esphome/esphome #264: RGB+W RGB+CW+WW light in which RGB part is fastled_based

wuub, 2018-12-15:
**Is your feature request related to a problem? Please describe.**

Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A description of what you want to happen.

**Additional context**
Add any other context about the feature request here.

OttoWinter, 2018-12-19:
I don't this is worthy of a component. And also it would definitely not work well with the light architecture. A `LightState` object has exactly one `LightOutput`.

Similar feature: https://github.com/OttoWinter/esphomelib/pull/147

closed 2018-12-19 by OttoWinter

wuub, 2018-12-19:
sure thing. makes sense. I'll just disable discovery and configure mqtt light myself on home-assistants side.