esphome/esphome #266: Run fails when trying to use fonts

bphermansson, 2018-12-18:
**Operating environment (Hass.io/Docker/pip/etc.):**

Docker, Peppermint Linux
**ESP (ESP32/ESP8266/Board/Sonoff):**

ESP8266/Nodemcu
**Affected component:**

https://esphomelib.com/esphomeyaml/components/display/index.html

**Description of problem:**
The firmware fails to build when adding a font-section to the device file. 

**Problem-relevant YAML-configuration entries:**
```yaml
font:
  - file: "Comic Sans MS.ttf"
    id: my_font
    size: 20
```

**Traceback (if applicable):**

```
docker run --rm -p 6123:6123 -v "`pwd`":/config -it ottowinter/esphomeyaml display.yaml run
INFO Reading configuration...
Failed config
 font: 
   - Invalid config for [font]: Please install the pillow python package to use this feature. (pip2 install pillow). Got [OrderedDict([('file', 'Comic Sans MS.ttf'), ('id', 'my_font'), ('size', 20)])].
   - [OrderedDict([('file', 'Comic Sans MS.ttf'), ('id', 'my_font'), ('size', 20)])]

```

**Additional information:**
I try to follow the instructions, but:

pip2 install pillow
Requirement already satisfied: pillow in /usr/lib/python2.7/dist-packages (5.1.0)

OttoWinter, 2018-12-19:
Fixed in next version

closed 2018-12-19 by OttoWinter

bkpsu, 2019-01-15:
This issue is still occurring when I try to validate any yaml file with a font section.

bkpsu, 2019-01-15:
> @bkpsu <https://github.com/bkpsu> This is the correct behavior when using
> this through pip. What has been fixed is that python pillow was not
> installed in the docker package.
>
> So for pip installations you *do* have to explicitly install pillow (this
> is because pillow can be difficult to install, so I don't force the
> installation for everyone).

Copy that, but I am using the Docker container (both latest and rc exhibit
the same behavior)

OttoWinter, 2019-01-15:
Ok just tried and you're correct! Will be fixed soon.