esphome/esphome #269: Improved documentation of OTA process for devices with small flash chips

brandond, 2018-12-19:
**Is your feature request related to a problem/use-case? Please describe.**
Using esphomeyaml ota with esp01_1m boards.

**Describe the solution you'd like:**
Improved documentation of OTA requirements for flash-constrained devices.

**Additional context:**
I have a number of devices based on esp01_1m and would love to be able to do OTA updates. I can successfully push a small image (just wifi + ota) to them, with a firmware.bin size of 370816 bytes. However, trying to switch back to a larger image - even one with just mqtt enabled, at 403200 bytes - always fails with a binary size error:

```
ERROR Error binary size: Error: Couldn't prepare flash memory for update. Is the binary too big? Please try restarting the ESP.
```

Is there any way to build an even smaller ota bootstrap image that would be more likely to leave space for the full image? Is there any way to calculate, log, or report the space available for ota on a given device?

pove, 2018-12-26:
Same error here, even with a firmware.bin of 347280 bytes.

brandond, 2018-12-27:
I've been considering trying to put together a super-minimal OTA update bin, but I'm not really sure where to start. Was hoping @OttoWinter might have some idea.

brandond, 2018-12-27:
I was actually looking at trying to build it in platformio with just the espressif sdk, since even a minimal esphomeyaml build seemed like it was too large to successfully OTA on top of. Sounds like some of that may be due to the dev Arduino core sucking though.

@ottowinter - do you think your OTA rewrite is portable enough to stick in a micro image?

brandond, 2018-12-27:
Also - when I OTA, where does the image go? Does it just get stuck in flash directly after the current image? Is there a 50/50 split partition for current/new image? I guess I'm wondering if I need to worry about flash fragmentation eating up space when I do OTA updates instead of flashing from the uart bootloader.

closed 2018-12-31 by brandond

brandond, 2018-12-31:
@pove you were on the right track with fixing the issue by uploading a sketch with no SPIFFS. Unfortunately, it looks like Platformio defaults to reserving 1/4 of the flash for SPIFFS, so any image built and uploaded via UART by esphomeyaml with those defaults will have reduced space for OTA.

I opened #284 to try to fix this.