esphome/esphome #274: Time

meijerwynand, 2018-12-26:
Greetings, 

I have made the following observations relating to the time component: (https://esphomelib.com/esphomeyaml/components/time.html )

Maybe I am overlooking some details and you/the community could assist in this regard, it would be highly appreciated.

### Server update intervals:
I am unable to see in the documentation the frequency of the server update intervals (i.e how often it queries the sntp server). 

### No warning / message on SNTP failure
It appears only successful SNTP actions are logged and not failed attempts (this example is using default (esphomeyaml) ntp servers as per documention.

```
...
[15:57:24][D][time.sntp:082]: Synchronized time: Wed Dec 26 13:57:23 2018
...
```

I am running an SNTP server to my router (Mikrotik). I am able to query and set time on my local PC 
```
$ ntpdate -q 192.168.13.1
server 192.168.13.1, stratum 3, offset -0.005066, delay 0.02580
26 Dec 16:13:58 ntpdate[25006]: adjust time server 192.168.13.1 offset -0.005066 sec

$ ntpdate -q pool.ntp.org
server 197.82.150.123, stratum 2, offset -0.008851, delay 0.04404
server 196.43.2.190, stratum 2, offset -0.008035, delay 0.05125
26 Dec 16:14:43 ntpdate[25019]: adjust time server 197.82.150.123 offset -0.008851 sec
```

The ESP does not appear to accept details from my Mikrotik / 192.168.13.1 / SNTP. What makes this a bit more challenging is that fact that no details about the reason / retries are logged.

```
# Enable logging
logger:
  level: DEBUG
```

### on_time not working without SNTP update

It appears if the device boots up *without* successful SNTP the ```on_time``` actions are ignored.

I have attempted with (pseudo code) both time and cron options
```
     - cron: * /5 * * * *
```
and 
```
      - seconds: 0-5
        minutes: /5
```

Use case:

Running a water pump on the hour
```
      - seconds: 0-5
        minutes: 0
        hours: '*'
        then:
          - light.turn_on: esp_5ccf7f2c0f45_d1
```

and shut it off 10 minutes later 
```
      - seconds: 0-5
        minutes: 10
        hours: '*'
        then:
          - light.turn_off: esp_5ccf7f2c0f45_d1
```

The pump is not that concerned about a precise moment of time, just that it delivers water in the cycles stipulated. However with the on_time not working due to SNTP / default epoch time, this creates a concern as the pump is now unable to reliably continue a watering cycle.

### Thoughts

I can not express the amount of joy and fun I have had since I have discovered your project. My most appreciative and thankful gratitude and wishes. 

You are awesome!

HFsi, 2018-12-26:
And what about using delay commands instead o sntp, being that you are not concerned about a specific moments but intervals?

OttoWinter, 2018-12-26:
SNTP is all handled by the LwIP core - AFAIK there's no way to get callbacks for failures and update intervals.

But... in the next version there will be a new time platform:

```yaml
time:
- platform: homeassistant
```

which will use the time of your HA instance.

meijerwynand, 2018-12-26:
> And what about using delay commands instead o sntp, being that you are not concerned about a specific moments but intervals?

Occam's Razor... I am a sucker for aesthetics/punctuation, "...its 21:00 and the pump should start...". However your suggestion of using delays it a good idea. Thank you.

I will use this if all other plans fail.

meijerwynand, 2018-12-26:
> SNTP is all handled by the LwIP core - AFAIK there's no way to get callbacks for failures and update intervals.
> 
> But... in the next version there will be a new time platform:
> 
> ```yaml
> time:
> - platform: homeassistant
> ```
> 
> which will use the time of your HA instance.

Thanks for the insights on the non-warning being displayed. As for the SNTP retry (see Server update intervals) and not executing and on_time command (see on_time not working without SNTP update), do you have any suggestions ?

Thanks

OttoWinter, 2018-12-27:
> SNTP retry

also handled by lwip

> not executing and on_time command 

well because the real time is not known. Consider an automation like this `minutes: 0, hours: 0`. So you'd only want it to trigger at midnight. But if we don't know the actual time, we should not execute an automation. (internally the check is if the time is before the year 2018 - and if so doesn't execute any triggers)

meijerwynand, 2018-12-28:
### SNTP / LwIP:

I looked around in the source code, thanks for pointing me to the LwIP. 

```
https://github.com/esp8266/Arduino/blob/master/tools/sdk/lwip/src/core/sntp.c
```

Just answering my own questions so the ticket can be closed:

* SNTP update: yes, every hour once successful communication was made
* SNTP retry: yes, it does and it also has a timeout
* SNTP debug: yes, it has the capability, have not looked further into this option

Why it is not working with the Mikrotik sntp server, that I still need to figure out

### No automation if SNTP failed

I am glad to hear my observation was correct. No SNTP = No Automation. Thank you for giving some more insight into the question.

closed 2018-12-28 by meijerwynand