esphome/esphome #276: Partial discovery only

ToSa27, 2018-12-26:
**Operating environment (Hass.io/Docker/pip/etc.):**

Hass.io running on Ubuntu.

**ESP (ESP32/ESP8266/Board/Sonoff):**

ESP8266 - NodeMCU v2.

**Affected component:**

Dallas - but issue may be independent of component.

**Description of problem:**
I have six Dallas temperature sensors attached to one pin. The detection etc. seems to work perfectly fine and temperature reading over time as well. The DEBUG log looks perfect. Out of the six only four are properly discovered by HA. As the DEBUG log shows all six but watching the MQTT broker only four messages show up in homeassistant/sensor/eg_heizung/… even after several reboots and recompiles, I could imagine it's something like a timing issue - sending the messages too fast running out of buffer or so. The four that show up in MQTT and in HA are: eg_heizung_zulauf, eg_heizung_ruecklauf_diele, eg_heizung_ruecklauf_essen, eg_heizung_ruecklauf_wohnenr

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: eg_heizung
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

mqtt:
  discovery: true
  discovery_retain: true

logger:
  level: DEBUG

dallas:
  - pin: GPIO14

sensor:
  - platform: dallas
    address: 0xE9800000265D3928
    name: "EG Heizung Zulauf"
  - platform: dallas
    address: 0x5805168661E4FF28
    name: "EG Heizung Ruecklauf Diele"
  - platform: dallas
    address: 0x6C031685443AFF28
    name: "EG Heizung Ruecklauf Kueche"
  - platform: dallas
    address: 0x700416857DA3FF28
    name: "EG Heizung Ruecklauf Essen"
  - platform: dallas
    address: 0x1B041685AFD3FF28
    name: "EG Heizung Ruecklauf WohnenR"
  - platform: dallas
    address: 0x360416857F77FF28
    name: "EG Heizung Ruecklauf WohnenL"
```

**Traceback (if applicable):**

```
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from eg_heizung/debug
[23:10:36][I][helpers:258]: Rebooting safely... Reason: 'ota'
[23:10:50][I][mqtt.client:115]: MQTT Connected!
[23:10:50][C][wifi:338]: WiFi:
[23:10:50][C][wifi:323]:   SSID: 'xxxxxxxxxx'
[23:10:50][C][wifi:324]:   IP Address: xxx.xxx.xxx.xxx
[23:10:50][C][wifi:326]:   BSSID: xx:xx:xx:xx:xx:xx
[23:10:50][C][wifi:328]:   Hostname: 'eg_heizung'
[23:10:50][C][wifi:330]:   Channel: 6
[23:10:50][C][wifi:331]:   Subnet: 255.255.255.0
[23:10:50][C][wifi:332]:   Gateway: xxx.xxx.xxx.xxx
[23:10:50][C][wifi:333]:   DNS1: xxx.xxx.xxx.xxx
[23:10:50][C][wifi:334]:   DNS2: 0.0.0.0
[23:10:50][C][sensor.dallas:079]: DallasComponent:
[23:10:50][C][sensor.dallas:080]:   Pin: GPIO14 (Mode: OUTPUT)
[23:10:50][D][sensor.dallas:085]:   Found sensors:
[23:10:50][D][sensor.dallas:088]:     0xE9800000265D3928
[23:10:50][D][sensor.dallas:088]:     0x5805168661E4FF28
[23:10:50][D][sensor.dallas:088]:     0x6C031685443AFF28
[23:10:50][D][sensor.dallas:088]:     0x700416857DA3FF28
[23:10:50][D][sensor.dallas:088]:     0x1B041685AFD3FF28
[23:10:50][D][sensor.dallas:088]:     0x360416857F77FF28
[23:10:50][C][sensor.dallas:093]:   Device 'EG Heizung Zulauf':
[23:10:50][C][sensor.dallas:101]:     Address: 0xE9800000265D3928
[23:10:50][C][sensor.dallas:102]:     Resolution: 12
[23:10:50][C][sensor.dallas:093]:   Device 'EG Heizung Ruecklauf Diele':
[23:10:50][C][sensor.dallas:101]:     Address: 0x5805168661E4FF28
[23:10:50][C][sensor.dallas:102]:     Resolution: 12
[23:10:50][C][sensor.dallas:093]:   Device 'EG Heizung Ruecklauf Kueche':
[23:10:50][C][sensor.dallas:101]:     Address: 0x6C031685443AFF28
[23:10:50][C][sensor.dallas:102]:     Resolution: 12
[23:10:50][C][sensor.dallas:093]:   Device 'EG Heizung Ruecklauf Essen':
[23:10:50][C][sensor.dallas:101]:     Address: 0x700416857DA3FF28
[23:10:50][C][sensor.dallas:102]:     Resolution: 12
[23:10:50][C][sensor.dallas:093]:   Device 'EG Heizung Ruecklauf WohnenR':
[23:10:50][C][sensor.dallas:101]:     Address: 0x1B041685AFD3FF28
[23:10:50][C][sensor.dallas:102]:     Resolution: 12
[23:10:50][C][sensor.dallas:093]:   Device 'EG Heizung Ruecklauf WohnenL':
[23:10:50][C][sensor.dallas:101]:     Address: 0x360416857F77FF28
[23:10:50][C][sensor.dallas:102]:     Resolution: 12
[23:10:50][C][ota:133]: Over-The-Air Updates:
[23:10:50][C][ota:134]:   Address: xxx.xxx.xxx.xxx:xxxx
[23:10:50][C][ota:136]:   Using Password.
[23:10:50][C][mqtt.client:054]: MQTT:
[23:10:50][C][mqtt.client:055]:   Server Address: xxx.xxx.xxx.xxx:xxxx
[23:10:50][C][mqtt.client:056]:   Username: 'xxxxxxxxxx'
[23:10:50][C][mqtt.client:057]:   Client ID: 'eg_heizung-xxxxxxxxxxxx'
[23:10:50][C][mqtt.client:059]:   Discovery prefix: 'homeassistant'
[23:10:50][C][mqtt.client:060]:   Discovery retain: YES
[23:10:50][C][mqtt.client:062]:   Topic Prefix: 'eg_heizung'
[23:10:50][C][mqtt.client:064]:   Log Topic: 'eg_heizung/debug'
[23:10:50][C][mqtt.client:067]:   Availability: 'eg_heizung/status'
[23:10:50][C][sensor.mqtt:028]: MQTT Sensor 'EG Heizung Zulauf':
[23:10:50][C][sensor.mqtt:030]:   Expire After: 675s
[23:10:50][C][sensor.mqtt:032]:   Unit of Measurement: '°C'
[23:10:50][C][sensor.mqtt:033]:   Accuracy Decimals: 1
[23:10:50][C][sensor.mqtt:038]:   Unique ID: 'dallas-E9800000265D3928'
[23:10:50][C][sensor.mqtt:040]:   State Topic: 'eg_heizung/sensor/eg_heizung_zulauf/state'
[23:10:50][C][sensor.mqtt:028]: MQTT Sensor 'EG Heizung Ruecklauf Diele':
[23:10:50][C][sensor.mqtt:030]:   Expire After: 675s
[23:10:50][C][sensor.mqtt:032]:   Unit of Measurement: '°C'
[23:10:50][C][sensor.mqtt:033]:   Accuracy Decimals: 1
[23:10:50][C][sensor.mqtt:038]:   Unique ID: 'dallas-5805168661E4FF28'
[23:10:50][C][sensor.mqtt:040]:   State Topic: 'eg_heizung/sensor/eg_heizung_ruecklauf_diele/state'
[23:10:50][C][sensor.mqtt:028]: MQTT Sensor 'EG Heizung Ruecklauf Kueche':
[23:10:50][C][sensor.mqtt:030]:   Expire After: 675s
[23:10:50][C][sensor.mqtt:032]:   Unit of Measurement: '°C'
[23:10:50][C][sensor.mqtt:033]:   Accuracy Decimals: 1
[23:10:50][C][sensor.mqtt:038]:   Unique ID: 'dallas-6C031685443AFF28'
[23:10:50][C][sensor.mqtt:040]:   State Topic: 'eg_heizung/sensor/eg_heizung_ruecklauf_kueche/state'
[23:10:50][C][sensor.mqtt:028]: MQTT Sensor 'EG Heizung Ruecklauf Essen':
[23:10:50][C][sensor.mqtt:030]:   Expire After: 675s
[23:10:50][C][sensor.mqtt:032]:   Unit of Measurement: '°C'
[23:10:50][C][sensor.mqtt:033]:   Accuracy Decimals: 1
[23:10:50][C][sensor.mqtt:038]:   Unique ID: 'dallas-700416857DA3FF28'
[23:10:50][C][sensor.mqtt:040]:   State Topic: 'eg_heizung/sensor/eg_heizung_ruecklauf_essen/state'
[23:10:50][C][sensor.mqtt:028]: MQTT Sensor 'EG Heizung Ruecklauf WohnenR':
[23:10:50][C][sensor.mqtt:030]:   Expire After: 675s
[23:10:50][C][sensor.mqtt:032]:   Unit of Measurement: '°C'
[23:10:50][C][sensor.mqtt:033]:   Accuracy Decimals: 1
[23:10:50][C][sensor.mqtt:038]:   Unique ID: 'dallas-1B041685AFD3FF28'
[23:10:50][C][sensor.mqtt:040]:   State Topic: 'eg_heizung/sensor/eg_heizung_ruecklauf_wohnenr/state'
[23:10:50][C][sensor.mqtt:028]: MQTT Sensor 'EG Heizung Ruecklauf WohnenL':
[23:10:50][C][sensor.mqtt:030]:   Expire After: 675s
[23:10:50][C][sensor.mqtt:032]:   Unit of Measurement: '°C'
[23:10:50][C][sensor.mqtt:033]:   Accuracy Decimals: 1
[23:10:50][C][sensor.mqtt:038]:   Unique ID: 'dallas-360416857F77FF28'
[23:10:50][C][sensor.mqtt:040]:   State Topic: 'eg_heizung/sensor/eg_heizung_ruecklauf_wohnenl/state'
[23:10:50][I][application:108]: Running through first loop()
[23:10:50][I][application:125]: First loop finished successfully!
[23:10:58][D][sensor.dallas:157]: 'EG Heizung Zulauf': Got Temperature=27.9°C
[23:10:58][D][sensor.dallas:157]: 'EG Heizung Ruecklauf Diele': Got Temperature=24.8°C
[23:10:58][D][sensor.dallas:157]: 'EG Heizung Ruecklauf Kueche': Got Temperature=25.1°C
[23:10:58][D][sensor.dallas:157]: 'EG Heizung Ruecklauf Essen': Got Temperature=23.9°C
[23:10:58][D][sensor.dallas:157]: 'EG Heizung Ruecklauf WohnenR': Got Temperature=23.9°C
[23:10:58][D][sensor.dallas:157]: 'EG Heizung Ruecklauf WohnenL': Got Temperature=24.1°C
[23:11:13][D][sensor.dallas:157]: 'EG Heizung Zulauf': Got Temperature=27.9°C
[23:11:13][D][sensor.dallas:157]: 'EG Heizung Ruecklauf Diele': Got Temperature=24.8°C
[23:11:13][D][sensor.dallas:157]: 'EG Heizung Ruecklauf Kueche': Got Temperature=25.1°C
[23:11:13][D][sensor.dallas:157]: 'EG Heizung Ruecklauf Essen': Got Temperature=23.9°C
[23:11:13][D][sensor.dallas:157]: 'EG Heizung Ruecklauf WohnenR': Got Temperature=23.9°C
[23:11:13][D][sensor.dallas:157]: 'EG Heizung Ruecklauf WohnenL': Got Temperature=24.1°C
```

**Additional information:**

HFsi, 2018-12-26:
Did you try to define manually the sensors in configuration.yaml?
It's the only way I found to make all sensors work until (hopefully) 1.10

ToSa27, 2018-12-27:
My work-around would be to send the discovery message manually from an MQTT client. Did not try this yet as I wanted to see first if this is a known issue or I did something wrong... seems like I‘m not the only one experiencing this. Do you know there will be a fix in 1.10 or just hope for it?

voicevon, 2018-12-27:
In my case:
RespB3+, Hassio, Esp32, 12 Dallas sensors on one pin, interval = 2s.
In front end, I am using Mqtt-dash on android. Everything is OK for more than 1 week time. But I didn't use any HA auto discovery.

meijerwynand, 2018-12-27:
If I infer correctly from your post and issue:
You have 6 'devices' of which only 4 'devices' are discovered in HA, hence the "partial discovery".

I have had a similar issue in the past where not all the components on a single MCU will be discovered. 

I have found switching over to Mosquito MQTT instead of the built-in HASS/HA-MQTT has resolved this issue for me.

* Stop HA
* Install and config Mosquito 
* Change configuration.yaml and set broker IP
* Start Mosquito
* Start HA

Not sure what the Hassio equivalent would be.