esphome/esphome #277: esphomeyaml-edge: hass.io addon not starting

TheJulianJES, 2018-12-27:
**Operating environment (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266/Board/Sonoff):**

**Affected component:**

**Description of problem:**
Tried rebuilding the latest esphomeyaml-edge and I couldn't login via my hass.io credentials. Then I reinstalled the addon because it also didn't show the updated badges (like auth badge -> maybe the config.json doesn't get "checked again" when rebuilding) in the hassio interface. Now for some reason the network tab is missing (where you configure the port in hassio) and this is the error I see in the logs: https://pastebin.com/raw/SihacNUK

**Problem-relevant YAML-configuration entries:**
(Not relevant) / default hass.io addon config:
```yaml
{
  "ssl": false,
  "certfile": "fullchain.pem",
  "keyfile": "privkey.pem"
}
```

**Traceback (if applicable):**
 You are running the latest version of this add-on
[cont-init.d] 02-updates.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing... 
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[17:12:03+0100] INFO ----> Starting NGINX...
[services.d] done.
[17:12:03+0100] INFO ----> Starting esphomeyaml dashboard...
INFO Starting dashboard web server on port 80 and configuration dir /config/esphomeyaml...
Traceback (most recent call last):
  File "/usr/local/bin/esphomeyaml", line 11, in <module>
    load_entry_point('esphomeyaml==1.10.0.dev0', 'console_scripts', 'esphomeyaml')()
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 494, in main
    return run_esphomeyaml(sys.argv)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 470, in run_esphomeyaml
    return PRE_CONFIG_ACTIONS[args.command](args)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 360, in command_dashboard
    return dashboard.start_web_server(args)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/dashboard/dashboard.py", line 554, in start_web_server
    app.listen(args.port)
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 2042, in listen
    server.listen(port, address)
  File "/usr/local/lib/python2.7/dist-packages/tornado/tcpserver.py", line 143, in listen
    sockets = bind_sockets(port, address=address)
  File "/usr/local/lib/python2.7/dist-packages/tornado/netutil.py", line 168, in bind_sockets
    sock.bind(sockaddr)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use
[17:12:04+0100] INFO ----> Starting esphomeyaml dashboard...
INFO Starting dashboard web server on port 80 and configuration dir /config/esphomeyaml...
Traceback (most recent call last):
  File "/usr/local/bin/esphomeyaml", line 11, in <module>
    load_entry_point('esphomeyaml==1.10.0.dev0', 'console_scripts', 'esphomeyaml')()
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 494, in main
    return run_esphomeyaml(sys.argv)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 470, in run_esphomeyaml
    return PRE_CONFIG_ACTIONS[args.command](args)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 360, in command_dashboard
    return dashboard.start_web_server(args)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/dashboard/dashboard.py", line 554, in start_web_server
    app.listen(args.port)
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 2042, in listen
    server.listen(port, address)
  File "/usr/local/lib/python2.7/dist-packages/tornado/tcpserver.py", line 143, in listen
    sockets = bind_sockets(port, address=address)
  File "/usr/local/lib/python2.7/dist-packages/tornado/netutil.py", line 168, in bind_sockets
    sock.bind(sockaddr)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use
[17:12:05+0100] INFO ----> Starting esphomeyaml dashboard...
INFO Starting dashboard web server on port 80 and configuration dir /config/esphomeyaml...
Traceback (most recent call last):
  File "/usr/local/bin/esphomeyaml", line 11, in <module>
    load_entry_point('esphomeyaml==1.10.0.dev0', 'console_scripts', 'esphomeyaml')()
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 494, in main
    return run_esphomeyaml(sys.argv)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 470, in run_esphomeyaml
    return PRE_CONFIG_ACTIONS[args.command](args)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/__main__.py", line 360, in command_dashboard
    return dashboard.start_web_server(args)
  File "/usr/local/lib/python2.7/dist-packages/esphomeyaml/dashboard/dashboard.py", line 554, in start_web_server
    app.listen(args.port)
  File "/usr/local/lib/python2.7/dist-packages/tornado/web.py", line 2042, in listen
    server.listen(port, address)
  File "/usr/local/lib/python2.7/dist-packages/tornado/tcpserver.py", line 143, in listen
    sockets = bind_sockets(port, address=address)
  File "/usr/local/lib/python2.7/dist-packages/tornado/netutil.py", line 168, in bind_sockets
    sock.bind(sockaddr)
  File "/usr/lib/python2.7/socket.py", line 228, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 98] Address already in use
[17:12:06+0100] INFO ----> Starting esphomeyaml dashboard...
-->
```

```

**Additional information:**
Dashboard shows:
[image]

OttoWinter, 2018-12-27:
You have another service running on the 6052 port (or the port you configured in the Hass.io addon settings).

It's pretty clear because the page you're seeing there is the apache setup page, but esphomeyaml uses nginx. I bet you you will still see that page when thr add-on is not running.

Closing as configuration issue.

OttoWinter, 2018-12-27:
As for the missing network tab, it's probably due to a bug in Hass.io - please check yiu're using the latest version of that.

closed 2018-12-27 by OttoWinter

TheJulianJES, 2018-12-27:
Hey, thanks for the quick response!
I am actually running that page on port 80 (proxy stuff for hass).
But in the screenshot already provided the page is somehow "relayed" to the 6052 port which should display the esphomeyaml dashboard.

(It seems like the dashboard somehow starts on port 80 (which is occupied) and the apache server which is running there somehow gets "relayed" to the port 6052.)

Here you can see that this is not happening when the (edge) addon is shutdown.
[image]

Hass.io supervisor version 141 is currently the latest (which I'm running).
 
I already re-added the repo and reinstalled the addon two times, but I'll try again shortly.

TheJulianJES, 2018-12-27:
Uninstalled all esphomeyaml addons, removed the repo again.
Added the repo and only installed the esphomeyaml-edge addon and the network tab is still missing.
I'm not sure if this is a hass.io issue or a config.json issue of the esphomeyaml-edge addon.
I'll try to see if I can come up with a solution and will try installing the addon on a clean hass.io setup.

TheJulianJES, 2018-12-27:
Setting host_network to false (or removing the line) in the esphomeyaml-edge/config.json solved the issue.
(Tested using a fork)
Of course this is "against" the point of hass.io's implementation but it seems like the network tab only shows up, when host_network is disabled (like it is in the current esphomeyaml and esphomeyaml-beta addon)
```"host_network": false,```

When doing this, the addon correctly starts on port 6052 and also shows the correct dashboard.

TheJulianJES, 2018-12-27:
It seems like this is done intentionally on the hass.io side:
https://github.com/home-assistant/hassio/blob/dev/hassio/addons/addon.py#L261-L264

TheJulianJES, 2018-12-27:
Updated the issue to (hopefully) make it more clear on what's actually going on.
Sorry, because of the Apache part. It's running on port 80 and was displayed only when the addon was started also on port 6052 (due to the missing network tab -> causing the addon trying to start on port 80).