esphome/esphome #278: Reusable Firmware Builds / Device+Component Name Templating

brandond, 2018-12-28:
**Is your feature request related to a problem/use-case? Please describe.**
I would like to be able to flash the same firmware on multiple identical devices and have device and component names auto-generated based on the device ID.

**Describe the solution you'd like**
I would like to auto-name the device as well as the sensors/switches using some portion of the device ID or MAC address. Something like this perhaps, where the `{id:x}` portion of the device and component names were replaced at runtime with the last `x` characters of the id.

```yaml
esphomeyaml:
  name: 'smartplug-{id:4}'
  platform: ESP8266
  board: esp01_1m
  arduino_version: dev
  esphomelib_version: dev
  board_flash_mode: dout
  build_path: plugs

wifi:
  ssid: 'ssid'
  password: 'password'

mqtt:
  broker: '10.0.1.27'
  discovery_prefix: homeassistant
  username: ''
  password: ''

logger:

sensor:
  - platform: wifi_signal
    name: "Smart Plug {id:4} WiFi Signal"
    filters:
      - sliding_window_moving_average:
          window_size: 4
          send_every: 4

switch:
 - platform: gpio
   id: switch0
   pin:
     number: 14
     mode: OUTPUT
 - platform: template
   id: template0
   icon: "mdi:power-plug"
   name: "Smart Plug {id:4} Power"
   restore_state: yes
   turn_off_action:
     - light.turn_off:
         id: led0_light
         transition_length: 0.5s
     - switch.turn_off:
         id: switch0
   turn_on_action:
     - light.turn_on:
         id: led0_light
         transition_length: 0.5s
     - switch.turn_on:
         id: switch0
   lambda: "return id(switch0).state;"

binary_sensor:
  - platform: gpio
    id: button0
    pin:
      number: 4
      mode: INPUT
    on_press:
      then:
        - switch.toggle:
            id: template0

status_led:
  pin:
    number: 12
    mode: OUTPUT
    inverted: true

output:
  - platform: esp8266_pwm
    id: led0_output
    pin:
      number: 13
      mode: OUTPUT
      inverted: true

light:
  - platform: monochromatic
    id: led0_light
    output: led0_output
```

**Additional context**
I have a number of 'Tuya Smart' WiFi plugs that have an ESP-01M embedded in them. I've successfully reflashed them with esphomelib, but the process of building the firmware for each of them is somewhat irritating, in that I have to open the YAML file, replace all instances of '01' with '02', compile, flash, repeat until they're all done. Additionally, I have to manually keep track of which is which lest I accidentally flash more than one as 'Smart Plug 01'.

It would save me a great deal of time and manual device tracking if I could just build the firmware once and flash it to all of the devices, and have them auto-name themselves.

brandond, 2018-12-29:
I hacked up a quick and dirty POC, which works OK but isn't great since you have to manually rename each of the Nameable components, as well as the node itself and the topic prefix.

```
  std::string placeholder = "{id}";
  std::string short_id = get_mac_address().substr(8, 4);
  std::string::size_type pos;
  std::string name;

  name = App.get_name();
  pos = name.find(placeholder);
  if (pos != std::string::npos){
      name.replace(pos, 4, short_id);
      App.set_name(name);
      mqtt_mqttclientcomponent->set_topic_prefix(name);
  }

  name = sensor_wifisignalsensor->get_name();
  pos = name.find(placeholder);
  if (pos != std::string::npos){
      name.replace(pos, 4, short_id);
      sensor_wifisignalsensor->set_name(name);
  }

  name = template0->get_name();
  pos = name.find(placeholder);
  if (pos != std::string::npos){
      name.replace(pos, 4, short_id);
      template0->set_name(name);
  }
```

voicevon, 2018-12-30:
Based on my understanding, this feature is not suitable for esphomelib, unless you never use `lambda` and `automation`.Then reason is
1)  the `lambda` is compiled as a `function`, and that function will be called (use callback method) somewhere ,normally is a trigger.  
2) The "link"  between a `trigger `and `action `is hard coded in esphomelib.
 
In theory, the solution is 
1) Replace  `lambda` and the "link" to some kind of  "script_2".  
2) The `callback (*f)` is still there, and the execution will be started at a certain of  entrance of the "script_2" . 
3) The "script_2" will be explained to some code, the real execution is on those code.
4) When the execution is finished, return to where is the callback.
 
This will cause a big change of the framework of esphomelib.  I am not sure @OttoWinter has started to go this way or not, I noticed the new version(Ver1.9.3) supports `script_1 ` already. But the goal of  `script_1`  is to reduce repeating yaml code, the "link" is hard coded,  the execution of code is compiled, no "script_2" at all, no big change of framework of esphomelib.

brandond, 2018-12-30:
Yeah I'm not sure what you're on about. All I'm doing is changing the hostname and component names. The code I included above goes after the autogenerated portion of main.cpp, not in a Lambda. I will note that I also had to patch esphomeyaml to allow {} in the name, which it will not by default.

It would be nice to have something that provides similar functionality in the core, instead of having to hack it in. I can't be the only one with a bunch of identical devices, these things come in four and ten packs.

OttoWinter, 2018-12-30:
No, this won't happen. The whole thing that makes esphomeyaml work is that esphomeyaml knows *exactly* how the node will behave when it starts up. That includes knowing how to reach the ESP (`wifi.hostname`/`esphomeyaml.name`). If that is parameterized, esphomeyaml will no longer be able to connect to the node directly - so many thing like OTA, new native API, ping, HA config gen etc. won't work.

Apart from that, the changes required in esphomelib would be massive. Each method that is passed a `name` would need to get a different method signature. Also, in esphomeyaml many methods that directly process the string would need to have the `{id}` pattern whitelisted manually - For example `esphomeyaml.name: smartplug_{id}` would fail because `{` is an invalid char for names. For `lambdas` this wouldn't work anyway (except if we would bundle a whole C++ compiler with the firmware, which of course won't happen)

And then comes the problem that users will complain if option A doesn't support `{id}` substitution etc. The list of stuff that would need to happen just goes on an on. I suggest you use the dashboard view of esphomeyaml and OTA uploads, then you can easily flash many devices quickly.

(Of course compilation speed is an issue then, but that's another issue. Let's not fight symptoms here, there are plenty of ways we can still explore to increase compilation speed.)

closed 2018-12-30 by OttoWinter

escoand, 2018-12-30:
Isn't something like this working (not tested) if you set a fixed IP address?

```yaml
esphomeyaml:
  on_boot:
    then: !lamba |>
      auto wifi = id(...);
      auto name = wifi.get_hostname();
      auto mac = get_mac_address();
      auto pos = name.find("{id}");
      name.replace(pos, 4, mac);
      wifi.set_hostname(name);
```

escoand, 2018-12-30:
Or maybe not use replace put append the mac address.

quazzie, 2019-02-11:
Couldn't this be handled just in the dashboard without affecting esphomelib at all ?
It would still generate a yaml/firmware for each node but collapse them on the dashboard.
Some kind of templating system to generate/edit the yaml-files.
I agree that this is something needed, its getting ridiculous when you have 20 nodes with the same yaml except the name of the node/entities.
Hmm when i get some time i need to see if i can't make a PR.

escoand, 2019-02-11:
Or maybe some kind of include mechanism? https://www.home-assistant.io/docs/configuration/splitting_configuration/

OttoWinter, 2019-02-11:
@escoand Please read the FAQ. That's literally the first line in there.