esphome/esphome #279: Does script support lambda?

voicevon, 2018-12-30:
**Operating environment (Hass.io/Docker/pip/etc.):**
Hassio
**ESP (ESP32/ESP8266/Board/Sonoff):**
Esp8266

**Affected component:**
script

**Description of problem:**
Can't pass validation.

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: test
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: 'abc'
  password: '1234567890'

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: '1234567890'

ota:
  password: '1234567890'

script:
  - id: my_script
    then:
      - delay: 1s
      
     # without the below code, validation is ok.
      - lambda: |
        if(1<2)
        {}
```

**Traceback (if applicable):**

```
INFO Reading configuration...
ERROR Error while reading config: while scanning a simple key
  in "/config/esphomeyaml/test.yaml", line 25, column 9
could not find expected ':'
  in "/config/esphomeyaml/test.yaml", line 26, column 9
```

**Additional information:**
I am not sure this is a bug , or `script` doesn't support `lambda`. anyway, the` INFO` is not helping me.

voicevon, 2018-12-30:
I forgot to put extra space for the body of lambda.

voicevon, 2018-12-30:
But could esphomeyaml show some friendly info?

OttoWinter, 2018-12-30:
It's a YAML read error. These errors are generated by PyYAML. Unfortunately that library is not the best with error messages, but it's the best we can do - esphomeyaml will **not** create its own YAML parser.

closed 2018-12-30 by OttoWinter