esphome/esphome #283: Current SNTP time for device

meijerwynand, 2018-12-31:
**Is your feature request related to a problem/use-case? Please describe.**
Display the current SNTP time as a component. To be able to see what the current time is on the device. In case of a failed SNTP update to be easier to identify or troubleshoot the issue.

**Describe the solution you'd like:**
Similar to the wifi signal or uptime sensors, 
```
sensor:
  - platform: sntp_time
    name: "esp_xxx_system_sntp_time"
    id: esp_xxx_system_sntp_time
    format_time: "%c"
    update_interval: 15s
```

**Additional context:**
 I have been unsuccessful to display the current time into HA using lamba's or to find a good/working example. If such a option is possible, please supply the correct method / code how to achieve it and I will gladly add to the cookbook docs.

Thanks

brandond, 2018-12-31:
Seems like you could do this pretty easily using:
https://esphomelib.com/esphomeyaml/components/time.html#use-in-lambdas
and
https://esphomelib.com/esphomeyaml/components/text_sensor/template.html

Just have the `text_sensor` lambda print `id(sntp_time).now().time` or something like that.

meijerwynand, 2018-12-31:
Thank you for the suggestion,

I suspected it might be something obvious / pretty easy as you alluded to, however something is always obvious once you know how to do it or how it works... :)

Having an extremely limited C++ background, I infer lambdas allow you to run code directly.

Thus from my understanding I can return the string (the formatted time) by using the ```text_sensor```component using the ```return``` value.

This is where it gets a bit tricky and I need some assistance.

- Examples found in the documentation for ```strftime``` all reference a '[display object](https://esphomelib.com/esphomeyaml/components/time.html#strftime)', there is no "text only"-example
- I tried to look at the c++ manual on both functions but that just become more confusing and frustrating [printf](http://www.cplusplus.com/reference/cstdio/printf/) and [strftime](http://www.cplusplus.com/reference/ctime/strftime/).
- I have tried various combinations ```printf``` and ```strftime``` but it fails compiling every time

I _really_ have tried to the best of my knowledge but I am unable to resolve this matter myself...

> Just have the text_sensor lambda print id(sntp_time).now().time or something like that.

Could you please provide a working example showing ```YYYY-MM-DD hh:mm:ss``` (format "%Y-%m-%d %H:%M:%S" if I understand correctly), this was where I ended...

```
text_sensor:
  - platform: template
    name: "Current time"
    lambda: |-
      auto time = id(sntp_time).now();
      return {printf("%Y-%m-%d %H:%M:%S", time)};
    update_interval: 15s
```
I would like to 

I will gladly contribute to the documentation to have this readily available for others who would like to make use of this method.

Thank you

meijerwynand, 2019-01-03:
Aha! well, sort of...

This appears to be a way of using the ```text_sensor``` to display the SNTP time in text on HA

```
text_sensor:
  - platform: template
    name: "Current time"
    lambda: |-
      char str[17];
      time_t currTime = id(sntp_time).now().time;
      strftime(str, sizeof(str), "%Y-%m-%d %H:%M", localtime(&currTime));
      return { str };
    update_interval: 15s
```

I get in my debug logs
```
[20:03:35][D][text_sensor.text_sensor:017]: 'Current time': Sending state '1970-01-01 08:00'
[20:03:35][D][time.sntp:082]: Synchronized time: Thu Jan  3 18:03:33 2019
[20:03:46][D][text_sensor.text_sensor:017]: 'Current time': Sending state '2019-01-03 18:03'
[20:04:01][D][text_sensor.text_sensor:017]: 'Current time': Sending state '2019-01-03 18:04'

```

However, once I have enabled this I need to disable other features (in my case ```sensor.platform: uptime``` and ```sensor.platform: wifi_signal``` as the sketch is then too big to fit onto the Sonoff Basic (RR R2 V1.0)

@brandond would you mind sharing some knowledge whether this approach is correct using ```lambda's``` or if there is a more efficient method to attempt.

Thanks

meijerwynand, 2019-01-03:
@brandond 

Dev release : I will look into using it in the very near future. Just need to find the ladder and soldering iron for some of the devices, it's going to be a full day's work.

SNTP: Glad to hear I am heading in the general direction, I appreciate the ```text_sensor``` suggestion.

Thanks for the feedback.

closed 2019-01-03 by meijerwynand