esphome/esphome #284: Platformio default board config reserves excessive space for SPIFFS

brandond, 2018-12-31:
**Operating environment (Hass.io/Docker/pip/etc.):**
pip in virtualenv

**ESP (ESP32/ESP8266/Board/Sonoff):**
esp01_1m

**Affected component:**
core / flash image

**Description of problem:**
The platformio board configuration for esp8266 boards reserves unnecessary space for SPIFFS - 64k on 512k boards, 256k on 1m boards, and 1m on 4m boards. Since OTA updates can only write to the space between the end of the current sketch and the start of SPIFFS, this means that 1m boards cannot OTA images over about 384k.

The config included below comes out to 300k on stable versions of the arduino framework, and 360k on dev, which is fine. However, adding any additional components (even just mqtt) will push it over the 384k limit past which OTA will run out of space to upload anything useful.

The easy fix for this (assuming esphomeyaml doesn't plan on making use of SPIFFS any time soon) is to disable SPIFFS. This can be done by modifying platformio.ini's environment config: `board_build.ldscript = eagle.flash.1m.ld`

Rather than have esphomeyaml maintain its own board list (although it kind of already does, for pin names) it would probably be easiest to simply document this limitation, and expose the platformio setting, similar to `flash_mode`, so that users can override it if necessary.

Alternately, if the intent is to default to no SPIFFS, you could extract the appropriate SPIFFS-free ldscripts for the various boards and set this option by default.

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: minimal
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: 'ssid'
  password: 'password'

ota:

logger:
  level: VERY_VERBOSE
```

OttoWinter, 2019-01-01:
In general, yes that's a good idea.

The problem that has stopped me from doing so by default is https://github.com/esp8266/Arduino/pull/5103. 

the linker scripts that we'd need to use have been renamed in 2.5.0 - which would break esphomeyaml whenever that is released and the user is using `arduino_version: latest` + for `arduino_version: dev`

brandond, 2019-01-02:
Right, so it seems like you can't just hardcore the ldscript value unless/until you require framework 2.5.0 or newer. In the mean time, would you accept a YAML+docs MR that allowed it to be passed through like flash_mode?

closed 2019-01-02 by OttoWinter

brandond, 2019-01-02:
Awesome, thanks! In the same vein, you might also consider setting `board_build.partitions = min_spiffs.csv` for non-BLE esp32 builds; it gives you another 640k per app section.