esphome/esphome #286: More triggers

escoand, 2019-01-01:
I've stumbled over the new Shelly H&T and tried to create it by myself with my existing Wemos D1 Mini. But IMO there are some triggers missing like:

* wifi.on_connect
* mqtt.on_connect
* sensor.on_published

With this it would be possible to measure and going to deep sleep as soon as possible. Currently you have to "poll" if connected and wait some time after publish to be sure it's on the net and going to sleep. I realized some missing telegrams when not waiting 1 second.

escoand, 2019-01-01:
My current configuration looks like this:

```yaml
deep_sleep:
  id: deep_sleep_
  sleep_duration: 10min

sensor:
  - platform: adc
    pin: A0
    id: temperature
    name: Temperatur
    unit_of_measurement: "°C"
    accuracy_decimals: 1
    update_interval: never
    expire_after: 11min
    icon: mdi:thermometer
    filters:
      - lambda: !lambda >-
          static double uMax = 1.0;
          static double tStatic = 25 + 273.15;
          static double beta = 4090;
          static double rStatic = 10000;
          double rThermistor = rStatic * (uMax / x - 1);
          double tKelvin = 1.0 / ( (1.0/tStatic) + (1.0/beta)*log(rThermistor/rStatic) );
          return tKelvin - 273.15;
    on_value:
      - delay: 1s
      - lambda: 'id(deep_sleep_).begin_sleep_();'
  - platform: template
    id: update_temperature
    update_interval: 1s
    lambda: |-
      if(!id(mqtt_).is_connected()) { return {}; }
      id(thermistor).turn_on();
      id(temperature).update();
      id(thermistor).turn_off();
      return {};
```

I use the yaml delay and not the api delay because [it's "smart"](https://esphomelib.com/esphomeyaml/guides/automations.html#delay-action).

escoand, 2019-01-02:
Another cool feature for the Shelly H&T replica would be to power the device up without wifi starting. So we could measure and just connect and transmit the data if something changed.
Of course this is possible at the moment but not with the default yaml options.

OttoWinter, 2019-01-02:
`sensor.on_published` -> `sensor.on_value`
`mqtt.on_connect` `binary_sensor.status.on_press`
`wifi.on_connect` doesn't exist yet though (though you can use `update_interval: 0s` for immediate response; doesn't really affect performance)

> Another cool feature for the Shelly H&T replica would be to power the device up without wifi starting. 

Not going to happen in YAML (nor lib, AFAIK this isn't possible atm). That would kind of break the whole setup priority thing of esphomelib components. Plus I don't think it's really necessary.

escoand, 2019-01-02:
Thanks, seems to work really well except of this error:
```
src/main.cpp: In function 'void setup()':
src/main.cpp:58:51: error: expected type-specifier before '::' token
::UpdateComponentAction<NoArg> *_action_4 = new ::UpdateComponentAction(temperature);
```

The config (the commented line works):
```yaml
binary_sensor:
  - platform: status
    id: status
    on_press:
      - output.turn_on: thermistor
      - component.update: temperature
#      - lambda: 'id(temperature).update();'
      - output.turn_off: thermistor
      - delay: 1s
      - lambda: 'id(deep_sleep_).begin_sleep_();'

sensor:
  - platform: adc
    pin: A0
    id: temperature
    ...
```

closed 2019-01-08 by OttoWinter