esphome/esphome #306: Neo Coolcam PIR, help appreciated

Styx85, 2019-01-07:
I recently ordered one of these [NEO Coolcam PIR WIFI sensors ](https://de.aliexpress.com/item/NEO-COOLCAM-Wei-WiFi-PIR-Motion-Sensor-Detektor-Home-Alarm-System-Super-Mini-Form/32908948741.html).

The PIR itself is quite handy and small (sized like a golf ball).
When you take it apart you find a ESP8266EX and a lot of pinholes to connect to. Have a look at forum diskussion [here](https://community.home-assistant.io/t/coolcam-wifi-motion-sensor-pir/54783). 

The standard software is placed in the tuya/smart life universe and the PIR supports IFTT. Some sellers mention amazon alexa and google home support, but that is not integrated (yet?).

The original behaviour of the PIR is to detect movement, give a signal with the integrated LED's when movement is detected and return to deep sleep if there isn't any movement for 60 seconds.

[image]

when connecting the marked pins it is possible to flash the pir and extract the original firmware.
 the esphomeyaml I created so far looks like this:

````
esphomeyaml:
  name: pir2
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: MyHomeNetwork
  password: VerySafePassword
  domain .veryowndomain

mqtt:
  broker: ip
  username: user
  password: MyMQTTPassword

logger:
  level: VERY_VERBOSE
debug:

ota:
  password: passwd

binary_sensor:
  - platform: gpio
    pin: 
#      number: GPIO1
#      number: GPIO2
#      number: GPIO3
#      number: GPIO4
#      number: GPIO5
#      number: GPIO12
#      number: GPIO13
#      number: GPIO14
#      number: GPIO15
    name: "esp_xxx_pir"
    device_class: motion
 ## https://esphomelib.com/esphomeyaml/components/binary_sensor/status.html
  - platform: status
    name: "esp_xxx_pir_system_status"
    id: "esp_xxx_pir_system_status"

sensor:
  ## https://esphomelib.com/esphomeyaml/components/sensor/uptime.html
  - platform: uptime
    name: "esp_xxx_pir_system_uptime"
    id: esp_xxx_pir_system_uptime

  ## https://esphomelib.com/esphomeyaml/components/sensor/wifi_signal.html
  - platform: wifi_signal
    name: "esp_xxx_pir_system_wifi_signal"
    id: esp_xxx_pir_system_wifi_signal
    update_interval: 15s
````

With that configuration boot-up is working, mqtt is working wifi-OTA is working. 
So far none of the GPIO's was sucessful in keeping it alive or sending a signal. Exactly after 60 seconds the PIR is dead and has to be repowered manually.
So I think I have to figure out how to get the signal after deep sleep wake-up (that has to happen without programming I think).

Would it be wise to set the GPIO's to OUTPUT mode?
Any ideas how to procede?

OttoWinter, 2019-01-07:
> So far none of the GPIO's was sucessful in keeping it alive or sending a signal.

What do you mean by that? Also, from the config it looks like you haven't tried pin 16 yet. The PIR sensor is pretty likely connected to pin 16 as that's the pin that supports waking from deep sleep.

> Exactly after 60 seconds the PIR is dead and has to be repowered manually.

So the ESP crashes after exactly 60 seconds? That's weird.

> Would it be wise to set the GPIO's to OUTPUT mode?

No

OttoWinter, 2019-01-08:
> Also, I don't think there's any great support for deep sleep at the moment, is there? As far as esphomelib is concerned, it's just another powerup; the only difference is the reset reason, right?

https://esphomelib.com/esphomeyaml/components/deep_sleep.html

> Also, iirc GPIO16 has to be wired up to the reset pin in order to be used as a wakeup timer, right?

That's if you want to it in wakeup "timer" mode. But I think in this case the product is designed in such a way that the RST pin is pulled high by the PIR directly, so the ESP will boot up exactly when the PIR sensor activates.

> at which point the system powers up again with all memory reset as if from a cold boot.

except for RTC memory, which esphomelib uses as a store for all of its persistent data with ESP8266s. But I think for this application persistent data isn't required anyway?

brandond, 2019-01-08:
By 'great support' I meant there wasn't any special handling for deepsleep wakeup. The sleep itself is definitely well supported - but when it boots it still has to go through the whole init, associate, dhcp, connect to mqtt, etc process as if it were a cold boot. There are no shortcuts or speedups. Any handling of determining a sensor wakeup vs timer wakeup would need to be done manually, correct?

Styx85, 2019-01-08:
> > So far none of the GPIO's was sucessful in keeping it alive or sending a signal.
> 
> What do you mean by that? Also, from the config it looks like you haven't tried pin 16 yet. The PIR sensor is pretty likely connected to pin 16 as that's the pin that supports waking from deep sleep.
> 
I forgot GPIO16, true. I believed like @brandond that it was connected to the reset pin and just doing its magic and I just have to fetch some output signal tu use it in HA

> > Exactly after 60 seconds the PIR is dead and has to be repowered manually.
> 
> So the ESP crashes after exactly 60 seconds? That's weird.
> 
no it doesn't crash. It's the standard behaviour of the PIR and it seems it is realized in hardware that it goes into deepsleep after 60 seconds. if you put your ear next to the board it's probably realized with some electric coil or inductor.
> > Would it be wise to set the GPIO's to OUTPUT mode?
> 
> No
straight answers.
thanks very much, I'll tinker a little more and report what I found.

Styx85, 2019-01-08:
> By 'great support' I meant there wasn't any special handling for deepsleep wakeup. The sleep itself is definitely well supported - but when it boots it still has to go through the whole init, [...]

To wait for the boot process is a general problem of the esp module used here. the vendor seems to circumvent the problem in prolonging the PIR signal with some kind of timer or filter. 
The process is as follows.After triggering the PIR you get a LED blink after 0.2 seconds. And shortly thereafter a signal in the tuya ap that there is a movement. That could mean that the LED signals the completed boot process and ahortly afterwards a wifi connection is established to transfer the signal.
As with the hardware implemented deep sleep only a sensor wakeup is possible (another guess so far).
I'll try some things and report.

Styx85, 2019-01-08:
After the last test GPIO16 itself is also giving no signal, too.
So the circuit must be somewhat different. Maybe like this one described [here](https://stackoverflow.com/questions/39481196/how-to-wake-esp8266-from-deep-sleep-without-continuous-resets). I'll test it another time.

brandond, 2019-01-08:
I just ordered one off Alibaba since it seemed like an interesting piece of hardware, and I've been meaning to play with PIR sensors anyway. Won't be here for about a month though :) I've got a logic analyzer so at the very least I should be able to do some tracing on the stock firmware.

ciscozine, 2019-01-14:
I'm interested in this discussione because I have 2 PIR sensors.
Anyone is able to flash with custom firmware?
Thanks

brandond, 2019-01-17:
My device just came in. Haven't got much further than cracking it open and putting a barebones esphomelib firmware on it, which was pleasantly easy to do given the exposed vias.

I have a sneaking suspicion that the secondary MCU controls both the LEDs and the PIR, and the ESP is only woke if some internal threshold is triggered on that side. Other bits of stock functionality like blinking the LEDs at powerup, illumining them when the button is held down, etc all seem to work even with alternate firmware loaded.

If that's the case, there's probably some additional firmware running on there. It seems to persist even if the ESP flash is rewritten, which is good - but we're probably going to have to figure out how to talk to it if we want the chip to get woken up. Hopefully it's as simple as just pulling up a pin at the right time and not a whole complicated serial protocol, although there do appear to be another set of clock and data pins along the edge of the board...

brandond, 2019-01-17:
The pictures don't do this board justice - it's tiny. I've got as far as figuring out that the SB1 'Sleepy Bee' chip is wired up to the external reset pin on the ESP8266. I can confirm that GPIO16 is not wired up to the reset pin, so there's no way for the ESP to wake itself. I can also confirm that the SB1 pulls the external reset pin low after about 110 seconds, so clearly there's some sort of deadline in play. I'm unable to get the unit to wake from motion, so I'm guessing the SB1 is expecting the ESP to do some sort of setup that's not occurring, but I can't find anything else that's wired up between the two chips. I'll keep poking at it, but this is a weird one.

brandond, 2019-01-17:
I managed to kill the ESP core - the SB1 still blinks the LEDs and such but I can't even get the ESP into the bootloader. That's probably the end of my physical exploration of the board - I'll see if I can figure anything out from the flash dump I grabbed, but I'm not optimistic.

Pity, it'd be such a great device if we could just figure out what that other core wanted.

Styx85, 2019-01-19:
Real pity you managed to kill it. Or was it only temporary? I thought I killed mine once, then reflashed the original firmware though getting an error but it worked miraculously.
If it's dead I have to figure out by trial and error which Pin of the ESP pulls up the SB1.
Guess the SP1 is resetting the ESP and awaiting kind of an alive signal to keep the ESP alive.
Have you figured out more connections? And true the board is really small and painted with thick white colour. It's not that easy to follow the connections with ones eyes.

brandond, 2019-01-19:
I was messing around with it while powered up and accidentally shorted the external reset header to 3v3 instead of ground for a couple seconds, after that I can't get anything at all from the ESP - scoping the serial pins shows some output on boot but the timing is all off. Pretty sure I fried something :)

I might try removing the SB1, there are apparently four pads under the corners that I can't hit with a probe. I'm also playing with decompiling the firmware with ScratchABit, but I'm having a hard time figuring out how to load the bits of dumped firmware into the tool and haven't made any meaningful progress.

brandond, 2019-01-20:
Update: I was able to find another set of connections after decapping the SB1.

Using your picture above as a reference, all the via pads above the notch go to the ESP8266, and are labeled intuitively. You can flash firmware over RXD0 and TXD0 by holding GPIO0 to ground as usual. GPIO2 puts out the usual bootloader serial messages at 76923 baud, PLUS a bunch of debug messages from the Tuya firmware. The pads below the notch are for 3v3 and ground, plus another set labeled TDX and RXD. These go to the same pins on the ESP8266 as the RXD0 and TXD0, but on the other side of a pair of resistors. Also on the other side of the resistors are two pins at the top right of the SB1; Multifunction IO P0.4 and P0.5 according to the datasheet. The ESP8266's reset pin is down on the bottom at SB1.5, also connected through a set of resistors. The pins at the far bottom of the board go to the C2 Debug Clock and Data pins on the SB1, which could be interesting if anyone really wanted to get into figuring out what it was doing; it's basically a proprietary bidirectional JTAG protocol.

Interestingly, the board has a 26MHz crystal on it, instead of the more common 40MHz - probably to further reduce battery usage. This might have contributed to my confusion when troubleshooting esphomelib on the board, as there are some things that aren't clocked properly if you compile for the wrong frequency.

Anyways, my board is definitely dead. Knowing what I know now I will probably order another one and see if I can get some logic analyzer traces off the SB1 side of the RXD/TXD pins while running the stock firmware.

Also, the stock firmware seems to run fine on another ESP board I had sitting around. I never actually used the Tuya app to set it up, and I threw away the instructions, so if anyone has any tips on how to use the app to configure it, that might be useful while I wait for the new sensor to come in.

Styx85, 2019-01-26:
> Interestingly, the board has a 26MHz crystal on it, instead of the more common 40MHz - probably to further reduce battery usage. This might have contributed to my confusion when troubleshooting [...]

So what is the setting I have to set here in esphomeyaml? Or does it only matter for external debugging?

Styx85, 2019-01-26:
> Also, the stock firmware seems to run fine on another ESP board I had sitting around. I never actually used the Tuya app to set it up, and I threw away the instructions, so if anyone has any tips on how to use the app to configure it, that might be useful while I wait for the new sensor to come in.

using the tuya app is quite easy. you just have to create an account and then add a device. for the pir you have to add "other devices" then it searches for the pir, otherwise its just telling me its not a zwave (guess what...)

brandond, 2019-01-28:
I'm waiting on my replacement devices to get in; in the mean time I've been playing with the PIR firmware flashed to a development board. Unfortunately I can't get the Tuya app to pick it up. After the normal boot messages on GPIO2, it just sits there sending out a fixed sequence on UTX0 at 9600 baud, at alternating 1 second/2 second intervals.
`0x55 0xAA 0x00 0x01 0x00 0x00 0x00`

I'd love to get it registered with Tuya and see if this changes. At the very least it's something to look for once I get the new devices in.

Styx85, 2019-01-28:
That could be the connection sequence, have a look [here](https://www.heise.de/newsticker/meldung/No-Name-Smart-Home-Security-flaw-allows-easy-firmware-upload-4285085.html).

brandond, 2019-01-28:
Oh, [tuya-convert](https://github.com/ct-Open-Source/tuya-convert) is just what I've been looking for. Might save me some soldering in the future.

I might grab my rPi and see if I can it running on there. Nor sure how the pairing works, but the fact that my AP is dual-band with a 5.8ghz preference and the app says it only supports 2.4gh may be breaking it. Could also be that the firmware is looking for some handshake from the SB1 to go into pairing mode though, since it has no direct connection to the button on the board that's used to put it into that state.

OttoWinter, 2019-02-02:
Same here, awesome work! I'll also close this issue because this is primarily a bug tracker. Discussion can however of course continue here.

closed 2019-02-02 by OttoWinter

Styx85, 2019-02-08:
This is great work. 
But I actually don't know yet what it means to us to get it work with epshomelib. You got any idea?

brandond, 2019-02-09:
I should be able to write a pretty basic UART component to talk to the SB1, but first we need to move the standard serial logging output to GPIO2 to free up GPIO1 and GPIO3. I'll probably put in a PR to allow for setting an option in the logger component to do that, unless @OttoWinter has a different preference.

Once that's done, I'll put together a UART component to talk to the SB1. That would include:
* sending status messages during the boot / OTA process to prevent early sleep
* acking the reset / motion messages to allow the system to go back to sleep as quickly as possible
* sending an update via API / MQTT when a motion event is triggered

Styx85, 2019-02-10:
That would be gorgeous. I'm happily testing it when its done. Unfortunately programming itself is not one of my skills.

mplinuxgeek, 2019-02-11:
I ordered one of these a few weeks ago (arriving today) hoping to "decloud" it with tuya-convert but obviously the SB1 throws a spanner in the works, I'm happy to offer what ever help or testing I can, please hit me up if theres something I can do to help. I've now had intruders in my backyard twice in a month and wanted to deploy at least 4 of these!

After reading this I'd be fairly confident that I could wire the "motion" LED to a GPIO and GPIO16 to RST to bypass the SB1... thoughts?

brandond, 2019-02-11:
The traces on this thing are tiny, I would not plan on rewiring anything. The LEDs are controlled by the SB1, and flash in predetermined patterns - slow blink when configuring in AP mode, fast blink when configuring via SmartConfig, long blink when motion detected followed by a single short blink when the motion has been transmitted to the server, etc. All of this is coordinated by the status messages send back and forth via the UART.

brandond, 2019-02-11:
If you wanted a challenge, you could dump out the flash on the SB1 and reprogram that thing, though. Then you could do whatever you wanted with the LEDs and power state on the ESP.

mplinuxgeek, 2019-02-11:
It looks like the SB1 uses Silicon Labs' Simplicty Studio for development and can be programmed via UART...
https://www.silabs.com/products/development-tools/software/simplicity-studio

When you said you were trying to decompile the firmware were you referring to the firmware on the ESP or the SB1? It looks as though you could possibly reprogram the SB1 over UART via the ESP if the bootloader is still in tact... excerpt from the datasheet:
_1.10 Bootloader
All devices come pre-programmed with a UART bootloader. This bootloader resides in flash and can be erased if it is not needed._

brandond, 2019-02-11:
I have only dumped the flash off the ESP. The SB1 is pretty well documented and does have exposed pads for the Silicon Labs C2 Debug interface, but I  happy to let that chip do it's thing as long as we know how to talk to it.

brandond, 2019-02-11:
I have the UART remap stuff waiting in #427, once that's set I'll start on the SB1 UART stuff and see if I can't get at least a basic handshake working this week.

mplinuxgeek, 2019-02-12:
Love your work, unfortunately I'm not much of a software engineer!
But after reading some of the comments on other posts where people say the ESP still goes to sleep with alternate firmware makes me think it's not actually going to sleep but is being turned off by a transistor or mosfet... so my guess is that the SB1 triggers the "deep sleep" simply by interupting power to the ESP if it doesn't recieve the handshake messages.

brandond, 2019-02-12:
@mplinuxgeek I know they're long but if you're interested, read through the comments above. I already traced all that out and even have some logic analyzer captures if you're curious - the SB1 is pulling the reset pin down, and then releasing it again when there's a motion event... but it only wakes on motion events if the ESP has successfully completed a handshake sequence. If that doesn't happen it assumes the device isn't configured yet (or is misconfigured) and stays asleep until you hold down the config button again.

mplinuxgeek, 2019-02-12:
Oh, I must have missed that.

brandond, 2019-02-16:
OK, so I have a custom UART component that interacts with the SB1 enough to make it happy:
https://github.com/brandond/esphome-tuya_pir/blob/master/sb1_uart.h

There's also a sample esphome YAML file in there if you want to play with it.

Next step is wiring up the motion event to a binary output, and tweaking the MQTT config so that it shows as triggered when it goes off and then clears instead of going unavailable when it gets powered down.

mplinuxgeek, 2019-02-16:
@brandond how do I use your component?

When I attempt to build as is the validation barfs with:
`ERROR Unable to find component esphome`

Changing esphome to esphomeyaml I then get:
`[hardware_uart] is an invalid option for [logger]`

Is there a dev branch of esphomelib I should be using to build?

Disclaimer I started with esphomelib 3 days ago...

OttoWinter, 2019-02-16:
@brandond Amazing work!!! From a brief look the code looks really clean too. Will take a closer look later.

@mplinuxgeek Yes you need the dev branch for that (see FAQ). ESPHome is currently going through a rename so if you use the project through docker or hass.io it's bit different now (ask if you use one of these two)

mplinuxgeek, 2019-02-16:
@OttoWinter will check out the FAQ, thanks, no I'm not using docker or hass.io, I have esphomelib installed directly on my Fedora desktop machine. Cheers.

OttoWinter, 2019-02-16:
A few comments:

https://github.com/brandond/esphome-tuya_pir/blob/4589ea0487abee007895f45f504dcc1fb5cb0fea/sb1_uart.h#L234

This should be refactored into a more general method like `network_client_is_connected` (probably there are better names; method should be in `util.h`). That's because I would like to support setups without MQTT too (native API). 

https://github.com/brandond/esphome-tuya_pir/blob/4589ea0487abee007895f45f504dcc1fb5cb0fea/sb1_uart.h#L193

Is this logic correct? If we have STA shouldn't the state be `STA`? Also I think we should check for STA first, then AP (in case user has STA and AP configured, with AP as fallback if STA fails to connect).

https://github.com/brandond/esphome-tuya_pir/blob/4589ea0487abee007895f45f504dcc1fb5cb0fea/sb1_uart.h#L251

Yes the component should inherit from `binary_sensor::BinarySensor` and call `publish_state`. We could also just call publish on immediately followed by off. Then a binary sensor delayed_off filter can take care of exposing ON for x seconds for frontend.

https://github.com/brandond/esphome-tuya_pir/blob/4589ea0487abee007895f45f504dcc1fb5cb0fea/sb1_uart.h#L250

Is it possible that events from `SB1_STATE_RUNNING` are received while waiting for `MOTION_ACK_DELAY`?

brandond, 2019-02-17:
> This should be refactored into a more general method like `network_client_is_connected` (probably there are better names; method should be in `util.h`). That's because I would like to support setups without MQTT too (native API).

I'll put together a PR for that. It looks like APIServer doesn't currently have an interface to expose whether or not any clients are currently connected; I'll have to add that as well.
 
> Is this logic correct? If we have STA shouldn't the state be `STA`? Also I think we should check for STA first, then AP (in case user has STA and AP configured, with AP as fallback if STA fails to connect).

The AP and STA modes are actually only used while provisioning the device through the app - once you've set the SSID and password via the app and it has successfully registered with Tuya it no longer boots into either of these modes. The only thing that's particularly interesting about them is that it flashes the blue LEDs (fast for AP, slow for STA) as long as it remains in either of these states. The only other time it blinks the LEDs is one long blink when it detects motion, and then one short blink when the ESP acks the motion event, right as it's putting the ESP to sleep.

A better use of these states would probably be to use them to keep the device awake for OTA. Since the device only comes online briefly when motion is detected, catching it for OTA is going to be tricky. We can use the reset button to trigger a safe-mode reboot, and then blink the LEDs while waiting for OTA.

> Yes the component should inherit from `binary_sensor::BinarySensor` and call `publish_state`. We could also just call publish on immediately followed by off. Then a binary sensor delayed_off filter can take care of exposing ON for x seconds for frontend.

Is there any way to tell when the API or MQTT state has been successfully published? It's easy enough to wait for a fixed amount of time, but I'd prefer to go back to sleep as soon as the message is successfully sent off, rather than assuming it will make it out in some arbitrary amount of time.

> https://github.com/brandond/esphome-tuya_pir/blob/4589ea0487abee007895f45f504dcc1fb5cb0fea/sb1_uart.h#L250

> Is it possible that events from `SB1_STATE_RUNNING` are received while waiting for `MOTION_ACK_DELAY`? 

Motion events can't be re-triggered while the ESP is awake, but the reset could theoretically happen at any time. I might put in some special handling for the reset event that will trigger an immediate safe reboot regardless of what else it was looking for.

brandond, 2019-02-17:
As I add the binary sensor component, I'm trying to figure out how to best handle the fact that this device is designed to be online for as short as possible. Most of the time it's not going to be connected to MQTT, or available to be connected to by the native API.

I can probably handle this properly with MQTT by setting up the last will messages so that the motion sensor shows as triggered when it sends a motion event, and then clear when it goes back to sleep and disconnects. However, I'm not sure how this will work with the native API. I've basically got to wait around for the server to connect to it before I can send the message and then go back to sleep, which results in more time on battery. Also, is there any way to tell the native API to publish a state other than unavailable when the client goes offline, as I can do with a MQTT last will message?

mplinuxgeek, 2019-02-17:
> Is there any way to tell when the API or MQTT state has been successfully published?

What about subscribing to the topic it publishes and waiting for an MQTT message?

brandond, 2019-02-17:
@mplinuxgeek the client should get a puback message. Not sure if I can wait for that somehow, or just wait for the queue to go down to 0. I need to look at the code; I was hoping Otto would have a recommendation on a method that might work for both MQTT and API.

brandond, 2019-02-19:
OK, I have totally rewritten the custom component after realizing that I had been going about a bunch of things the wrong way entirely. For starters, the protocol is a super simple type-length-value with a modulo-265 checksum byte at the end. Handling that properly fixes a bunch of stuff. I also added support for OTA mode, where holding down the button inside the battery component puts it into config mode, where the LEDs will blink for about 2 minutes while it waits for an OTA update to be sent.

I still have a few things left to do(battery level, etc), but as of today it should give you an actual functional motion sensor in Home Assistant.

mplinuxgeek, 2019-02-20:
I think the SB1 is pulling the reset line low before state OFF is sent:

This event never sent OFF
```
[19:21:17][D][sb1:353]: Motion event:
[19:21:17][D][sb1:355]: 00: 0x65
[19:21:17][D][sb1:355]: 01: 0x01
[19:21:17][D][sb1:355]: 02: 0x00
[19:21:17][D][sb1:355]: 03: 0x01
[19:21:17][D][sb1:355]: 04: 0x01
[19:21:17][D][binary_sensor:037]: 'PIR00 Motion Sensor': Sending state ON
[19:21:17][D][sb1:175]: state: 12 -> 13 after 38 ms
```
However this one did:
```
[19:22:17][D][sb1:353]: Motion event:
[19:22:17][D][sb1:355]: 00: 0x65
[19:22:17][D][sb1:355]: 01: 0x01
[19:22:17][D][sb1:355]: 02: 0x00
[19:22:17][D][sb1:355]: 03: 0x01
[19:22:17][D][sb1:355]: 04: 0x00
[19:22:17][D][binary_sensor:037]: 'PIR00 Motion Sensor': Sending state ON
[19:22:18][D][sb1:175]: state: 12 -> 13 after 38 ms
[19:22:18][D][binary_sensor:037]: 'PIR00 Motion Sensor': Sending state OFF
```

OttoWinter, 2019-02-20:
> As I add the binary sensor component, I'm trying to figure out how to best handle the fact that this device is designed to be online for as short as possible. Most of the time it's not going to be connected to MQTT, or available to be connected to by the native API.

Yeah so native API won't work for now (because ESP is server, and HA only checks every 60s for reconnect) - a native API server that runs on HA will be supported at some point too, but not yet.

For MQTT, the deep sleep component does automatically disable availability for HA - maybe we could generalize that so that either deep sleep or SB1 disables availability reporting

> Is there any way to tell when the API or MQTT state has been successfully published? It's easy enough to wait for a fixed amount of time, but I'd prefer to go back to sleep as soon as the message is successfully sent off, rather than assuming it will make it out in some arbitrary amount of time.

So no, there's not an API for that (and it would be really hard to add). However, the state will be re-sent until it is published - usually it works on the first try but it doesn't take much longer either. Waiting a fixed amount of time should be fine - the MQTT/API client will make sure the message is delivered on a best-effort basis

> I can probably handle this properly with MQTT by setting up the last will messages so that the motion sensor shows as triggered when it sends a motion event, and then clear when it goes back to sleep and disconnects.

We could also just tell users to use the status binary sensor for that. It will pretty much do the same thing - and I'd rather not have the SB1 component mess with MQTT directly (separation of components).

For new code:

https://github.com/brandond/esphome-tuya_pir/blob/master/sb1_uart.h#L267

This should be called from `setup()`

brandond, 2019-02-20:
@mplinuxgeek Pull an update - after playing with it a bit more I figured out that the last byte in those events indicates motion or clear. I also updated the docs on general setup, although if you're far enough along to be getting log output you've clearly got it pretty well figured out ;)
https://github.com/brandond/esphome-tuya_pir/blob/master/README.md

@OttoWinter I commented on the reason for this here: https://github.com/brandond/esphome-tuya_pir/blob/master/sb1_uart.h#L204 - would you accept a PR to run the shutdown hooks in reverse order (lower level hooks register first, get called last), or maybe allow components to indicate a priority as they do for setup and loop? My hook needs to be the LAST one called, as it sends a message that causes the SB1 to pull RST low as soon as the last character is received.

splaspood, 2019-02-20:
Followed your README instructions and got your custom component flashed to my 'Wasserstein' PIR which appears to be the same board as this Coolcam PIR.    After flashing I'm getting MQTT events for the battery voltage and the wifi signal every 60s or so, but no motion events and the device appears to never shut off.

After enabling DEBUG logging I see this, repeated over and over...  (This is with esphome 1.12.0-dev)

Feb 20 13:10:43 home/garage/pir/debug [D][sb1:175]: state: 1 -> 0 after 1006 ms
Feb 20 13:10:43 home/garage/pir/debug [D][sb1:175]: state: 0 -> 1 after 16 ms

Any thoughts?

Awesome work btw!   I just bought this PIR with the hopes of getting it working with esphome and it was great to find your work. :)

brandond, 2019-02-20:
@splaspood that's the initial handshake retry loop. If the ESP isn't getting a response to that message and isn't getting turned off after a couple minutes, it sounds like the SB1 isn't operational. Usually the fix for that is to reset it using the process described in the second paragraph of the [OTA Updates](https://github.com/brandond/esphome-tuya_pir/blob/master/README.md#ota-updates) section of my readme.

If that still doesn't fix it, can you post your YAML and collect some logs from level VERY_VERBOSE? It's might also be interesting to share a picture of exposed side of the PCB to see if it's the same as what I've got.

FWIW I just ordered one off Amazon; even though I've got a few Neo Coolcams sitting around, 11 bucks is a really good price for these if they're the same thing.

splaspood, 2019-02-20:
I did follow the procedure from the README a few times, with seemingly no effect.    I'll get some log output in a little bit, but here's my YAML:

`esphome:
  name: garage_motion
  platform: ESP8266
  board: esp01_1m
  arduino_version: 2.4.2
  esphome_core_version: dev
  board_flash_mode: dout
  includes:
    - sb1_uart.h

mqtt:
  broker: 192.168.86.15
  username: garage_motion
  password: !secret mqtt_password
  topic_prefix: home/garage/pir
  will_message:

wifi:
  ssid: '123Chadwick'
  password: !secret wifi_password
  fast_connect: true
  # hostname: garage_motion
  domain: .lan

logger:
  level: DEBUG
  id: logger
  hardware_uart: UART1

ota:
  id: ota
  password: !secret ota_password

uart:
  - tx_pin: 1
    rx_pin: 3
    baud_rate: 9600
    id: uart0

sensor:
  - platform: wifi_signal
    id: wifi_signal
    name: "Garage Motion WiFi Signal"
    expire_after:
    filters: []
  - platform: adc
    id: voltage
    name: "Garage Motion Battery"
    expire_after:
    pin: VCC
    filters: []

binary_sensor:
  - platform: template
    id: motion
    name: "Garage Motion"
    filters: []
    device_class: motion
    lambda: "return {};"

custom_component:
  - id: sb1_uart
    lambda: |-
      auto component = new SB1UARTComponent(id(uart0), id(motion));
      return {component};
`

And here are photos of both sides of the board (taken before I found your ongoing work):

[image]
[image]

splaspood, 2019-02-20:
Actually, something just occurred to me.    When I was first playing around with this device I discovered that it had the same mac address as a wemos d1 mini I had on my network.     I hacked up some esphome code to set the MAC to something else on the PIR, but it seems as if the MAC got reset back to the original, conflicting one.    I suspect that might be causing my problem.   

I also ordered this from amazon, and $11 seemed like a *GREAT* price..    That pricing might explain the MAC situation ;)

brandond, 2019-02-20:
Yeah, that's the same thing alright, down to the version and date on the silk screen. The SB1 usually shuts the ESP down after about 110 seconds if it doesn't get the responses it's looking for - are you seeing yours stay up longer than that?

The duplicate MAC address is pretty concerning. I haven't run into that yet, but given the huge number of ESP8266s out there (I saw a press release announcing they've shipped 100 million of them) I guess it would make sense that you might run into duplicates at some point. Might have to add some code to esphome to allow setting the MAC address!?

Styx85, 2019-02-20:
@brandond I have to admit this is really amazing work. I just had time to test it and it's working flawlessly over MQTT with latest dev esphome version.

brandond, 2019-02-20:
I see that the Amazon store page for this thing references using the 'Wstein Mobile App', while the Coolcam branded ones just use the Tuya Home app. I wonder if this is just a rebranding, or if they're actually running some different firmware.

splaspood, 2019-02-20:
I attempted to use tuya-convert and was unable to get it to work (I have successfully used it with other Tuya devices).    However, I was able to flash it directly through the pins.

I suspect that it's 100% Tuya, just branded with possibly different servers on the back end. (which is why tuya-convert didn't work, I guess)

splaspood, 2019-02-20:
> Yeah, that's the same thing alright, down to the version and date on the silk screen. The SB1 usually shuts the ESP down after about 110 seconds if it doesn't get the responses it's looking for - are you seeing yours stay up longer than that?

It seems to never shut off, but until I get the MAC situation squared away it probably makes sense to ignore any of my results.   I'll get that squared later tonight and re-test.

> The duplicate MAC address is pretty concerning. I haven't run into that yet, but given the huge number of ESP8266s out there (I saw a press release announcing they've shipped 100 million of them) I [...]

I find it highly suspect that I would just happen to have two devices with the same MAC due to some exhaustion of address space.    I figure instead that someone didn't set the macs properly in the production line so there are now X number of devices out there all with the same 'default' mac.

splaspood, 2019-02-20:
Issue was 100% the MAC conflict.   After much trial and error getting it re-flashed via TTL serial (turns out I didn't need to push the button after powering on with GPIO0 tied to ground.   Once I skipped that, I was able to flash.    

The device is now shutting off properly after motion is detected.   I still need to do some better testing to see if motion is detected accurately, but it's looking much better than before.

In case anyone encounters a similar MAC issue...  I just ended up sticking the following two lines into the custom_component lambda (probably better to put this into the startup code you can define in esphome):
``` 
uint8_t mac[6] {0xb8, 0xd7, 0x63, 0x00, 0xfe, 0xef};
wifi_set_macaddr(STATION_IF, mac);
```

After booting with that once, the code can be removed.   Obviously you should pick a different mac address than that example.

splaspood, 2019-02-20:
Looks like it detects motion, turns on and sends the ON state to MQTT, then powers off, waits 60 seconds and powers back on and sends the OFF.   That sound right?  

If that's the expected behavior, it seems to be working great for me now :)

brandond, 2019-02-20:
@splaspood yeah that's what it does. I'm not sure what the exact delay is, but it seems to debounce the events a bit - it won't power the ESP up to send an OFF message until it's gone a while without seeing motion. 

If we find too many more devices with duplicate MACs, it might be worth adding an option to the wifi component to allow setting it manually.

brandond, 2019-02-21:
Since this seems to be working for a few folks other than me, I'm going to try to prepare it for merging into core. @OttoWinter I was planning on setting this up as a binary_sensor platform; does that make sense? I'll also probably include a new 'device' page for the docs, as this thing needs some special handling in terms of how you flash it and get it set up for OTA.

There's also that issue with the shutdown hook order vs setup order; what do you think is the best way to resolve that?

OttoWinter, 2019-02-22:
So, I haven't read everything here (quite a lot of stuff to do in real life right now), but:

Re duplicate mac address: Even with so many devices sold, the chance of having a duplicate mac address are =0 for all intents and purposes. Did you maybe flash one firmware to both ESPs or copy one from an other ESP? (The MAC address is stored in rfcalib section I think, which is accessible from flash and can be overridden by a flash)

@brandond Hmm, I mean it is a binary sensor, but it also feels a lot like a real top-level component to me. Is there other things that this device could add at some point in the future (status LED, some sensor, etc?). I think you talked about other things being hidden in these messages like battery level. If so, this should be a component.

> There's also that issue with the shutdown hook order vs setup order; what do you think is the best way to resolve that?

Ok so rather than reversing it, I think adding a priority (like setup priority) to these hooks is better and more future-proof. Implementation would be quite simple, just create a small holder struct that stores them and sort them right before they're executed with `std::stable_sort` (see application.cpp for example)

brandond, 2019-02-22:
@OttoWinter it outputs a binary reading, but does a lot more than just talk to a device to get that reading. I'll put in a PR that sets it up that way and you can suggest changes if it doesn't feel right. The only other thing I've gotten out of the message payloads is some sort of power loss / battery replacement indicator. The battery voltage just comes from reading the ADC, and the LEDs are controlled by the coprocessor based on the messages we send it - which also determine whether or not we get sent motion events, so they're not really directly controllable.

For the shutdown order, right now there's just a list of callback functions; it doesn't appear to store the components they're linked to. I suppose I could change it to accept a priority in addition to a callback function pointer. Alternately, I could rework it so that all components can override shutdown() and get_shutdown_priority() to hook shutdown and set their shutdown priority. That would give it better parity with setup(). Do you have a preference?

[77 more replies]