esphome/esphome #307: clean-mqtt does not clean all topics

brandond, 2019-01-08:
**Operating environment (Hass.io/Docker/pip/etc.):**
pip + virtualenv

**ESP (ESP32/ESP8266/Board/Sonoff):**
esp01_1m

**Affected component:**
mqtt

**Description of problem:**
Running a clean-mqtt only cleans the discovery prefix, it does not clean any of the actual component state topics.

Additionally, it would be nice if esphomelib would subscribe it its own discovery topics and re-publish configuration when it detects that they have been cleaned. I guess it's easy enough to simply push a firmware update to the device to force it to reboot and re-publish the discovery topics after I've cleaned things out, but it's an extra step.

**Additional information:**
```
(python27) brandond@seago:~/repos/hass-esphome/nodes$ esphomeyaml bulb00.yaml clean-mqtt
INFO Reading configuration...
INFO Clearing messages from 'homeassistant/+/bulb00/#'
INFO Please close this window when no more messages appear and the MQTT topic has been cleared of retained messages.
Clearing topic homeassistant/sensor/bulb00/smart_bulb_00_wifi_signal/config
Clearing topic homeassistant/switch/bulb00/smart_bulb_gpio5/config
Clearing topic homeassistant/switch/bulb00/smart_bulb_gpio12/config
Clearing topic homeassistant/switch/bulb00/smart_bulb_gpio14/config
Clearing topic homeassistant/switch/bulb00/smart_bulb_gpio5_red/config
Clearing topic homeassistant/switch/bulb00/smart_bulb_gpio12_blue/config
Clearing topic homeassistant/switch/bulb00/smart_bulb_gpio14_green/config
Clearing topic homeassistant/switch/bulb00/smart_bulb_gpio13/config
Clearing topic homeassistant/switch/bulb00/smart_bulb_gpio15/config
Clearing topic homeassistant/light/bulb00/smart_bulb_00_light/config

(python27) brandond@seago:~/repos/hass-esphome/nodes$ esphomeyaml garage.yaml clean-mqtt --topic 'bulb00/#'
INFO Reading configuration...
INFO Clearing messages from 'bulb00/#'
INFO Please close this window when no more messages appear and the MQTT topic has been cleared of retained messages.
Clearing topic bulb00/status
Clearing topic bulb00/sensor/smart_bulb_00_wifi_signal/state
Clearing topic bulb00/switch/smart_bulb_gpio5/state
Clearing topic bulb00/switch/smart_bulb_gpio12/state
Clearing topic bulb00/switch/smart_bulb_gpio14/state
Clearing topic bulb00/switch/smart_bulb_gpio5_red/state
Clearing topic bulb00/switch/smart_bulb_gpio12_blue/state
Clearing topic bulb00/switch/smart_bulb_gpio14_green/state
Clearing topic bulb00/switch/smart_bulb_gpio13/state
Clearing topic bulb00/switch/smart_bulb_gpio15/state
Clearing topic bulb00/light/smart_bulb_00_light/state
```

OttoWinter, 2019-01-08:
> Running a clean-mqtt only cleans the discovery prefix, it does not clean any of the actual component state topics.

I guess that could also be added.

> Additionally, it would be nice if esphomelib would subscribe it its own discovery topics and re-publish configuration when it detects that they have been cleaned.

I like to keep the MQTT client implementation as simple as possible. Discovery payloads can be huge and so if we would subscribe to them too the MQTT library will allocate a whole bunch more memory for the receiving part (it already does for sending, but that's a separate buffer).

Anyway, I'm betting on the new ["native API"](https://beta.esphomelib.com/esphomeyaml/components/api.html) with esphomelib right now - it's far better for communicating with HA<->esphomelib (for a variety of reasons). Eventually I also want to remove the MQTT discovery aspect; or at least disable it by default.

> I guess it's easy enough to simply push a firmware update to the device to force it to reboot and re-publish the discovery topics after I've cleaned things out, but it's an extra step.

https://esphomelib.com/esphomeyaml/components/switch/restart.html

brandond, 2019-01-08:
Will the native api handle this any better? The place I usually run into this is when renaming devices, adding temporary components when debugging, etc. It leaves a bunch of cruft around in mqtt topics that I then have to manually clean out once things settle down.

closed 2019-01-17 by OttoWinter