esphome/esphome #314: No USB port in dashboard web interface !!

vincegre, 2019-01-10:
Operating environment (Hass.io/Docker/pip/etc.):
Hassio 141
HassOS 1.13
HA 0.85.0
esphomeyaml component in hassio: 1.10.0

ESP (ESP32/ESP8266/Board/Sonoff): D1 mini with previous version of esphomeyaml firmware in it

Affected component: USB upload feature unavalaible

Description of problem: Unable to do USB upload of firmware as USB port is not offered as choice in web interface. I connected my D1 mini (esp module alone) by USB to Raspberry running HA). I then restarted the esphomeyaml component in hassio but I still have only OTA choice in web interface :(

```
Logs on module startup but I don't see anything interesting in it:
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing... 
-----------------------------------------------------------
 Hass.io Add-on: esphomeyaml v1.10.0
 esphomeyaml Hass.io add-on for intelligently managing all your ESP8266/ESP32 devices.
 From: esphomeyaml Hass.io Add-Ons
 By: Otto Winter <contact@otto-winter.com>
-----------------------------------------------------------
 armhf / HassOS 1.13 / HA 0.85.0 / SU 141 / stable
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] 02-updates.sh: executing... 
[11:46:32+0100] INFO ----> You are running the latest version of this add-on
[cont-init.d] 02-updates.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing... 
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-esphomeyaml.sh: executing... 
[cont-init.d] 30-esphomeyaml.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[11:46:33+0100] INFO ----> Starting NGINX...
[11:46:33+0100] INFO ----> Starting esphomeyaml dashboard...
INFO Starting dashboard web server on unix socket /var/run/esphomeyaml.sock and configuration dir /config/esphomeyaml...
INFO 304 GET / (0.0.0.0) 74.37ms
INFO 304 GET /static/materialize.min.css?hash=1126843982 (0.0.0.0) 25.37ms
INFO 304 GET /static/esphomeyaml.css?hash=1649691102 (0.0.0.0) 2.75ms
INFO 304 GET /static/materialize-stepper.min.css?hash=-14757235 (0.0.0.0) 2.69ms
INFO 304 GET /static/jquery.min.js?hash=750217762 (0.0.0.0) 3.40ms
INFO 304 GET /static/materialize.min.js?hash=-1933846751 (0.0.0.0) 3.92ms
INFO 304 GET /static/jquery-ui.min.js?hash=564504033 (0.0.0.0) 4.41ms
INFO 304 GET /static/jquery.validate.min.js?hash=-1686981287 (0.0.0.0) 4.12ms
INFO 304 GET /static/materialize-stepper.min.js?hash=-757351638 (0.0.0.0) 2.89ms
INFO 304 GET /static/ace.js?hash=-1353738198 (0.0.0.0) 5.00ms
INFO 304 GET /static/esphomeyaml.js?hash=216580722 (0.0.0.0) 2.77ms
INFO 304 GET /static/mode-yaml.js (0.0.0.0) 3.46ms
INFO 304 GET /static/theme-dreamweaver.js (0.0.0.0) 3.25ms
INFO 304 GET / (0.0.0.0) 14.88ms
INFO 304 GET /static/materialize.min.css?hash=1126843982 (0.0.0.0) 4.63ms
INFO 304 GET /static/materialize-stepper.min.css?hash=-14757235 (0.0.0.0) 4.54ms
INFO 304 GET /static/esphomeyaml.css?hash=1649691102 (0.0.0.0) 2.51ms
INFO 304 GET /static/jquery.min.js?hash=750217762 (0.0.0.0) 2.48ms
INFO 304 GET /static/materialize.min.js?hash=-1933846751 (0.0.0.0) 4.77ms
INFO 304 GET /static/jquery-ui.min.js?hash=564504033 (0.0.0.0) 4.30ms
INFO 304 GET /static/jquery.validate.min.js?hash=-1686981287 (0.0.0.0) 2.63ms
INFO 304 GET /static/materialize-stepper.min.js?hash=-757351638 (0.0.0.0) 2.52ms
INFO 304 GET /static/ace.js?hash=-1353738198 (0.0.0.0) 2.50ms
INFO 304 GET /static/esphomeyaml.js?hash=216580722 (0.0.0.0) 2.40ms
INFO 304 GET /static/theme-dreamweaver.js (0.0.0.0) 7.36ms
INFO 304 GET /static/mode-yaml.js (0.0.0.0) 2.86ms
```

HugoVonHory, 2019-01-11:
I face the same issue. I have three ESPs connected to raspberry (1x NodeMCU, 2x ESP32) for testing purposes. After last update USB ports disappeared from web UI. Tried plugin restart, USB reconnection, homeassistant restart, nothing helped.
(armhf / Resin OS 2.3.0+rev1 / HA 0.85.0 / SU 141 / stable)

vincegre, 2019-01-12:
@HugoVonHory Talked with Otto on discord channel and he is aware of problem that is already fixed and should be pushed with next update ;) So we just have to wait for it !

OttoWinter, 2019-01-12:
For everyone wanting to get USB sticks back, you can already try my patch (it will go live with 1.10.1)

- Stop all esphomeyaml addons
- Remove the esphomeyaml addon repo
- Re-add the `https://github.com/OttoWinter/esphomeyaml` repo
- Re-install addon

SirRedZ, 2019-01-24:
I installed today, repo to fresh Hassio install on VirtualBox,  (Windows 10x64 Pro Machine) using D1 Mini previously flashed with Tasmota 6.4.1. Numerous restarts, and could not get USB interface to flash 1st time. Used EasyESP Flasher and all OTA updates working great after that. I am hesitant to post, as I wonder if my Virtual machine is improperly configured. Willing to test or make changes as required if it helps - this is a test hassio for EspHome Yaml learning  specifically.

vincegre, 2019-01-24:
Confirm it's working fine here now with update of esphome addon in hassio ;)

Great work @OttoWinter

closed 2019-01-24 by vincegre