esphome/esphome #315: Create local mDNS responder

OttoWinter, 2019-01-10:
**Is your feature request related to a problem/use-case? Please describe.**

Hass.io .local resolution doesn't always work (I'm guessing this is due to the OS - specifically in`getaddrinfo` syscall).

**Describe the solution you'd like:**

Esphomeyaml should try to set up a local mDNS responder if the getaddrinfo syscall fails and the user isn't using a static IP.

oscar-b, 2019-01-12:
A bit of tetsing against my esphome nodes, running v1.9.0:

Node `koket_fonster_shelly1.local` doesn't resolve mDNS requests:
```$ dig @224.0.0.251 -p 5353 +short koket_fonster_shelly1.local
;; connection timed out; no servers could be reached```

It does resolve on ping:
```$ ping koket_fonster_shelly1.local
PING koket_fonster_shelly1.local (10.5.5.104): 56 data bytes
64 bytes from 10.5.5.104: icmp_seq=0 ttl=255 time=8.725 ms```

Using `dns-sd`, it doesn't find the nodes:
```$ dns-sd -B
Browsing for _http._tcp
DATE: ---Sat 12 Jan 2019---
13:38:58.107  ...STARTING...
Timestamp     A/R    Flags  if Domain               Service Type         Instance Name
13:38:58.108  Add        3  10 local.               _http._tcp.          Hogwarts
13:38:58.108  Add        2  10 local.               _http._tcp.          HP Color LaserJet Pro```

olskar, 2019-01-12:
@oscar-b Are you using HassOS as host?

oscar-b, 2019-01-12:
@olskar the tests above are from a MacBook. And no I don’t use HassOS for my Home Assistant setup.