esphome/esphome #317: Missing functionality after switching to native API

escoand, 2019-01-10:
**Is your feature request related to a problem/use-case? Please describe.**
I use a thermistor to measure the temperature and go for one hour into deep sleep afterwards. Until 1.10 I used MQTT with empty `will_message` and `mqtt_subscribe` to check if in OTA mode. Now after switching to native API this is not possible anymore. So every time the device is going into deep sleep it and all sensors are immediately reported as unavailable in HA.

```yaml
mqtt:
  broker: 192.168.1.45
  will_message:
text_sensor:
  - platform: mqtt_subscribe
    id: ota_mode
    topic: thermometer/ota_mode
```

**Describe the solution you'd like:**
I would like to have an yaml option to not set all sensors/... as unavailable. This will already happen with `expire_after`. Additionally should there be a way in HA to set OTA mode - maybe directly in the integration page as a switch.
Maybe this is more a ticket for the HA native api component.

**Additional context:**

OttoWinter, 2019-01-11:
If the ESP has deep sleep enabled, Home Assistant will not use the `unavailable` feature in the native API.

But ... deep sleep isn't really possible with native API yet. HA will only connect to the ESP every 60 seconds, so most of the times the ESP is already back to sleep then. The native API component will need to be restructured a bit to support deep sleep - specifically HA/or some other persistent app needs to be the server, not the ESP.

OttoWinter, 2019-02-02:
Closing for https://github.com/esphome/feature-requests/issues/46

closed 2019-02-02 by OttoWinter