esphome/esphome #318: Problem with OTA updates

tomdee, 2019-01-11:
**Operating environment (Hass.io/Docker/pip/etc.):**
pip

**ESP (ESP32/ESP8266/Board/Sonoff):**
ESP32

**Affected component:**
OTA

**Description of problem:**
I've never been able to get OTA working. I was using the last release 1.9.x and have just tried with 1.10 now. I've also switched to static IPs but it's still not working.

Usually, I don't see it get past 2 or 3 %, but this time it got a bit higher
```
22:25:58][D][ota:180]: Starting OTA Update from 192.168.10.142...
[22:25:58][D][ota:336]: OTA in progress: 0.1%
[22:25:59][D][ota:336]: OTA in progress: 5.3%
[22:26:00][D][ota:336]: OTA in progress: 10.6%
[22:26:01][D][ota:336]: OTA in progress: 14.8%
[22:26:02][D][ota:336]: OTA in progress: 20.0%
[22:26:03][D][ota:336]: OTA in progress: 22.6%
[22:26:04][D][ota:336]: OTA in progress: 28.1%
[22:26:05][D][ota:336]: OTA in progress: 33.2%
[22:26:06][W][ota:398]: Error reading data!
[22:26:06][W][ota:369]: Update end failed! Error: No Error
[22:26:06]
[22:26:06][E][WiFiClient.cpp:338] write(): 12
[22:26:06][D][esp32_ble_tracker:363]: Xiaomi MiJia 4C:65:A8:DB:A6:01 Got temperature=18.7°C, humidity=72.0%
```

Any thoughts on what else I can try?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
```

**Traceback (if applicable):**

```

```

**Additional information:**

tomdee, 2019-01-11:
Looks like this could be because I'm using an ESP32 with BLE (as mentioned in https://github.com/OttoWinter/esphomeyaml/issues/191). Is there any workaround?

tomdee, 2019-01-11:
I retried a few time and hit a different error
```
[22:34:09][D][ota:180]: Starting OTA Update from 192.168.10.142...
[22:34:09][D][ota:336]: OTA in progress: 0.1%
[22:34:10][D][ota:336]: OTA in progress: 5.7%
[22:34:20][W][ota:403]: Timeout waiting for data!
[22:34:20][W][ota:369]: Update end failed! Error: No Error
[22:34:20]
[22:34:20][E][WiFiClient.cpp:338] write(): 113
[22:34:20][W][wifi:1212]: Event: Disconnected ssid='iot' bssid=36:B5:C2:AA:32:91 reason=Beacon Timeout
[22:34:20][W][wifi:080]: WiFi Connection lost... Reconnecting...
[22:34:20][W][wifi:386]: Restarting WiFi adapter...
[22:34:20][W][esp32_ble_tracker:082]: Too many BLE events to process. Some devices may not show up.
```

tomdee, 2019-01-11:
And another time, this time I got a crash
```
[22:36:25]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[22:36:25]Core 1 register dump:
[22:36:25]PC      : 0x40173903  PS      : 0x00060630  A0      : 0x801cb618  A1      : 0x3fff26f0  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d livingroom -t idedata
WARNING Decoded 0x40173903: tcp_output at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp_out.c:983
[22:36:27]A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000011  A5      : 0x0009c506  
[22:36:27]A6      : 0x00000000  A7      : 0x000010d4  A8      : 0x8017d79c  A9      : 0x3fff26c0  
[22:36:27]A10     : 0x3fff0028  A11     : 0x3ffc1170  A12     : 0x3fff0088  A13     : 0x00000000  
[22:36:27]A14     : 0x3fff00b8  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
[22:36:27]EXCVADDR: 0x00000038  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[22:36:27]
[22:36:27]Backtrace: 0x40173903:0x3fff26f0 0x401cb615:0x3fff2710 0x4016bb9d:0x3fff2730
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40173903: tcp_output at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/core/tcp_out.c:983
WARNING Decoded 0x401cb615: _tcp_output_api(tcpip_api_call*) at AsyncTCP.cpp
WARNING Decoded 0x4016bb9d: tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/api/tcpip.c:474
[22:36:27]
[22:36:27]Rebooting...
[22:36:27]ets Jun  8 2016 00:22:57
[22:36:27]
[22:36:27]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:36:27]configsip: 0, SPIWP:0xee
[22:36:27]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:36:27]mode:DIO, clock div:2
[22:36:27]load:0x3fff0018,len:4
[22:36:27]load:0x3fff001c,len:952
[22:36:27]load:0x40078000,len:6084
[22:36:27]load:0x40080000,len:7944
[22:36:27]entry 0x40080310
[22:36:27][I][logger:071]: Log initialized
[22:36:27][C][ota:461]: There have been 0 suspected unsuccessful boot attempts.
[22:36:27][I][application:053]: Running through setup()...
[22:36:27][D][esp32_ble_tracker:167]: Starting scan...
```

tomdee, 2019-01-11:
Thanks @OttoWinter for the explanation (and all the great work you do on this awesome project)!

Is see that arduino-esp32 has just had a new release - is it worth me trying that? https://github.com/espressif/arduino-esp32/releases

brandond, 2019-01-11:
It looks like the platformio folks pulled in the updated core earlier today, you can try it if you point at the develop branch:

```yaml
esphomeyaml:
  platform: ESP32
  arduino_version: https://github.com/platformio/platform-espressif32#develop
```

brandond, 2019-01-11:
Looks the AsyncMQTT/AsyncTCP stuff does not compile against 1.0.1 :(

OttoWinter, 2019-01-12:
@brandond Try
```yaml
esphomeyaml:
  platform: ESP32
  arduino_version: dev
```

tomdee, 2019-01-12:
The `idf-update` branch just got merged to master