esphome/esphome #319: Add an option to prevent MQTTTextSensor being generated

pauln, 2019-01-11:
**Is your feature request related to a problem/use-case? Please describe.**

When creating text sensors, the generated code creates an `MQTTTextSensor` for each text sensor.  For nodes which are using the native API (especially those running on ESP8266 modules, given how little RAM they have), it'd be nice to be able to skip this.

**Describe the solution you'd like:**

Ideally, `esphomeyaml` would detect that MQTT is not set up (but the native API is), and thus automatically skip the `MQTTTextSensor` creation - but if that's not feasible (at least in the short term), a more manual approach (such as a `skip_mqtt: True` flag in the `text_sensor` yaml) would certainly suffice.

**Additional context:**

I'm attempting to build an e-paper ambient display, using a 7.5in WaveShare screen and their ESP8266-based driver board.  The small amount of RAM on the ESP8266 regularly results in crashes due to memory allocation failures, so I've been trying to remove anything superfluous - hence this feature request.

I'm currently using template text sensors while I build up the lambda for the display component, but (once [esphomelib#368](https://github.com/OttoWinter/esphomelib/issues/368) is resolved) these will need to be replaced with Home Assistant text sensors.  Another option might therefore be to refactor the Home Assistant text sensor component to not do the MQTT setup (while leaving it in place for other text sensors), as it seems a bit pointless to rebroadcast Home Assistant states over MQTT anyway.

OttoWinter, 2019-01-11:
Have you seen this line? https://github.com/OttoWinter/esphomelib/blob/dev/src/esphomelib/application.cpp#L1219

pauln, 2019-01-11:
I hadn't seen that line - I'm new to `esphomelib` and haven't had a chance to delve into its inner workings yet.  I was looking at the `main.cpp` generated by `esphomeyaml` (where each text sensor gets a line like `text_sensor::MQTTTextSensor *text_sensor_mqtttextsensor = application_maketemplatetextsensor.mqtt;`) - and was drawn into that when I noticed the `AsyncMqttClient` library showing up in the dependencies.

If you're satisfied that the current approach is not resulting in extra memory usage when MQTT is not enabled (bearing in mind how little RAM the ESP8266 has), please feel free to close this issue - though it still might be nice to strip the MQTTTextSensor stuff from the generated code entirely and remove the `AsyncMqttClient` dependency for more efficient builds.

OttoWinter, 2019-01-11:
> where each text sensor gets a line like text_sensor::MQTTTextSensor *text_sensor_mqtttextsensor = application_maketemplatetextsensor.mqtt;

Yep, that's mostly because I'm lazy (and because it's kind of hard for the C++ generator to know if a variable is used or not). So because the compiler also detects that it's an unused variable, it even gets removed in the assembly as well (you can check with objdump).

> and was drawn into that when I noticed the AsyncMqttClient library showing up in the dependencies.

Long story, currently platformio will always load in that library even though it's not explicitly added in `lib_deps` - Spent a ton of time weeks ago disabling the platformio library dependency finder, but ran into issues which unfortunately make ESPHome load all libraries.

> If you're satisfied that the current approach is not resulting in extra memory usage when MQTT is not enabled

Yes, AFAIK that should not affect runtime memory. It does affect flash size a bit (compiler isn't *quite* smart enough to notice the MQTT client is never instantiated), but that will take lots of refactoring to get rid of + I haven't seen lots of flash size issues recently.

closed 2019-01-11 by OttoWinter