esphome/esphome #321: Cant set dns1 address. "some but not all values in the same group of inclusion 'dns'."

pixiandreas, 2019-01-11:
**Operating environment (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266/Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml

   manual_ip:
     static_ip: 192.168.50.101
     gateway: 192.168.50.1
     subnet: 255.255.255.0
     dns1: 192.168.50.1

```

**Traceback (if applicable):**

```

```

**Additional information:**

The config check complain about the dns1
Without dns1 it works as it should.

OttoWinter, 2019-01-11:
The error means you need to specify both `dns1` and `dns2`

nacbooth, 2019-01-19:
I had the same problem and discovered that DNS2 had to be set as well; I simply use the same IP address for both (as I only have one [practical] DNS server on my network.

brandond, 2019-01-19:
This should be fixed in #357 which hasn't yet been merged to dev.

closed 2019-01-27 by OttoWinter