esphome/esphome #322: Force update option for binary sensor

HugoVonHory, 2019-01-11:
**Is your feature request related to a problem/use-case? Please describe.**
Binary sensor value is logged to homeassistant DB only on value change event via MQTT. Displaying graphs from values doesn't show correct state, if last state change isn't in displayed timeframe (no data in result). Hartbeat filter option in config publishes repeatedly state to MQTT correctly, force_update option should enable loging to DB as well.

**Describe the solution you'd like:**
add configuration option "force_update"
https://www.home-assistant.io/components/binary_sensor.mqtt/#force_update

**Additional context:**

OttoWinter, 2019-01-11:
I never understood why people would use the force_update option. It just increases the DB size considerably and floods the HA state machine - and most graph frontends (like grafana) have options to not have to rely on force update.

HugoVonHory, 2019-01-11:
I understand that it is standard behaviour and it saves data and processing. Can anyone advise, how to solve it? If there are no data in time range, grafana displays nothing (or graph only on part of time range, where are values). Still can't find any option, or procedure for correct display, force_update is only thing I currently see to make diplay consistent.

casperklein, 2019-01-26:
@OttoWinter When not every event is logged to a database (because the value haven't changed), how do you distinguish later, wether for example the temperature was the same or the device just failed and stopped sending sensor data?

brandond, 2019-01-26:
If you're querying a DB, you could look at the timestamp (age) of the most recent reading to decide whether or not to use it. If you're using HA automations, you could check the device status to see if it's online or unavailable. Just a couple suggestions; I'm sure you can come up with more solutions that are specific to your use case.

OttoWinter, 2019-01-27:
>  If there are no data in time range, grafana displays nothing

There's a fill option just for that

> wether for example the temperature was the same or the device just failed

The state of the sensor is `unavailable` when HA can't connect to it.

It's not that much work to add `force_update` to ESPHome, but I'm against it for several reasons (mostly because it's really bad for HA performance). And for all the reasons I've heard so far there was a simple solution.

If someone can convince me that force_update is required and there's no simple other way, I can add it.