esphome/esphome #323: Can't flash and compile v1.10.0

flemssound, 2019-01-11:
I have Updated the Homeassitant Plugin and want to flash new wemos d1 mini devices with just the basic settings. But after a while the process will be stoped.

Running 
HassOS 1.13 on rpi 3 32bit

`[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x75623670>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=1)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x75623b30>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=2)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x75630070>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=3)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x75630590>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=4)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x75630ab0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=5)
Error: [API] Could not connect to PlatformIO API Service. Please try later.
`

closed 2019-01-11 by flemssound

poldim, 2019-04-04:
What did you do to solve this @flemssound ? 

I'm getting a similar error:
```[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP+Async+WebServer%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f2cbc3e5f10>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=1)```

Naesstrom, 2019-04-04:
Oh great, it's not just me then!
```log
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x75f33d10>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=1)
```
It tries a couple of times and then gives a error!

poldim, 2019-04-04:
It turns out my pihole docker was locked up and I wasn't getting DNS resolution.  Make sure you have a valid DNS server and can resolve URLs