esphome/esphome #324: Shows as offline, but is online

olskar, 2019-01-11:
[image]

Updated a nodemcu to latest version. Now in dash board it shows as offline but i can still se message coming in in debug

dreed47, 2019-01-11:
I also get this using the latest beta

olskar, 2019-01-12:
Could this be related to mdns which I am having troubles with. It does not
find bastu.local, but if I change the settings to a static IP it says
online.

OttoWinter, 2019-01-12:
Internally, just `ping <HOSTNAME>.local` is called (or `ping <STATIC_IP>` if configured). So this is related to https://github.com/OttoWinter/esphomeyaml/issues/315

olskar, 2019-01-12:
Makes sense, closing this @OttoWinter and hopes for a solution in #315

closed 2019-01-12 by olskar