esphome/esphome #326: Emulate WeMo device

highground88, 2019-01-12:
Fantastic work on the current state of esphomeyaml!!! Have just started playing with it this week and it's a dream in most respects. But.. onto the topic at hand: Missing WeMo support

Currently I'm using Tasmota and a mixture of others for my ESP's, however I would love to fully move over, but unless I'm mistaken the ability to emulate a WeMo device (or similar), to allow discovery in Alexa and Google Home, is missing.

Adding this ability into the system would provide voice control for switches etc, and potential simple binary sensors to communicate and work within these AI systems (routines etc..)

I

OttoWinter, 2019-01-12:
Currently, the goal of this project is not to do *everything* like Wemo etc. For example the web interface looks horrible, so the way ESPHome is _meant_ to be used is together with a Home Automation solution (like Home Assistant or OpenHAB).

_These_ hubs should be in charge of relaying stuff to Alexa, Google Home etc (and it's very easy to set it up too for them). Why? Running yet another service on these tiny ESPs is just asking for trouble (talking mostly about memory usage here)

highground88, 2019-01-12:
Sure I understand, thank you for the feedback.  I had hoped it was a simple addition that it would be picked up by Alexa in addition to HA.  At the moment I'm still learning, but the Tasmota style WeMo emulation has been a favourite with the wife approval factor... I will need to spend more time obviously (and apparently a subscription now) to implement the remaining devices into Alexa.

highground88, 2019-02-11:
Hi Otto,

Just on the off chance that this might be something you hadn't seen or considered. I've thought about this some more and perhaps, memory dependant, this could be an optional library to include?

[FauxWemo](https://bitbucket.org/xoseperez/fauxmoesp)

This is not to use the device in the Wemo interface - it's simply to allow Alexa etc to recognise the ESP as a switch automatically, which saves extra configuration (e.g. no mqtt required).

Please feel free to comment or close as you wish - appreciate your hard work.

OttoWinter, 2019-02-11:
Yes I am aware of that library but I still do not think it's good to include. It would also be difficult because the web_server component uses the same port too (so using that library would probably not be possible)

> which saves extra configuration (e.g. no mqtt required).

Have you seen the new native API for esphome? With that it's just a couple of clicks to get the esp connected to HA.

highground88, 2019-02-11:
I've just started using the native API and it is indeed excellent, and all the sensors you've included is fantastic.

But for me at the moment I'm still using Tasmota on roughly half of my devices, purely because I can voice control them with Alexa due to the WeMo emulation, without having to pay a monthly cloud subscription for the HA component for just a few devices. When money is tight this option has a much higher Wife Approval Factor too.

If things change down the line I really do think something like this would be a fantastic option to include, but it does sound like there are a few hurdles.

OttoWinter, 2019-02-12:
Well if ESPs can emulate wemo I think HA can do the same. Haven't checked, but I think there's a high chance somebody already implemented wemo emulation for HA (either integrated or as a custom component)