esphome/esphome #328: ESP offline after OTA update

romanpeters, 2019-01-12:
**Operating environment (Hass.io/Docker/pip/etc.):** Hass.io, ESPHome add-on v1.10.0
 

**ESP (ESP32/ESP8266/Board/Sonoff):** Sonoff

**Description of problem:**
I decided to replace MQTT with the new Native API for Home Assistant, on a Sonoff I've had up and running for a while. I removed the MQTT configuration and added `api:` (without a password), through the Hass.io add-on. Validation went through and the Sonoff was online.
After compiling and uploading, I got a warning that it used the Legacy OTA method as a fallback (I hadn't updated this Sonoff in a while, so expected behaviour I guess).
After that, the logs said the upload was a success, however it would repeatedly warn me that it couldn't connect to the API.
Closing the upload screen, showed the Sonoff now being offline in the Hass.io add-on.
I tried restarting the Sonoff, by unplugging it from the socket and back in, but to no avail.
Once plugged in, the green LED on the Sonoff turns on for a fraction of a second, then turns off.
I tried to connect to the Sonoff through Home Assistant, but got a connection error.
I tried to ping the Sonoff, but got request timeout and host is down.

It looks like the Sonoff has become completely unresponsive.

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: sonoff2
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout

wifi:
  ssid: 'redacted'
  password: 'redacted'

api:

logger:

ota:
  password: 'redacted'

switch:
  - platform: gpio
    name: "Heater"
    pin: GPIO12

output:
  - platform: esp8266_pwm
    id: basic_green_led
    pin:
      number: GPIO13
      inverted: True

light:
  - platform: monochromatic
    name: "Heater LED"
    output: basic_green_led

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Heater Button"
  - platform: status
    name: "Sonoff Status"
```

**Traceback (if applicable):**
Running "Show Logs" in the Hass.io add-on repeats this warning:

```
INFO Connecting to sonoff2.local:6053 (10.10.10.196)
WARNING Couldn't connect to API (Error connecting to 10.10.10.196: [Errno 113] No route to host). Trying to reconnect in 1 seconds
```

**Additional information:**

alvarea, 2019-01-13:
Hello,

First of all, congratulations for this excellent work! 

I've experience almost the same issue after upgrading to 1.10 version. After this failed firmware updated, the sonoff basic is offline on the dashboard and it has also disappeared from the network (no ping available).  During the 1st. upload attempt I saw an error message similar to:  Failed to upload firmware using OTA v2. 

In my case the most extrange thing is that it has just happened with 3 sonoff basic, but there is no issue with a sonoff pow and a sonoff S20, which I've been able to change to api connection. 

I guess the only way for this 3 sonoff basic is connect them by USB and try to upload the firmware again.

mssmison, 2019-01-18:
I've been experiencing this with the device I'm testing esphomeyaml on as well. Config validates without issue but it doesn't connect. 
 I checked the serial logger and I could see it was crashing and rebooting continually.

OttoWinter, 2019-01-19:
> I checked the serial logger and I could see it was crashing and rebooting continually.

Yes that would lead to no connection. Please post logs or I won't be able to help

romanpeters, 2019-01-20:
I reflashed my Sonoff Basic over USB and now it works perfectly again.

shendriksza, 2019-01-22:
Exact same thing happened to me. My configuration stayed exactly the same, I just removed the MQTT configuration (I had api and mqtt running at the same time, was causing issues) and did OTA upload. I didn't check exactly what the output was when I uploaded, but it succeeded. Sonoff went offline and never came back, according to my router it's not even connecting to wifi