esphome/esphome #329: template cover state OPEN/CLOSED not working in version 1.10.0

MiKuBB, 2019-01-12:
hass 0.85.1 in virtualenv running on debian 9

ESP8266 (wemos d1_mini rev. 2.3)

Template cover

After upgrading esphomelib to 1.10.0 and hass to 0.85.1 and succesful migration to esphome native api previously working template cover stoped reporting OPEN/CLOSE state

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: status
    name: "ESP Braana: Status"
  - platform: gpio
    name: Zvoncek
    id: button_zvoncek
    pin:
      number: D1
      inverted: true
      mode: INPUT_PULLUP
    filters:
      - delayed_off: 10ms
  - platform: gpio
    name: Braana otvorena
    id: sensor_braana
    device_class: garage_door
    pin:
      number: D2
      inverted: true
      mode: INPUT_PULLUP
    filters:
      - delayed_on: 100ms

switch:
  - platform: restart
    id: restart
    name: "ESP Braana: Restart"
  - platform: gpio
    name: "ESP Braana: Switch"
    id: relay_1
    pin:
      number: D0
      inverted: true
    internal: true

cover:
  - platform: template
    name: "Braana"
    open_action:
      - switch.turn_on: relay_1
      - delay: 100ms
      - switch.turn_off: relay_1
    close_action:
      - switch.turn_on: relay_1
      - delay: 100ms
      - switch.turn_off: relay_1
    stop_action:
      - switch.turn_on: relay_1
      - delay: 100ms
      - switch.turn_off: relay_1
    lambda: |-
      if (id(sensor_braana).value) {
        return cover::COVER_OPEN;
      } else {
        return cover::COVER_CLOSED;
      }

```

**Traceback (if applicable):**

```

```

**Additional information:**

talondnb, 2019-01-13:
Came here to post the same - I have a garage door opener that no longer responds to gpio relay trigger since upgrading to 1.10. Manually publishing to the topic also doesn't not work.

```
# api:

esphomeyaml:
  name: garage_opener
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: 'xxxxxx'
  password: 'xxxxxx'
  manual_ip:
    static_ip: 192.168.0.131
    gateway: 192.168.0.1
    subnet: 255.255.255.0

mqtt:
  broker: '192.168.0.201'
  username: 'xxxx'
  password: 'xxxx'

# Enable logging
logger:

ota:
  password: 'xxxx'

binary_sensor:
  - platform: gpio
    pin:
      number: D2
      inverted: True
      mode: INPUT_PULLUP
    name: "Garage Door Closed"
    filters:
      - delayed_on: 10ms
      - delayed_off: 10ms
    device_class: garage_door

switch:
  - platform: gpio
    name: "Garage Door Trigger"
    pin: D7

sensor:
  - platform: ultrasonic
    filters:
      - filter_nan:
    trigger_pin: D5
    echo_pin: D6
    update_interval: 30s
    name: "Ultrasonic Car Sensor"
```

talondnb, 2019-01-13:
Also, API integration was hit and miss so I've disabled this to stick to MQTT, but still having issues. Any way to roll back to the previous version?

OttoWinter, 2019-01-13:
Will be fixed with https://github.com/home-assistant/home-assistant/pull/20048

@talondnb If you're using the command line (or docker) you can install using `pip2 install esphomeyaml==1.9.3`, but from Hass.io it's currently not possible.

closed 2019-01-13 by OttoWinter