esphome/esphome #347: platformio and esptool are not meant to be imported

dotlambda, 2019-01-16:
**Operating environment (Hass.io/Docker/pip/etc.):**
distribution package

**ESP (ESP32/ESP8266/Board/Sonoff):**
n/a

**Affected component:**
n/a

**Description of problem:**
I want to package ESPHome for NixOS: https://github.com/NixOS/nixpkgs/pull/53990
However, I had to create a patch that e.g. replaces `import platformio.__main__` and calling that by something that uses `subprocess.run()`.
The problem is that in NixOS, we have to write a wrapper script for platformio to work correctly and have to call that script instead of `platformio.__main__()`.
Since platformio and esptool are not meant to be imported anyway and you doing so relies on using internal APIs they could change at any time, I was wondering why you don't simply use `subprocess.run()` in the first place. I know you'd have to tell people to install three pieces of software to get ESPHome up and running but that doesn't seem like such a big deal as it can be done in one command line (`pip install platformio esptool esphomeyaml`).

**Problem-relevant YAML-configuration entries:**
n/a

**Traceback (if applicable):**
n/a

**Additional information:**

OttoWinter, 2019-01-16:
The reason they're imported in-process is because I've had a bad experience with python setups.
If you start `platformio` as a subprocess, there's a ton of things that can go wrong. For example if you're in a virtual environment and for some reason the `platformio` hook has not been installed, it will fail.

I'm also monitoring all platformio (and all other directly imported packages) closely, so I know if will break beforehand. 

> I know you'd have to tell people to install three pieces of software to get ESPHome up and running but that doesn't seem like such a big deal as it can be done in one command line (pip install platformio esptool esphomeyaml).

a) About that last part: With this tool I want to have a good installation experience, as that matters a lot for acquisition (also one of the areas ESPHome excels at the most, you can go from nothing to a fully working RGB-light integrated into HA in 20 minutes). Making the installation more difficult, even if it's only a bit, needs to be carefully considered

b) Why would they need to install all three? I mean it's already an installation requirement for esphomeyaml in setup.py

---------

So I'd rather want to keep the default way to have it imported in-process. If there's no other way for having this packaged I guess I could also add an environment option for that. So for example if `ESPHOME_USE_SUBPROCESS` (or whatever) is set, it will use `subprocess`

dotlambda, 2019-01-18:
> So I'd rather want to keep the default way to have it imported in-process. If there's no other way for having this packaged I guess I could also add an environment option for that. So for example if ESPHOME_USE_SUBPROCESS (or whatever) is set, it will use subprocess

This seems like a fair compromise. I've created a draft PR here: https://github.com/OttoWinter/esphomeyaml/pull/359
If you feel like it's better to not merge this, I'll apply and maintain a similar patch locally.

PR #359 (2019-01-19): call platformio and esptool using subprocess if $ESPHOME_USE_SUBPROCESS is set

## Description:

**Related issue (if applicable):** fixes #347 

**Pull request in [esphomedocs](https://github.com/OttoWinter/esphomedocs) with documentation (if applicable):** OttoWinter/esphomedocs#<esphomedocs PR number goes here>
**Pull request in [esphomelib](https://github.com/OttoWinter/esphomelib) with C++ framework changes (if applicable):** OttoWinter/esphomelib#<esphomelib PR number goes here>

## Checklist:
  - [x] The code change is tested and works locally.
  - [ ] Tests have been added to verify tha [... cut]
closed 2019-01-19 by OttoWinter