esphome/esphome #349: Compile -> Ter

ioudeveldhuis, 2019-01-16:
**Operating environment (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266/Board/Sonoff):**
ESP32

**Affected component:**
N/A

**Description of problem:**
When compiling the yaml below from the ESPHome Dashboard, the process stops with : Terminated process. 
No firmware can be downloaded after this.

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: crazylongname
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: 'obnoxiousssid'
  password: 'totallyobviouspassword'
  id: my_wifi

esp32_ble_tracker:
  scan_interval: 300s

binary_sensor:
```

**Traceback (if applicable):**
N/A

**Additional information:**
https://imgur.com/a/38SHZ3Y

OttoWinter, 2019-01-16:
I've seen this error before. I think it's related to the system that's running esphomeyaml somehow crashing (due to lack of resources, or some error in the compiler provided by espressif)

Just to confirm, what _system_ is this error happening on? Is it a RPi B (v1)? Anyway, the compiler is supplied by espressif (+ pre-compiled by platformio), and so I have no control over this part of the compilation process.

ioudeveldhuis, 2019-01-17:
The system Hass.io is running on (docker) is a Netgear ReadyNas RN314.
I am able to compile the following yaml file but that is without the _ble stuff which I need.

```
esphomeyaml:
  name: crazylongname
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: 'obnoxiousssid'
  password: 'totallyobviouspassword'

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'XXXXX'

ota:
  password: 'XXXXX'
```

Ilpredatore, 2019-01-18:
Hi Otto,
i have the same problem compiling code with option "esp32 ble tracker", after a while in the phase of linking, the system terminate process. If i start the procedure for updating with Ota flagged, instead of serial port, the procedure endes correctly. Then i restart upload procedure for flash via USB.

Without option "esp32 ble tracker" there isn't compiling problems.

I need of BLE gateway because i have 3 Xiaomi Mijia Temp sensor and i need btle extender for reach Home Assistant HUB.

Hardware:
-ESP32 WROOM32
-Raspberry 3 B+ running Hass.io
-3 Xiaomi Btle temp sensors

I have also disconnections issues with Hass.io, after some minutes or hours, Hass.io doen't receive data from sensors, but for that i open later a new thread..

ioudeveldhuis, 2019-01-18:
I can confirm that it is a host/platform issue. I installed esphomyaml on my windows 10 pc and I can compile and upload the yaml with esp32 ble stuff in it without any problems.
The only issue I am running into now is that if there is no connection with home assistant, the node 'hangs' itself after 3-5 minutes:
```
[09:37:44]Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[09:37:44] - IDLE (CPU 0)
[09:37:44]Tasks currently running:
[09:37:44]CPU 0: wifi
[09:37:44]CPU 1: IDLE
...........
[09:41:54]Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[09:41:54] - IDLE (CPU 0)
[09:41:54]Tasks currently running:
[09:41:54]CPU 0: wifi
[09:41:54]CPU 1: IDLE
[09:41:55]ASSERT_PARAM(-218959118 0), in arch_main.c at line 305
[09:41:55]Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
[09:41:55]Core 0 register dump:
[09:41:55]PC      : 0x40085f05  PS      : 0x00060434  A0      : 0x801a0770  A1      : 0x3ffc04e0
[09:41:55]A2      : 0x00000001  A3      : 0x00000000  A4      : 0x0000f2f2  A5      : 0x60008054
[09:41:55]A6      : 0x3ffc1030  A7      : 0x60008050  A8      : 0x80085f05  A9      : 0x3ffc04c0
[09:41:55]A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff
[09:41:55]A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005
[09:41:55]EXCVADDR: 0x00000000  LBEG    : 0x40085e3d  LEND    : 0x40085e44  LCOUNT  : 0x00000000
[09:41:55]Core 0 was running in ISR context:
[09:41:55]EPC1    : 0x40009203  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40085f05
[09:41:55]
[09:41:55]Backtrace: 0x40085f05:0x3ffc04e0 0x401a076d:0x3ffc0500 0x40019fb5:0x3ffc0520 0x40046683:0x3ffc0550 0x40047515:0x3ffc0570 0x4008767e:0x3ffc0590 0x4008814e:0x3ffc05b0 0x4008908e:0x3ffc05d0 0x4008a003:0x3ffc05f0 0x400817d9:0x3ffc0610 0x4008a9e7:0x00000000
[33mWARNING Found stack trace! Trying to decode it[0m
[09:41:55]
[09:41:55]Core 1 register dump:
[09:41:55]PC      : 0x401cff96  PS      : 0x00060034  A0      : 0x8009179a  A1      : 0x3ffd75b0
[09:41:55]A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffd70e0
[09:41:55]A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc8cec  A9      : 0x3ffc8cb0
[09:41:55]A10     : 0x00000000  A11     : 0x00000001  A12     : 0x80092abc  A13     : 0x3ffda550
[09:41:55]A14     : 0x00000000  A15     : 0x3ffd7260  SAR     : 0x00000000  EXCCAUSE: 0x00000005
[09:41:55]EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
[09:41:55]
[09:41:55]Backtrace: 0x401cff96:0x3ffd75b0 0x40091797:0x3ffd75d0
[33mWARNING Found stack trace! Trying to decode it[0m
[09:41:55]
[09:41:55]Rebooting...
[09:41:55]ets Jun  8 2016 00:22:57
[09:41:55]
[09:41:55]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[09:41:55]configsip: 0, SPIWP:0xee
[09:41:55]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[09:41:55]mode:DIO, clock div:2
[09:41:55]load:0x3fff0018,len:4
[09:41:55]load:0x3fff001c,len:952
[09:41:55]load:0x40078000,len:6084
[09:41:55]load:0x40080000,len:7944
[09:41:55]entry 0x40080310
[09:41:56]\033[0;32m[I][logger:071]: Log initialized\033[0m
```

Ilpredatore, 2019-01-18:
> I can confirm that it is a host/platform issue. I installed esphomyaml on my windows 10 pc and I can compile and upload the yaml with esp32 ble stuff in it without any problems.
> The only issue I am [...]

Ok, thank you, this evening i'll try with linux.

Can you report if have you some disconnection in next hours with hassio?

OttoWinter, 2019-01-18:
@ioudeveldhuis If you're seeing disconnects, I've heard the following helps (will become default in next version):

```yaml
esphomeyaml:
  arduino_version: dev
```

brandond, 2019-01-18:
Seems like embedded platforms (NAS, RPi, etc) frequently don't have enough memory to link a complicated image. That doesn't explain the stack trace once the code is successfully built and uploaded, but it might explain all the folks seeing a 'terminated process' error. I wonder if there's a good way to run the build with a memory limit, and handle OOM properly such that it doesn't kill the add-on? Folks building from the CLI can figure out from the error message what's going on, but if the add-on just crashes that doesn't make diagnosing the issue very easy.

OttoWinter, 2019-01-19:
> That doesn't explain the stack trace once the code is successfully built and uploaded

Yes, that's a separate issue (and seems to be fixed with recent esp-idf versions using `arduino_version: dev`)

> I wonder if there's a good way to run the build with a memory limit, and handle OOM properly such that it doesn't kill the add-on?

It doesn't kill the add-on from what I know. It just terminates gcc (guessing segmentation fault), I guess the add-on *could* parse the platformio output at real time and print a "nicer" error message.

ioudeveldhuis, 2019-01-28:
Added the arduino_dev but still crashes:

```
[0;32m[I][wifi:348]: WiFi connected![0m
[0;32m[I][application:089]: setup() finished successfully![0m
[0;32m[I][application:097]: You're running esphomelib v1.11.0-dev compiled on Jan 24 2019, 21:07:41[0m
[0;32m[I][application:114]: Running through first loop()[0m
[0;32m[I][application:141]: First loop finished successfully![0m
[0;32m[I][ota:154]: Boot seems successful, resetting boot loop counter.[0m
CORRUPT HEAP: Bad head at 0x3fff18f4. Expected 0xabba1234 got 0x3fff1c0c
assertion "head != NULL" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x401522e7 on core 0

Backtrace: 0x40091f24:0x3ffe8180 0x40092155:0x3ffe81a0 0x401522e7:0x3ffe81c0 0x40097709:0x3ffe81f0 0x4008498e:0x3ffe8210 0x4008b4fd:0x3ffe8230 0x4000bec7:0x3ffe8250 0x40171923:0x3ffe8270 0x40171983:0x3ffe8290 0x401719cb:0x3ffe82b0 0x4017414e:0x3ffe82d0 0x401741b6:0x3ffe82f0 0x401a3a75:0x3ffe8310 0x401702fc:0x3ffe8330 0x40096c21:0x3ffe8360

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5860
entry 0x40080698
[0;32m[I][logger:098]: Log initialized[0m
[0;32m[I][application:053]: Running through setup()...[0m
[0;32m[I][wifi:307]: - 'MyAccessPoint' (XX:XX:XX:XX:XX:XX) [0;33m?_[0;37m?¦[0m[0m
[0;32m[I][wifi:178]: WiFi Connecting to 'MyAccessPoint'...[0m
[0;32m[I][wifi:348]: WiFi connected![0m
[0;32m[I][application:089]: setup() finished successfully![0m
[0;32m[I][application:097]: You're running esphomelib v1.11.0-dev compiled on Jan 24 2019, 21:07:41[0m
[0;32m[I][application:114]: Running through first loop()[0m
[0;32m[I][application:141]: First loop finished successfully![0m
[0;32m[I][ota:154]: Boot seems successful, resetting boot loop counter.[0m
[1;31m[E][api:063]: No client connected to API. Rebooting...[0m
[0;32m[I][helpers:277]: Forcing a reboot... Reason: 'api'[0m
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5860
entry 0x40080698
[0;32m[I][logger:098]: Log initialized[0m
[0;32m[I][application:053]: Running through setup()...[0m
[0;32m[I][wifi:307]: - 'MyAccessPoint' (XX:XX:XX:XX:XX:XX) [0;32m?_?¦[0m[0m
[0;32m[I][wifi:178]: WiFi Connecting to 'MyAccessPoint'...[0m
[0;32m[I][wifi:348]: WiFi connected![0m
[0;32m[I][application:089]: setup() finished successfully![0m
[0;32m[I][application:097]: You're running esphomelib v1.11.0-dev compiled on Jan 24 2019, 21:07:41[0m
[0;32m[I][application:114]: Running through first loop()[0m
[0;32m[I][application:141]: First loop finished successfully![0m
[0;32m[I][ota:154]: Boot seems successful, resetting boot loop counter.[0m
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
```