esphome/esphome #350: Wrong password in main.cpp

itProfi, 2019-01-16:
In version 1.10.1 - i have password for my AP - "20182018."
[image]

In main.cpp after compile it we have! -"20182018.0" !
I try repruduce it 5 times afer delete all libs, same effect!

brandond, 2019-01-16:
You probably need to quote the password in your YAML file - the parser is treating it as a number instead of a string because it is numeric and unquoted. This is standard YAML behavior.

OttoWinter, 2019-01-16:
Yes this is how YAML works. If you type in something like `42.` it will get converted to the _number_ `42.0` and the information about how it was typed in originally is lost. You have to put quotes around these values in YAML

PR #351 (2019-01-18): Use strict string mode for WiFi password
Fixes #350

## Description:

**Related issue (if applicable):** fixes <link to issue>

**Pull request in [esphomedocs](https://github.com/OttoWinter/esphomedocs) with documentation (if applicable):** OttoWinter/esphomedocs#<esphomedocs PR number goes here>
**Pull request in [esphomelib](https://github.com/OttoWinter/esphomelib) with C++ framework changes (if applicable):** OttoWinter/esphomelib#<esphomelib PR number goes here>

## Checklist:
  - [ ] The code change is tested and works locally.
  - [ ] Tests have been added to verify that the new code wo [... cut]
commit d8fd5e9b (2019-01-18): Use strict string mode for WiFi password (#351)

Fixes #350
closed 2019-01-18 by OttoWinter