esphome/esphome #356: OTA 100% but still fails :(

vincegre, 2019-01-18:
Operating environment (Hass.io/Docker/pip/etc.): Hassio with ESPHome plugin in hassio 32 bits on Pi3B+ (updated at 1.10.1)
ESP (ESP32/ESP8266/Board/Sonoff): D1 Mini (but problem same with other ESP)
Affected component: OTA update
Description of problem: On an ESP running already 1.10.0 firmware, I try to do an OTA to push it at 1.10.1 but it fails. The upload of the firmware goes up to 100% but after it gets an error and fails. The ESP becomes unresponsive and I have to power reboot it and then it goes back online but still with old firmware ! I could reproduce with different ESP (so not specific to D1 Mini used here).

[image]

OttoWinter, 2019-01-18:
See the error please:
> Error: Finishing update failed. **See the MQTT/USB logs for more information.**

OttoWinter, 2019-02-02:
Closing for no response. Please re-open if error still exists and with more info

closed 2019-02-02 by OttoWinter