esphome/esphome #360: update_interval not working for dallas sensor

balk77, 2019-01-18:
**Operating environment (Hass.io/Docker/pip/etc.):**
Ubuntu 18.04
Sensor readout via MQTT to Home Assistant

**ESP (ESP8266): **
Wemos d1 mini

**Affected component:**
Dallas temperature sensor
https://esphomelib.com/esphomeyaml/components/sensor/dallas.html

**Description of problem:**
If I specify "update_interval: 30s" in my yaml file I am getting a compile error. According to the change log I should be able to specify this
https://esphomelib.com/esphomeyaml/changelog/v1.10.0.html#breaking-changes

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: dallas
    address: 0x9E0117B30B28FF28
    name: "CH aanvoer"
    state_topic: zolder/ketel/chaanvoer/temperatuur
    filters:
    id: chaanvoer
    update_interval: 30s
```

**Traceback (if applicable):**

```
INFO Reading configuration...
Failed config

sensor.dallas: [source config/keteltemp8266.yaml:28]
  platform: dallas
  address: 11385407390723145512
  name: CH aanvoer
  state_topic: zolder/ketel/chaanvoer/temperatuur
  filters: 
  id: chaanvoer
  
  [update_interval] is an invalid option for [sensor.dallas].
  update_interval: 30s
```

**Additional information:**
How do I specify the update interval for a Dallas sensor if this is a feature instead of a bug?

brandond, 2019-01-18:
`update_interval` goes in the [dallas](https://esphomelib.com/esphomeyaml/components/dallas.html) component, not the sensor

```yaml
dallas:
  - pin: 23
    update_interval: 30s
    id: dallas_23
sensor:
  - platform: dallas
    address: 0x9E0117B30B28FF28
    name: "CH aanvoer"
    dallas_id: dallas_23
```

balk77, 2019-01-19:
Ah, I see. That works. The docs state that the default interval for dallas is 15 seconds
(see https://esphomelib.com/esphomeyaml/components/dallas.html)
So I thought that was the internal update and that data push was by default 60 seconds. MQTT topics were pushed out on a 60 secs interval.

closed 2019-01-19 by balk77

brandond, 2019-01-19:
You're probably running into the default filter, which is I believe a 4 sample sliding window moving average. With a default 15 second interval, you'll only see an update pushed every 60 seconds. You can disable this by specifying an empty filter on the sensor.