esphome/esphome #366: after changing node name using native API old sensors remain in integrations

blakadder, 2019-01-19:
**Operating environment (Hass.io/Docker/pip/etc.):**
pip install under windows

**ESP (ESP32/ESP8266/Board/Sonoff):**
any board

**Affected component:**
Native API: https://esphomelib.com/esphomeyaml/components/api.html?highlight=native%20api

**Description of problem:**
After changing node name in .yaml Home Assistant retains all the old configuration in the integrations under the new name but using old entity_id names. 
Tried deleting by clicking on the trashcan icon and reconfiguring. didn't work
Tried deleting by clicking on the trashcan icon, resetting home assistant and reconfiguring. didn't work

**Additional information:**
http://prntscr.com/m9a83o
http://prntscr.com/m9a88v

mjoshd, 2019-01-20:
I ran into this issue too. The items are stored in 'config/.storage/core.entity_registry' in .json format. As of now, the only way I know of to remove the non-existent items is by manually deleting them from core.entity_registry and rebooting :-( Hopefully someone can find a better way.

OttoWinter, 2019-01-27:
That is an architectural issue with Home Assistant, and there's no way to fix it on ESPHome's end.

The HA guys wanted to add an entity registry editing feature some time ago (z-wave users also suffer from this I think), but never got around to.

Closing as not related to ESPHome

closed 2019-01-27 by OttoWinter