esphome/esphome #371: DHT22 issue

mssmison, 2019-01-20:
I'm running into issues with my DHT22
```
[22:39:33][W][sensor.dht:141]: Requesting data from DHT failed!
[22:39:33][W][sensor.dht:076]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
```
It works without issue on Tasmota.

My config is as follows (wifi info removed)

```
esphomeyaml:
  name: temp_humid_test
  platform: ESP8266
  board: d1_mini

sensor:
  - platform: dht
    pin: D2
    model: DHT22
    temperature:
      name: "temperature"
    humidity:
      name: "humidity"
    update_interval: 15s
```

mjoshd, 2019-01-20:
I'd suggest trying to define a different board model, sensor model, or defining the pin by GPIO#.

Here's what works for me with multiple different ESP8266 boards and a DHT22 ([the one with 3 pins](https://www.amazon.com/Gowoops-Temperature-Humidity-Measurement-Raspberry/dp/B073F472JL/ref=sr_1_3?ie=UTF8&qid=1547984626&sr=8-3&keywords=dht22)):

```yaml
esphomeyaml:
  name: esphome_dht_test
  platform: ESP8266
  board: nodemcuv2

sensor:
  - platform: dht
    pin: GPIO5 # D1
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    model: AM2302
    update_interval: 15s
```

mssmison, 2019-01-20:
I'll play around, but for sure there's an issue here. Even to your example your model is am2303 for a dht22 board.
 If it's the pin that's the issue hopefully it can be standardized how they are referenced.

 Thanks for your ideas though