esphome/esphome #374: Switch - miniaturized Sonoff clone

Styx85, 2019-01-20:
So this is one more basic question. Not even yet in the state using esphomeyaml, but maybe I can get some more ideas.

I bought some sonoff clones [here](https://www.amazon.de/gp/product/B07KW6BQVJ/ref=ppx_yo_dt_b_asin_title_o01__o00_s00?ie=UTF8&psc=1).

So first question why the clones. It's the size! They are tiny and therefore quite attractive for me to use. Have a look at the photos.

But they are not easy to flash so far. I broke the first one by desoldering the board (remember, never turn up the heat over 220 °C, even if in a hurry), but I could figure out which Pins are connected.

So there is easy access to
GPIO12
GPIO13
GPIO0
GND
VCC

There is an ESP8285 chip on the board. Does anyone know how to use these for flash access? I'm stuck so far. 

[image]
[image]
[image]
[image]
[image]

OttoWinter, 2019-01-20:
You need to access GPIO1 and 3 for flashing (see ESP8266 under devices section). For where to find these pins look at tge esp8265 datasheet.

Styx85, 2019-01-20:
Shut, aren't there any other flash modes that can be used. 
According to the ESP 8285  manual there are alternative modes, but I don't understand if it can be used to flash.
GPIO 12; HSPI_MISO
and
GPIO 13; HSPI_MOSI; UART0_CTS

If not I guess it's more soldering action.
As far as I can see the required pins 25 and 26 in the upper right aren't connected at all. And alternative pin 14 isn't connected, too.

Pin layout of the ESP8285 and ESP8266 is the same.

[ESP8266_Pin_List_0.zip](https://github.com/OttoWinter/esphomeyaml/files/2777051/ESP8266_Pin_List_0.zip)

[image]

brandond, 2019-01-21:
I'm really surprised they didn't bring out the bootloader pins they're almost always available for initial program load during manufacturing. I also don't see an external flash chip anywhere?

Styx85, 2019-01-21:
I think the flash chip is this one centralized on the main board. I was wondering about the boatloader pins to, somehow you have to flash the firmware during production and to keep things simple and cheap its probably done by machines.

[image]
[image]

Styx85, 2019-01-22:
OK, call me crazy but I did it. But a little bit more improvised using household pins to pin the solder  of RXD and UXD and a bridged connection with paper clips.

It is working with the reference for basic sonoff so far

> esphomeyaml:
>   name: <NAME_OF_NODE>
>   platform: ESP8266
>   board: esp01_1m
>   board_flash_mode: dout
> 
> wifi:sp01_1m
board_flash_mode: dout

wifi:
ssid: <YOUR_SSID>
password: <YOUR_PASSWORD>

mqtt:
broker: <YOUR_MQTT_BROKER>
username: <YOUR_USERNAME>
password: <YOUR_PASSWORD>

logger:
>   ssid: <YOUR_SSID>
>   password: <YOUR_PASSWORD>
> 
> mqtt:
>   broker: <YOUR_MQTT_BROKER>
>   username: <YOUR_USERNAME>
>   password: <YOUR_PASSWORD>
> 
> logger:
> 
> ota:
> 
> binary_sensor:
>   - platform: gpio
>     pin:
>       number: GPIO0
>       mode: INPUT_PULLUP
>       inverted: True
>     name: "Sonoff Basic Button"
>     on_press:
>       - switch.toggle: relay
> 
> switch:
>   - platform: gpio
>     name: "Sonoff Basic Relay"
>     pin: GPIO12
>     id: relay
> 
> output:
>   - platform: esp8266_pwm
>     id: basic_green_led
>     pin:
>       number: GPIO13
>       inverted: True
> 
> light:
>   - platform: monochromatic
>     name: "Sonoff Basic Green LED"
>     output: basic_green_led

[image]
[image]

brandond, 2019-01-22:
That is incredible. I thought I was out there with my extensive collection of oddly trimmed and bent probe clips.

meijerwynand, 2019-01-24:
Man, you have some patience and determination. Well done!

Styx85, 2019-01-26:
One thing to mention is that these switches are really stable so far.
I have another [Sonoff clone](https://github.com/esphome/ESPHome-Core/issues/343) and that is rebooting randomly on some days, losing connection randomly and is sometimes really annoying.
 The ones here are harder to flash but so far enjoyable.

akasma74, 2019-01-26:
It says "Currently unavailable" on Amazon.de
Thanks for supplying the yaml, could you describe what was exactly done to flash it (the hardware bit)?

Styx85, 2019-01-26:
First some alternative [amazon link](https://www.amazon.de/Flower205-Intelligente-Modifikation-Sprachsteuerung-Modifizierte/dp/B07D5572VJ/ref=pd_sbs_60_3/257-6517271-6546904?_encoding=UTF8&pd_rd_i=B07D5572VJ&pd_rd_r=c333187c-21ba-11e9-9599-f7e3d20a9d9c&pd_rd_w=qp6qf&pd_rd_wg=MWzFt&pf_rd_p=51bcaa00-4765-4e8f-a690-5db3c9ed1b31&pf_rd_r=HGDGN8STMJCV44YF6AG2&psc=1&refRID=HGDGN8STMJCV44YF6AG2). These are essentially the same. In fact, it's the first one I bought and disassembled. 

You can find them on [aliexpress](https://de.aliexpress.com/item/LX-00G-DIY-Wi-Fi-Wireless-Mini-Smart-Switch-Fernbedienung-Modul-f-r-Smart-Home-Arbeit/32862726911.html?spm=a2g0x.search0104.8.38.61145900zeD9hq) and [banggood](https://www.banggood.com/de/EJLink-LX-00G-DIY-Wi-Fi-Wireless-Smart-Switch-Remote-Control-Module-for-Smart-Home-p-1190993.html?cur_warehouse=CN), too. Try to search for LX-00G DIY Wi-Fi. I think they are also sold on ebay.
You can get them for round about 6-7€.

OK, I'll try to draw.

The pin out is as follows:
GPIO0: switch
GPIO12: relay
GPIO13: LED

[image]

I recommend to solder GND and VCC to some connection cables, because it's a little bit tricky to pin RXD and TXD. I also soldered GPIO0, because I didn't want to move the board by pressing the switch later.

[image]

Next step is to pin TXD and RXD in the upper right of the chip.

[image]

Now connect them all to your serial to USB adapter and you are good to go. don't forget to cross over TXD/RXD while connecting.

akasma74, 2019-01-26:
Thanks for the writeup. What are dimensions of the board itself?

brandond, 2019-01-27:
I picked up a couple Shelly 1s. They're working great so far, can be wired up to use the wall switch as an input, and have an exposed set of pins for flashing. They're tiny, but a little pricier.

Styx85, 2019-01-27:
@akasma74 Here are some pics for you to see the size. But as @brandond said, the shelly 1 is tiny and not that pricy. With some luck they are 10€.

[image]
![img_20190
[image]
127_132721](https://user-images.githubusercontent.com/32103696/51801018-95e77e00-2238-11e9-8065-fdafc24301c1.jpg)
[image]
[image]

akasma74, 2019-01-27:
Unfortunately, the board itself doesn't look much smaller than the Sonoff Basic's one.
If space is an issue, I'd rather use Wemos D1 mini + small 230v to 3.3V converter + relay/sensor etc.

OttoWinter, 2019-02-02:
Awesome work! I'll close this issue now (because this is a bug tracker). It would be great if this could be converted to a device guide on the website.

closed 2019-02-02 by OttoWinter