esphome/esphome #376: Devices get discovered with .local

frog32, 2019-01-21:
**Operating environment (Hass.io/Docker/pip/etc.):**

hassio 0.86.0b1
**ESP (ESP32/ESP8266/Board/Sonoff):**

ESP8266
**Affected component:**

API

**Description of problem:**
Home Assistant discovers ESP home devices with .local even if domain is set to .lan

**Problem-relevant YAML-configuration entries:**
```yaml
wifi:
  ssid: 'WIFI'
  password: '123456'
  domain: .lan
```

**Traceback (if applicable):**

```

```

**Additional information:**

frog32, 2019-01-26:
I had problem that the esphomeyaml hassio addon was not able to resolve .local domains. How is this addon supposed to resolve those domains?

frog32, 2019-01-26:
Just tested it without changing the domain and the result is that Home Assistant sees the node but the esphomeyaml addon is unable to resolve .local domains.

OttoWinter, 2019-01-27:
> I had problem that the esphomeyaml hassio addon was not able to resolve .local domains. How is this addon supposed to resolve those domains?

The addon uses the `domain` option just like you specified in your config.

Please describe your issue further. Is the issue that the integration screen in Home Assistant auto-discovers the node with `.local` ?

Because in that case there's a very simple fix: In that exact screen, just change it to `.lan` and everything should be fine.

JanSchnacki, 2019-02-03:
The issue (at least for me, I just came here because this annoys me and I started searching) is, that Home Assistant keeps telling me, that it disovered new ESPHome devices. I currently have exactly one such device. I have configured it. It's running fine. But the little bell-symbol on the start page keeps sprouting the little orange "hi there, there is something you need to take care of" dot.

Also the "history" view keeips showing the "notifying" state for "config entry discovery".

You are correct, I could just change the ".local" to my local domain name, and the sensor now works just like it should. But the whole reason for this little dot on the bell-symbol is to tell me that there's something new requiring my attention. The way the ESPHome integration works right now this is broken.