esphome/esphome #378: Header File linking error with custom component

swampyclark, 2019-01-23:
**Operating environment (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266/Board/Sonoff):**

**Affected component:**

**Description of problem:**
Using the include in the yaml config file results in a unlinkable header in main.cpp.  Validate will pass but you can not compile.   

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: alarm
  platform: ESP8266
  board: nodemcuv2
  includes:
   - my_custom_component.h

```

**Traceback (if applicable):**

```
src/main.cpp:5:42: fatal error: ../../../my_custom_component.h: No such file or directory
#include "../../../my_custom_component.h"
^
compilation terminated.
*** [.pioenvs/alarm/src/main.cpp.o] Error 1
```

**Additional information:**
current work around, put the header file in
/config/esphomeyaml/alarm/.piolibdeps/esphomelib/src/esphomelib/my_custom_component.h
with alarm being my componet name

OttoWinter, 2019-01-25:
Where did you put the `my_custom_component.h` file?

yawor, 2019-01-29:
There's a problem with `#include` path calculation. I've created a PR with the fix, which is really simple.

closed 2019-01-29 by OttoWinter

swampyclark, 2019-01-31:
Looks like this PR may have broke all the base headers or might be something else.  Just tried the basic binary sensor example and got
Compiling .pioenvs/test/lib4d9/ESP8266WiFi/ESP8266WiFi.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:21:3: error: 'MakeGPIOBinarySensor' is not a member of 'esphomelib::Application'
Application::MakeGPIOBinarySensor application_makegpiobinarysensor = App.make_gpio_binary_sensor("Living Room Window", 4);
^
src/main.cpp:21:37: error: expected ';' before 'application_makegpiobinarysensor'
Application::MakeGPIOBinarySensor application_makegpiobinarysensor = App.make_gpio_binary_sensor("Living Room Window", 4);
^
src/main.cpp:22:87: error: 'application_makegpiobinarysensor' was not declared in this scope
binary_sensor::GPIOBinarySensorComponent *binary_sensor_gpiobinarysensorcomponent = application_makegpiobinarysensor.gpio;
^
src/main.cpp:23:3: error: 'MQTTBinarySensorComponent' is not a member of 'esphomelib::binary_sensor'
binary_sensor::MQTTBinarySensorComponent *binary_sensor_mqttbinarysensorcomponent = application_makegpiobinarysensor.mqtt;
^
src/main.cpp:23:45: error: 'binary_sensor_mqttbinarysensorcomponent' was not declared in this scope
binary_sensor::MQTTBinarySensorComponent *binary_sensor_mqttbinarysensorcomponent = application_makegpiobinarysensor.mqtt;
^
Compiling .pioenvs/test/lib4d9/ESP8266WiFi/ESP8266WiFiAP.cpp.o

PR #390 (2019-01-29): Includes should be relative to the src directory, not main.cpp file
## Description:
The `includes` option checks file existence relative to the directory containing a yaml file. On the other hand the path inserted into the generated main.cpp file is calculated to be relative to the file. The problem is that the path is calculated relative to the file itself and not the directory it's in (src directory) making the path invalid by adding too much parent directories (`..`) in the path. This small change fixes it.

**Related issue (if applicable):** fixes #378 

**Pull request in [esphomedocs](htt [... cut]