esphome/esphome #380: RGBW light: brightness should not affect white value (W led brightness)?

peterforeman, 2019-01-23:
**Operating environment (Hass.io/Docker/pip/etc.):**

Docker repo

**ESP (ESP32/ESP8266/Board/Sonoff):**

MagicHome LED controller (PLATFORMIO_ESP8285) (https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller)

**Affected component:**

RGBW light

**Description of problem:**
I have a RGBW light strip with, as you can see in the config, four GPIO pins, one for each channel. In Home Assistant I'm expecting to be able to set the RGB and W leds independently from each other. Meaning setting the brightness should affect the RGB led, and the "White value" should affect the W led brightness.

Unfortunately, the brightness applies to both the RGB and the W leds, meaning I cannot turn on the W-leds and leave the RGB off.

**Problem-relevant YAML-configuration entries:**
```yaml
light:
  - platform: rgbw
    name: RGBW light
    red: pwm_r
    green: pwm_g
    blue: pwm_b
    white: pwm_w

output:
  - platform: esp8266_pwm
    pin: GPIO12
    frequency: 1000 Hz
    id: pwm_r

  - platform: esp8266_pwm
    pin: GPIO5
    frequency: 1000 Hz
    id: pwm_g

  - platform: esp8266_pwm
    pin: GPIO13
    frequency: 1000 Hz
    id: pwm_b

  - platform: esp8266_pwm
    pin: GPIO15
    frequency: 1000 Hz
    id: pwm_w
```

**Traceback (if applicable):**

```

```

**Additional information:**

OttoWinter, 2019-01-25:
Fixed in dev

closed 2019-01-25 by OttoWinter

peterforeman, 2019-01-27:
I tried dev (docker container), flashed it but did not work for me. Perhaps I missed something?