esphome/esphome #381: 4CH Pro R1 not working (bootloop)

peterforeman, 2019-01-23:
**Operating environment (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266/Board/Sonoff):**

Sonoff 4CH Pro R1

```
ESP Chip Id 10442015
Flash Chip Id 0x144051
Flash Size 1024kB
```

**Affected component:**

"Generic Sonoff"

**Description of problem:**
Uploading compiled firmware on my Tasmota-flash Sonoff 4CH Pro R1 creates a bootloop. The device never connects to WiFi anymore. Tried flashing via OTA (first time), and several times via serial. I tried the ESP8285 board (which is in the 4CH Pro R1) as well as the esp8266 board config. Nothing worked. Flashed Tasmota binary again which worked instantly (even remembered the settings).

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: 4chpro
  platform: ESP8266
  board: esp8285
  board_flash_mode: dout
```

**Traceback (if applicable):**

```
Not available yet, will update asap with serial logging.
```

**Additional information:**

meijerwynand, 2019-01-24:
Greetings, 

I am not sure about the difference in the 4CH and 4CH Pro. I have the former and flashed it today without any issue.

Here is the config I use.

```yaml
esphomeyaml:
  name: ghrotto
  platform: ESP8266
  board: esp01_1m
  board_flash_mode: dout
  esphomelib_version: 1.10.1

...

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ## use fast_connect to connect to hidden WiFi SSID
  fast_connect: true

```

OttoWinter, 2019-01-25:
Yes, serial logs would be required for debugging this issue. Please also make sure you read the FAQ entry about Wifi.

snicker, 2019-01-31:
just passing through doing some research for a project to see if the pro will work for me... is it possible this is the problem?

The "FW/IO0" button (Switch 1) is not directly connected to GPIO0 of the ESP module. A different method has to be used to program this board.

see here for more and possible solution:

 https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-4CH-and-4CH-Pro 

Really interested in using esphome for this project but I also want the isolated relays for mains and low voltage control for this project that only the Pro has.

OttoWinter, 2019-02-01:
@snicker GPIO0 needs to be LOW to enter programming mode (see generic ESP8266 docs). You need to find some way to access that pin.

brandond, 2019-02-01:
It sounds like some models you can just hold down the button to pull it low; on others you have to solder a connection. This appears to be pretty well documented by the tasmota folks though.