esphome/esphome #383: Cannot compile anything due to LibraryManager SSL-related error (OS X)

akasma74, 2019-01-26:
**Operating environment (Hass.io/Docker/pip/etc.):**
OS X El Capitan (10.11.6)

**ESP (ESP32/ESP8266/Board/Sonoff):**
Wemos D1 mini

**Affected component:**
don't know

**Description of problem:**
I'm trying to build a simple project.
Install of the esphomeyaml did't went through until I added `--ignore-installed six` to `sudo pip install esphomeyaml` as it was unable to uninstall six because it is part of another package. There was another message "matplot requires nose, that is not installed". Don't know if these things affect anything though.

I was able to generate a yaml file but get the following error on `esphomeyaml wemos_d1_mini.yaml run`:

SSLError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /OttoWinter/esphomelib/archive/v1.10.1.zip (Caused by SSLError(SSLError(1, u'[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)'),))

It complains about the very same string in _ssl.c code as it did when I tried to upgrade platformio using the old version of pip that could not handle some ssl configurations properly as far as I understand.
Since then I updated pip to `19.0.1 from /Library/Python/2.7/site-packages/pip (python 2.7)`
Platformio's version is 3.6.4
I successfully compiled and uploaded to the board in question the latest Sonoff Tasmota sources using platformio, but pio run in dht_on_wemos folder fails with the above SSL-related error.

**Problem-relevant YAML-configuration entries:**
```yams
esphomeyaml:
  name: dht_on_wemos
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'Xxx'
  password: 'xx'

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'xx'

ota:
  password: 'xx'

sensor:
  - platform: dht
    pin: D4
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    update_interval: 30s
```

**Traceback (if applicable):**
esphomeyaml wemos_d1_mini.yaml run
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d dht_on_wemos
Processing dht_on_wemos (platform: espressif8266@1.8.0; framework: arduino; board: d1_mini)
---------------------------------------------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing v1.10.1
Error: Please read http://bit.ly/package-manager-ioerror
Error: Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/Library/Python/2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Library/Python/2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Python/2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Python/2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Library/Python/2.7/site-packages/platformio/commands/run.py", line 107, in cli
    result = (envname, ep.process())
  File "/Library/Python/2.7/site-packages/platformio/commands/run.py", line 207, in process
    result = self._run()
  File "/Library/Python/2.7/site-packages/platformio/commands/run.py", line 302, in _run
    self.verbose)
  File "/Library/Python/2.7/site-packages/platformio/commands/run.py", line 325, in _autoinstall_libdeps
    ctx.invoke(cmd_lib_install, libraries=[lib], silent=not verbose)
  File "/Library/Python/2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Library/Python/2.7/site-packages/click/decorators.py", line 26, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "/Library/Python/2.7/site-packages/platformio/commands/lib.py", line 103, in lib_install
    library, silent=silent, interactive=interactive, force=force)
  File "/Library/Python/2.7/site-packages/platformio/managers/lib.py", line 352, in install
    force=force)
  File "/Library/Python/2.7/site-packages/platformio/managers/package.py", line 721, in install
    name, url, requirements, track=True)
  File "/Library/Python/2.7/site-packages/platformio/managers/package.py", line 508, in _install_from_url
    dlpath = self.download(url, tmp_dir, sha1)
  File "/Library/Python/2.7/site-packages/platformio/managers/package.py", line 199, in download
    raise e
SSLError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /OttoWinter/esphomelib/archive/v1.10.1.zip (Caused by SSLError(SSLError(1, u'[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)'),))
```

```

**Additional information:**

brandond, 2019-01-26:
This is a Python error. Your Python version appears to be too old to support the version of SSL required to communicate with github.com for package downloads. What OS is this on? What does `python --version` show?

akasma74, 2019-01-26:
Python 2.7.10, it's possibly the original one that came with the OS (El Capitan).
I understand that platformio is not compatible with python 3.x

brandond, 2019-01-27:
I would try grabbing 2.7.15, either from python.org, or via brew if you're familiar with that.

akasma74, 2019-01-27:
I installed both 2.7.15 and 3.7.2 through home-brew and ran "Install Certificates.command":
python -V
Python 2.7.15

python3 -V
Python 3.7.2

but still have the same error.. apparently I'm missing something I need to do so Library Manager calls the right python, but I don't know what else to do :(
Any ideas?

OttoWinter, 2019-01-27:
Try running `which python` - and see what `brew doctor` and `brew link --overwrite python` output.

OttoWinter, 2019-01-27:
This looks like it might be an issue:

```
Having additional scripts in your path can confuse software installed via
Homebrew if the config script overrides a system or Homebrew provided
script of the same name. We found the following "config" scripts:
/Library/Frameworks/Python.framework/Versions/3.7/bin/python3-config
/Library/Frameworks/Python.framework/Versions/3.7/bin/python3.7-config
/Library/Frameworks/Python.framework/Versions/3.7/bin/python3.7m-config
```

I don't know how to help with that further, but I'm sure there's some solution on the internet

akasma74, 2019-02-06:
Ok, posting it here to anyone else having the same issue.
Yes, the key is to install everything properly. I solved my problem doing the following (after reading [this](https://docs.brew.sh)):
1. Uninstalled all python instances I'd installed as administrator earlier via brew, by logging as administrator and issuing something like `brew uninstall --ignore-dependencies python`
2. Logged as a normal user
3. Installed the latest python2: brew install python2 (it installs the latest python3 as well, but don't worry)
4. Install virtualenv: python2 -m pip install virtualenv
5. Run virtualenv: python -m virtualenv -p $$(brew --prefix)/bin/python2 path_to_your_env_folder
6. source path_to_your_env_folder/bin/activate
7. python -V should show Python 2.X.XX
8. Now install easyhomeyaml: python -m pip install esphomeyaml --user
9. You may need to add something to your PATH variable if there are appropriate messages during installation.
10. It's time to run esphomeyaml: esphomeyaml youryaml.yaml run