esphome/esphome #387: Remote Transmitter Switch does not support Sony codes fully

chris-jennings, 2019-01-28:
**Operating environment (Hass.io/Docker/pip/etc.):**
Ubuntu linux with running esphome in a python2 virtual environment

**ESP (ESP32/ESP8266/Board/Sonoff):**
WeMos D1 mini lite

**Affected component:**
Remote Transmitter Switch (in particular Sony remote support)
[Remote Transmitter](https://esphomelib.com/esphomeyaml/components/switch/remote_transmitter.html)

**Description of problem:**
I have successfully installed the WeMod D1 mini lite hardware for both collecting IR codes and transmitting IR codes. I have managed to get my Harmon Kardon receiver to work with the NEC codes that I was able to collect.

I tried to continue this process with my Sony TV and I didn't have any luck. I understand that the Sony TV requires 3 repeats of the same code with a delay between each repeat. I attempted this using the following yaml entry:

```
switch:
  - platform: remote_transmitter
    name: "Sony TV Power"
    sony:
      data: 0x00000A90
      nbits: 12
    repeat: 3
  - platform: remote_transmitter
    name: "Sony TV Volume Up"
    sony:
      data: 0x00000490
      nbits: 12
    repeat: 3
  - platform: remote_transmitter
    name: "Sony TV Volume Down"
    sony:
      data: 0x00000C90
      nbits: 12
```

I get no response using this method.

I looked at other esp8266 projects used for IR remote control and found this one:
[ESP8266-HTTP-IR-Blaster](https://github.com/mdhiggins/ESP8266-HTTP-IR-Blaster)
Using that code and reprogramming the same hardware with this system allowed to me successfully send a power on signal to the Sony TV. However I needed to use the following request:
`http://192.168.86.94/msg?code=A90:SONY:12&pulse=3&pdelay=100&pass=1234&simple=1`

This sends the same Sony A90 code, but three times and a delay between repeats of 100ms. I believe that the Sony  remote_transmitter entry requires these extra parameters to enable it to work in some cases.

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: wemos2
  platform: ESP8266
  board: d1_mini_lite

wifi:
  ssid: ''
  password: ''

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

switch:
  - platform: remote_transmitter
    name: "Sony TV Power"
    sony:
      data: 0x00000A90
      nbits: 12
    repeat: 3
  - platform: remote_transmitter
    name: "Sony TV Volume Up"
    sony:
      data: 0x00000490
      nbits: 12
    repeat: 3
  - platform: remote_transmitter
    name: "Sony TV Volume Down"
    sony:
      data: 0x00000C90
      nbits: 12
    repeat: 3
  - platform: remote_transmitter
    name: "Sony TV Source"
    sony:
      data: 0x00000A50
      nbits: 12
    repeat: 3
  - platform: remote_transmitter
    name: "Sony TV Menu"
    sony:
      data: 0x00000070
      nbits: 12
    repeat: 3
  - platform: remote_transmitter
    name: "Sony TV Menu Up"
    sony:
      data: 0x000002F0
      nbits: 12
    repeat: 3
  - platform: remote_transmitter
    name: "Sony TV Menu Down"
    sony:
      data: 0x00000AF0
      nbits: 12
    repeat: 3
  - platform: remote_transmitter
    name: "Sony TV Menu Select"
    sony:
      data: 0x00000A70
      nbits: 12
    repeat: 3
  - platform: remote_transmitter
    name: "IR TX1 Amplifier On"
    nec:
      address: 0x010E
      command: 0x03FC
  - platform: remote_transmitter
    name: "IR TX1 Amplifier Off"
    nec:
      address: 0x010E
      command: 0xF906

remote_transmitter:
  pin: D5
  carrier_duty_percent: 50%

```

**Traceback (if applicable):**

```

```

**Additional information:**
Sorry I couldn't trace this back to the code and work it out. I'm not up to speed with Python.

OttoWinter, 2019-01-28:
See wait_time please. Or otherwise use raw codes.

closed 2019-01-28 by OttoWinter

chris-jennings, 2019-01-28:
> 
> 
> See wait_time please. Or otherwise use raw codes.

OK, I just used wait_time and I still does not respond. I also tried capturing a raw code and resending that but that didn't work. I'll try more troubleshooting, its strange that different code on the same hardware works successfully. I'll see if I can capture the difference between the two systems using another ESP8266.

chris-jennings, 2019-01-29:
OK, so I did further testing and the esphome IR transmitter switch does not transmitter the correct codes. Using another ESP8266 to receive the codes transmitted esphome gave the following results:

```
[16:31:57][D][remote.raw:064]: Received Raw: 2368, -603, 608, -609, 1266, -550, 606, -610, 1266, -551, 573, -643, 1265, -551, 606, -614, 602, -610, 1266, -551, 574, -641, 574, -643, 574, -643, 573, -10000
[16:31:58][D][remote.raw:064]: Received Raw: 2457, -556, 571, -641, 1267, -550, 574, -642, 1266, -550, 574, -643, 1265, -552, 572, -642, 606, -611, 1266, -550, 605, -611, 605, -611, 605, -612, 606, -10000
[16:31:58][D][remote.raw:064]: Received Raw: 2467, -547, 574, -641, 1266, -551, 607, -611, 1264, -550, 605, -611, 1267, -550, 605, -610, 606, -611, 1265, -553, 609, -605, 574, -642, 607, -609, 608, -10000
```
 When using ESP8266-HTTP-IR-Blaster on the same hardware, with the same circuit I get this result:
```

[16:40:59][D][remote.sony:095]: Received Sony: data=0x00000A90, nbits=12
[16:40:59][D][remote.sony:095]: Received Sony: data=0x00000A90, nbits=12
[16:40:59][D][remote.sony:095]: Received Sony: data=0x00000A90, nbits=12
[16:40:59][D][remote.sony:095]: Received Sony: data=0x00000A90, nbits=12
[16:40:59][D][remote.sony:095]: Received Sony: data=0x00000A90, nbits=12
[16:40:59][D][remote.sony:095]: Received Sony: data=0x00000A90, nbits=12
[16:40:59][D][remote.sony:095]: Received Sony: data=0x00000A90, nbits=12
[16:40:59][D][remote.sony:095]: Received Sony: data=0x00000A90, nbits=12
[16:40:59][D][remote.sony:095]: Received Sony: data=0x00000A90, nbits=12
```

So the receiving library appears to be working correctly but transmitting does not. I haven't determined the root cause. I need to look at what the A90 Sony code is as a raw signal then I might have a clue where the problem is.

chris-jennings, 2019-01-29:
The RAW Sony A90 code from ESP8266-HTTP-IR-Blaser looks like this:
```
[17:32:31][D][remote.raw:064]: Received Raw: 2414, -592, 1250, -569, 645, -564, 1249, -559, 646, -564, 1250, -559, 646, -564, 645, -563, 1251, -559, 645, -564, 645, -565, 646, -563, 645, -10000
[17:32:31][D][remote.raw:064]: Received Raw: 2429, -583, 1250, -564, 645, -565, 1249, -561, 643, -565, 1250, -560, 645, -563, 645, -565, 1249, -559, 646, -563, 645, -564, 645, -564, 645, -10000
[17:32:31][D][remote.raw:064]: Received Raw: 2459, -553, 1250, -559, 646, -564, 1250, -558, 646, -563, 1250, -559, 646, -564, 645, -563, 1250, -559, 646, -564, 645, -563, 648, -561, 648, -10000
[17:32:31][D][remote.raw:064]: Received Raw: 2430, -592, 1250, -568, 644, -566, 1250, -558, 645, -565, 1250, -559, 644, -565, 644, -565, 1251, -557, 646, -564, 646, -563, 644, -565, 645, -10000
[17:32:31][D][remote.raw:064]: Received Raw: 2461, -553, 1249, -560, 646, -562, 1250, -560, 645, -564, 1250, -559, 646, -564, 645, -565, 1248, -560, 645, -564, 644, -565, 644, -565, 646, -10000
[17:32:31][D][remote.raw:064]: Received Raw: 2458, -554, 1248, -560, 645, -565, 1250, -558, 645, -564, 1250, -560, 644, -565, 645, -563, 1248, -561, 645, -565, 644, -564, 646, -564, 644, -10000
[17:32:31][D][remote.raw:064]: Received Raw: 2424, -591, 1251, -563, 646, -562, 1251, -559, 646, -563, 1249, -560, 646, -563, 645, -565, 1249, -558, 646, -564, 645, -564, 645, -564, 646, -10000
[17:32:31][D][remote.raw:064]: Received Raw: 2441, -564, 1251, -558, 650, -560, 1249, -560, 645, -564, 1250, -559, 645, -564, 645, -564, 1249, -560, 644, -565, 646, -562, 646, -564, 645, -10000
[17:32:31][D][remote.raw:064]: Received Raw: 2420, -589, 1249, -568, 646, -563, 1249, -560, 645, -564, 1251, -558, 646, -563, 645, -565, 1250, -559, 646, -563, 645, -563, 645, -564, 647, -10000
```
It has two extra codes in the first section:
Correct
2500, -600, 1250, -550, 650,  550, -1250
Incorrect
2500, MISSING, MISSING, -600, 600, -600, 1250