esphome/esphome #388: Debounce Filters Don't Seem to be Working

MrPaulAR, 2019-01-28:
**Operating environment (Hass.io/Docker/pip/etc.):**
pip via pipenv

**ESP (ESP32/ESP8266/Board/Sonoff):**
ESP32 (Wemos D32)

**Affected component:**
binary-sensor

**Description of problem:**
The debounce filters don't seem to be working.  I've tried timings up to 1 seconds down to .01 second with similar results.  On my test bench I have a jumper plugged into ground of the D32.  I can then take the free end of the jumper and quickly drag it across the free pins or just tap a pin and after the delay defined in the filter I get an ON message.  I don't ever get an OFF message if I do this.  If I hold the jumper in place for a time period greater than the filters then I get the ON & OFF messages.

```
[18:44:51][D][binary_sensor:027]: 'PIN_22': Sending state ON
[18:44:52][D][binary_sensor:027]: 'PIN_21': Sending state ON
[18:44:52][D][binary_sensor:027]: 'PIN_19': Sending state ON
[18:44:52][D][binary_sensor:027]: 'PIN_18': Sending state ON
[18:44:52][D][binary_sensor:027]: 'PIN_17': Sending state ON
[18:44:52][D][binary_sensor:027]: 'PIN_16': Sending state ON
[18:44:52][D][binary_sensor:027]: 'PIN_4': Sending state ON
[18:44:52][D][binary_sensor:027]: 'PIN_2': Sending state ON
```

**Problem-relevant YAML-configuration entries:**
```
(esphome)  pthomas@raspberrypi  ◰² esphome-KxSwxlP7  ~/esphome  esphomeyaml lolin_d32_test.yaml config                                           6.9s  Mon 28 Jan 2019 18:51:04 UTC
INFO Reading configuration...
INFO Configuration is valid!
esphomeyaml:
  name: lolin_d32_test
  platform: ESP32
  board: lolin_d32
  build_path: lolin_d32_test
  arduino_version: espressif32@1.5.0
  esphomelib_version:
    repository: https://github.com/OttoWinter/esphomelib.git
    tag: v1.10.1
  use_custom_code: false
wifi:
  domain: .local
  networks:
  - password: XXXXXXXXXXXXXXXXXXX
    ssid: XXXXXXXXXXXXXXXXXXX
logger:
  baud_rate: 115200
api:
  password: ''
  port: 6053
ota:
  safe_mode: true
sensor:
- platform: wifi_signal
  name: WiFi Signal Sensor
  update_interval: 60s
output:
- platform: gpio
  pin:
    number: 5
  id: gpio_5
binary_sensor:
- platform: gpio
  pin:
    number: 35
    mode: INPUT_PULLUP
  name: PIN_35
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 32
    mode: INPUT_PULLUP
  name: PIN_32
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 33
    mode: INPUT_PULLUP
  name: PIN_33
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 25
    mode: INPUT_PULLUP
  name: PIN_25
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 26
    mode: INPUT_PULLUP
  name: PIN_26
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 27
    mode: INPUT_PULLUP
  name: PIN_27
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 14
    mode: INPUT_PULLUP
  name: PIN_14
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 23
    mode: INPUT_PULLUP
  name: PIN_23
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 22
    mode: INPUT_PULLUP
  name: PIN_22
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 21
    mode: INPUT_PULLUP
  name: PIN_21
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 19
    mode: INPUT_PULLUP
  name: PIN_19
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 18
    mode: INPUT_PULLUP
  name: PIN_18
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 17
    mode: INPUT_PULLUP
  name: PIN_17
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 16
    mode: INPUT_PULLUP
  name: PIN_16
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 4
    mode: INPUT_PULLUP
  name: PIN_4
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 2
    mode: INPUT_PULLUP
  name: PIN_2
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
- platform: gpio
  pin:
    number: 15
    mode: INPUT_PULLUP
  name: PIN_15
  filters:
  - delayed_off: 1s
  - delayed_on: 1s
```

**Traceback (if applicable):**

```

```

**Additional information:**

brandond, 2019-01-28:
try turning up the log level to VERBOSE, which is (counter-intuitively) higher than DEBUG. You should see some messages about the initial state of your pins. I had similar issues with binary inputs triggering STATE properly, but not CLICK. Turned out I needed to invert them; as far as esphome was concerned they were always 'on' and I was actually turning them 'off' for a brief moment when I clicked the button.

brandond, 2019-01-29:
right, you should also see some messages about the initial state of each of those sensors - whether they're ON or OFF at boot.

MrPaulAR, 2019-01-29:
Unless I'm missing it here is the complete messages from boot.

```
Hard resetting via RTS pin...
============================================================================ [SUCCESS] Took 95.63 seconds ============================================================================
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[01:07:28][D][WiFiGeneric.cpp:345] _eventCallback(): Event: 1 - SCAN_DONE
[01:07:28][V][wifi:1199]: Event: WiFi Scan Done status=0 number=2 scan_id=128
[01:07:28][D][wifi:275]: Found networks:
[01:07:28][I][wifi:308]: - 'XXXXXXXXXXXXX' (XXXXXXXXXXXXX) ▂▄▆█
[01:07:28][D][wifi:309]:     Channel: 1
[01:07:28][D][wifi:310]:     RSSI: -17 dB
[01:07:28][D][wifi:312]: - 'XXXXXXXXXXXXX' (XXXXXXXXXXXXX) ▂▄▆█
[01:07:28][I][wifi:179]: WiFi Connecting to 'XXXXXXXXXXXXX'...
[01:07:28]I (3118) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[01:07:29]I (3853) wifi: state: init -> auth (b0)
[01:07:29]I (3855) wifi: state: auth -> init (2c0)
[01:07:29]I (3855) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[01:07:29][D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[01:07:29][W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 202 - AUTH_FAIL
[01:07:29][W][wifi:1225]: Event: Disconnected ssid='XXXXXXXXXXXXX' bssid=XXXXXXXXXXXXX reason=Authentication Failed
[01:07:29][W][wifi:369]: Error while connecting to network.
[01:07:29][W][wifi:399]: Restarting WiFi adapter...
[01:07:29][V][wifi:983]: Disabling STA.
[01:07:29][D][WiFiGeneric.cpp:345] _eventCallback(): Event: 3 - STA_STOP
[01:07:29][V][wifi:1207]: E
[01:07:29]899) wifi: stop sw txq: WiFi STA stop
[01:07:29]I (3899) wifi: stop sw txq
[01:07:29]I (3908) wifi: lmac stop hw txq
[01:07:29][I][wifi:179]: WiFi Connecting to 'XXXXXXXXXXXXX'...
[01:07:29][V][wifi:981]: Enabling STA.
[01:07:29]I (4016) wifi: mode : sta (XXXXXXXXXXXXX)
[01:07:29][D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
[01:07:29][D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[01:07:29][V][wifi:1203]: Event: WiFi STA start
[01:07:29][V][wifi:1193]: Event: WiFi ready
[01:07:30]I (4233) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
[01:07:30]I (4968) wifi: state: init -> auth (b0)
[01:07:30]I (4974) wifi: state: auth -> assoc (0)
[01:07:30]I (4993) wifi: state: assoc -> run (10)
[01:07:30]I (5047) wifi: connected with XXXXXXXXXXXXX, channel 1
[01:07:30]I (5047) wifi: pm start, type: 0
[01:07:30]
[01:07:30][D][WiFiGeneric.cpp:345] _eventCallback(): Event: 4 - STA_CONNECTED
[01:07:30][V][wifi:1216]: Event: Connected ssid='Pretty Fly For A WiFi' bssid=XXXXXXXXXXXXX channel=1, authmode=WPA2 PSK
[01:07:31][D][WiFiGeneric.cpp:345] _eventCallback(): Event: 7 - STA_GOT_IP
[01:07:31][D][WiFiGeneric.cpp:389] _eventCallback(): STA IP: 192.168.86.21, MASK: 255.255.255.0, GW: 192.168.86.1
[01:07:31][V][wifi:1237]: Event: Got IP static_ip=192.168.86.21 gateway=192.168.86.1
[01:07:31][I][wifi:349]: WiFi connected!
[01:07:31][C][wifi:240]:   SSID: 'XXXXXXXXXXXXX'
[01:07:31][C][wifi:241]:   IP Address: 192.168.86.21
[01:07:31][C][wifi:243]:   BSSID: XXXXXXXXXXXXX
[01:07:31][C][wifi:245]:   Hostname: 'lolin_d32_test'
[01:07:31][C][wifi:250]:   Signal strength: -12 dB ▂▄▆█
[01:07:31][C][wifi:251]:   Channel: 1
[01:07:31][C][wifi:252]:   Subnet: 255.255.255.0
[01:07:31][C][wifi:253]:   Gateway: 192.168.86.1
[01:07:31][C][wifi:254]:   DNS1: 192.168.86.1
[01:07:31][C][wifi:255]:   DNS2: 0.0.0.0
[01:07:31][C][api:023]: Setting up Home Assistant API server...
[01:07:31][C][ota:129]: Over-The-Air Updates:
[01:07:31][C][ota:130]:   Address: 192.168.86.21:3232
[01:07:31][W][ota:136]: Last Boot was an unhandled reset, will proceed to safe mode in 7 restarts
[01:07:31][I][application:089]: setup() finished successfully!
[01:07:31][I][application:097]: You're running esphomelib v1.10.1 compiled on Jan 29 2019, 00:24:32
[01:07:31][C][wifi:341]: WiFi:
[01:07:31][C][wifi:240]:   SSID: 'XXXXXXXXXXXXX'
[01:07:31][C][wifi:241]:   IP Address: 192.168.86.21
[01:07:31][C][wifi:243]:   BSSID: XXXXXXXXXXXXX
[... 38 lines cut ...]
[01:07:31][C][binary_sensor.gpio:024]:   Pin: GPIO4 (Mode: INPUT_PULLUP)
[01:07:31][C][binary_sensor.gpio:023]: GPIO Binary Sensor 'PIN_2'
[01:07:31][C][binary_sensor.gpio:024]:   Pin: GPIO2 (Mode: INPUT_PULLUP)
[01:07:31][C][binary_sensor.gpio:023]: GPIO Binary Sensor 'PIN_15'
[01:07:31][C][binary_sensor.gpio:024]:   Pin: GPIO15 (Mode: INPUT_PULLUP)
[01:07:31][C][logger:099]: Logger:
[01:07:31][C][logger:100]:   Level: VERBOSE
[01:07:31][C][logger:101]:   Log Baud Rate: 115200
[01:07:31][C][sensor.wifi_signal:042]: WiFi Signal 'WiFi Signal Sensor'
[01:07:31][C][sensor.wifi_signal:042]:   Unit of Measurement: 'dB'
[01:07:31][C][sensor.wifi_signal:042]:   Accuracy Decimals: 0
[01:07:31][C][sensor.wifi_signal:042]:   Icon: 'mdi:wifi'
[01:07:31][V][sensor.wifi_signal:042]:   Unique ID: 'XXXXXXXXXXXXX-wifisignal'
[01:07:31][C][api:072]: API Server:
[01:07:31][C][api:073]:   Address: 192.168.86.21:6053
[01:07:31][C][ota:129]: Over-The-Air Updates:
[01:07:31][C][ota:130]:   Address: 192.168.86.21:3232
[01:07:31][W][ota:136]: Last Boot was an unhandled reset, will proceed to safe mode in 7 restarts
[01:07:31][I][application:114]: Running through first loop()
[01:07:31][V][sensor.sensor:020]: 'WiFi Signal Sensor': Received new state -13.000000
[01:07:31][I][application:141]: First loop finished successfully!
[01:07:42][V][api:500]: Hello from client: 'Home Assistant 0.86.2 (192.168.86.36)'
[01:07:42][D][api:531]: Client 'Home Assistant 0.86.2 (192.168.86.36)' connected successfully!
[01:08:23][V][sensor.sensor:020]: 'WiFi Signal Sensor': Received new state -13.000000
```

OttoWinter, 2019-02-01:
Fixed in https://github.com/esphome/ESPHome-Core/pull/454 (I think from looking at this quickly. Re-Open if not the case)

closed 2019-02-01 by OttoWinter