esphome/esphome #400: Can't compile binary sensor

swampyclark, 2019-01-31:
**Operating environment (Hass.io/Docker/pip/etc.):**

Hassio 0.86.2
**ESP (ESP32/ESP8266/Board/Sonoff):**

NodeMcu v2
**Affected component:**

https://esphomelib.com/esphomeyaml/components/binary_sensor/gpio.html

**Description of problem:**
Won't compile, use to work fine a few days ago.

**Problem-relevant YAML-configuration entries:**
binary_sensor:
  - platform: gpio
    pin: D2
    name: "Living Room Window"
    device_class: window

**Traceback (if applicable):**

```
LibraryManager: Installing id=305 @ 1.1.3
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing esphomelib
git version 2.17.1
Cloning into '/config/esphomeyaml/test/.piolibdeps/_tmp_installing-pBksOh-package'...
LibraryManager: Installing id=3837 @ 5.13.3
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=1826 @ 1.0.3
Downloading...
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.1.3
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphomelib> 1.11.0-dev #6b3bc24
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESPAsyncTCP> 1.1.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESP8266mDNS>
|   |   |   |-- <ESP8266WiFi> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <Wire> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/test/src/main.cpp.o
Generating LD script .pioenvs/test/ld/eagle.app.v6.common.ld
Compiling .pioenvs/test/lib4d9/ESP8266WiFi/BearSSLHelpers.cpp.o
Compiling .pioenvs/test/lib4d9/ESP8266WiFi/CertStoreBearSSL.cpp.o
Compiling .pioenvs/test/lib4d9/ESP8266WiFi/ESP8266WiFi.cpp.o
Compiling .pioenvs/test/lib4d9/ESP8266WiFi/ESP8266WiFiAP.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:21:3: error: 'MakeGPIOBinarySensor' is not a member of 'esphomelib::Application'
Application::MakeGPIOBinarySensor application_makegpiobinarysensor = App.make_gpio_binary_sensor("Living Room Window", 4);
^
src/main.cpp:21:37: error: expected ';' before 'application_makegpiobinarysensor'
Application::MakeGPIOBinarySensor application_makegpiobinarysensor = App.make_gpio_binary_sensor("Living Room Window", 4);
^
src/main.cpp:22:87: error: 'application_makegpiobinarysensor' was not declared in this scope
binary_sensor::GPIOBinarySensorComponent *binary_sensor_gpiobinarysensorcomponent = application_makegpiobinarysensor.gpio;
^
src/main.cpp:23:3: error: 'MQTTBinarySensorComponent' is not a member of 'esphomelib::binary_sensor'
binary_sensor::MQTTBinarySensorComponent *binary_sensor_mqttbinarysensorcomponent = application_makegpiobinarysensor.mqtt;
^
src/main.cpp:23:45: error: 'binary_sensor_mqttbinarysensorcomponent' was not declared in this scope
binary_sensor::MQTTBinarySensorComponent *binary_sensor_mqttbinarysensorcomponent = application_makegpiobinarysensor.mqtt;
^
*** [.pioenvs/test/src/main.cpp.o] Error 1
```

**Additional information:**

brandond, 2019-01-31:
You're running dev, but your core and yaml are out of sync. Update your add-on and try again?

swampyclark, 2019-01-31:
ok ill try that.  ?  how did you know the yaml version number mismatch from the log, i added the "esphomeyaml_version": "dev" in my config, but did not figure this out.  thanks for the info.

brandond, 2019-01-31:
You get weird errors like this when the esphome Python code generates C++ that doesn't match what's in esphome-core.

OttoWinter, 2019-02-01:
Yes, the esphomeyaml and esphomelib versions need to match

closed 2019-02-01 by OttoWinter

ioudeveldhuis, 2019-02-06:
For me the clean build doesn't work. I keep running into the following errors:

```
X:\Projecten\esphomeyaml>esphomeyaml config/woonkamerble.yaml clean
[32mINFO Reading configuration...[0m
[32mINFO Deleting config\woonkamerble\.piolibdeps[0m
[32mINFO Deleting config\woonkamerble\.pioenvs[0m
[32mINFO Done![0m

X:\Projecten\esphomeyaml>esphomeyaml config/woonkamerble.yaml run
[32mINFO Reading configuration...[0m
[32mINFO Generating C++ source...[0m
[32mINFO Compiling app...[0m
[32mINFO Running:  platformio run -d 'config\woonkamerble'[0m
Processing woonkamerble (platform: https://github.com/platformio/platform-espressif32.git#feature/stage; framework: arduino; board: nodemcu-32s)
-----------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing id=1826 @ 1.0.3
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing esphomelib
git version 2.19.2.windows.1
Cloning into 'X:\Projecten\esphomeyaml\config\woonkamerble\.piolibdeps\_tmp_installing-ymummb-package'...
remote: Enumerating objects: 388, done.
remote: Counting objects: 100% (388/388), done.
remote: Compressing objects: 100% (377/377), done.
remote: Total 388 (delta 40), reused 73 (delta 9), pack-reused 0
Receiving objects: 100% (388/388), 421.34 KiB | 967.00 KiB/s, done.
Resolving deltas: 100% (40/40), done.
Checking out files: 100% (363/363), done.
LibraryManager: Installing id=3837 @ 5.13.3
Downloading...
Unpacking  [####################################]  100%
LibraryManager: Installing id=305 @ 1.1.3
Downloading...
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 (Stage) > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <Update> 1.0
|-- <esphomelib> 1.11.0-dev #0bea62d
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <AsyncTCP> 1.0.3
|   |-- <Update> 1.0
|   |-- <WiFi> 1.0
|   |-- <Wire> 1.0.1
|   |-- <ArduinoOTA> 1.0
|   |   |-- <ESPmDNS> 1.0
|   |   |   |-- <WiFi> 1.0
|   |   |-- <Update> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <Preferences> 1.0
Compiling .pioenvs\woonkamerble\src\main.cpp.o
Generating partitions .pioenvs\woonkamerble\partitions.bin
Compiling .pioenvs\woonkamerble\libc97\AsyncTCP_ID1826\AsyncTCP.cpp.o
Compiling .pioenvs\woonkamerble\libdc6\WiFi\ETH.cpp.o
Compiling .pioenvs\woonkamerble\libdc6\WiFi\WiFi.cpp.o
Compiling .pioenvs\woonkamerble\libdc6\WiFi\WiFiAP.cpp.o
Compiling .pioenvs\woonkamerble\libdc6\WiFi\WiFiClient.cpp.o
Archiving .pioenvs\woonkamerble\libc97\libAsyncTCP_ID1826.a
Compiling .pioenvs\woonkamerble\libdc6\WiFi\WiFiGeneric.cpp.o
src\main.cpp: In function 'void setup()':
src\main.cpp:27:3: error: 'MQTTBinarySensorComponent' is not a member of 'esphomelib::binary_sensor'
binary_sensor::MQTTBinarySensorComponent *binary_sensor_mqttbinarysensorcomponent = App.register_binary_sensor(esp32blepresencedevice);
^
src\main.cpp:27:45: error: 'binary_sensor_mqttbinarysensorcomponent' was not declared in this scope
binary_sensor::MQTTBinarySensorComponent *binary_sensor_mqttbinarysensorcomponent = App.register_binary_sensor(esp32blepresencedevice);
^
*** [.pioenvs\woonkamerble\src\main.cpp.o] Error 1
============================================= [ERROR] Took 32.24 seconds =============================================

X:\Projecten\esphomeyaml>
```

Yaml file:
```
esphomeyaml:
  name: woonkamerble
  platform: ESP32
  board: nodemcu-32s
  esphomelib_version: dev
  arduino_version: dev

wifi:
  ssid: '<SSID>'
  password: '<PASSWORD>'

esp32_ble_tracker:
  scan_interval: 90s

binary_sensor:
  - platform: esp32_ble_tracker
    mac_address: X5:X0:X1:X5:X4:X9
    name: "MIBand Ivo woonkamer"

# Enable logging
logger:
  level: INFO

# Enable Home Assistant API
api:
  password: 'XXXXXX'

ota:
  password: 'XXXXXX'

X:\Projecten\esphomeyaml>
```

brandond, 2019-02-06:
@ioudeveldhuis update your esphomeyaml, then do a clean and build. Either one can get out of sync with the other when using the dev branch.