esphome/esphome #401: Template Binary Sensor doesn't send to MQTT

yottatsa, 2019-02-01:
**Operating environment (Hass.io/Docker/pip/etc.):** MacOS / venv from dev-branch

**ESP (ESP32/ESP8266/Board/Sonoff):** ESP-01

**Affected component:** https://esphomelib.com/esphomeyaml/components/binary_sensor/template.html

**Description of problem:** Sensor doesn't send to MQTT

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: template
    id: offline
    name: "Heater Offline"
    device_class: problem
    lambda: |-
```

**Traceback (if applicable):**

```
 MQTT:                                                                                                        
   Discovery prefix: 'homeassistant'                                                
   Discovery retain: YES                                                            
   Topic Prefix: 'heater'                                                           
   Log Topic: 'heater/debug'                                                        
   Availability: 'heater/status'                                                    
 MQTT Message Trigger:
   Topic: 'heater/switch/heater/command'                                            
   QoS: 0
 MQTT Sensor 'Heater WiFi Signal':                                                  
   Expire After: 120s
   State Topic: 'heater/sensor/heater_wifi_signal/state'                            
 MQTT switch 'Heater':
   State Topic: 'heater/switch/heater/state'                                        
   Command Topic: 'heater/switch/heater/command'
...
Jan 31 00:38:24 home esphomeyaml[751]: [00:38:24][D][binary_sensor:037]: 'Heater Offline': Sending state OFF
```

**Additional information:**

brandond, 2019-02-01:
I see you've got a template binary sensor with an empty lambda. Your log shows a successful 'OFF' state published to MQTT. What's the issue?

yottatsa, 2019-02-01:
@brandond it doesn't send the MQTT message, and it is not registered (only Sensor and switch are reported).

> OttoWinter: Yeah. the issue is in `make_template_binary_sensor`, it doesn't `register` the binary sensor (only the component)