esphome/esphome #403: Custom component sensor not creating sensor names

fransvdscheer, 2019-02-07:
Have used esphomeyaml to program couple of ESP's. Great stuff this code generator. Saves lot of time C++ programming. Thanks to the developer(s). 

Have a peacefair PZEM-016 (device to measure voltage, current, power, energy, cos-phi, etc. ; Low cost ~10 Euro). Device has a modbus connection and got it to work on my ESP by adding C++ code directly into main.cpp.  Next step to create it into a custom sensor component with multiple output values, but no sensor entities / names are created for the various parameters.   

Seen someone else raising same question on topic. Responds was a few adjustements in python scripts, but have no clue how to implement that into my hassio environment.     

Appreciate fix or work around idea

OttoWinter, 2019-02-08:
Fixed in dev.

closed 2019-02-08 by OttoWinter

fransvdscheer, 2019-02-09:
Tested the dev as you suggested.        
It creates a lot of compilation errors on mqtt, while the "latest" version compiles great
Dump from the compiler below

src/main.cpp: In function 'void setup()':
src/main.cpp:40:3: error: 'mqtt' has not been declared
mqtt::MQTTClientComponent *mqtt_mqttclientcomponent = App.init_mqtt("192.168.1.222", 1883, "xxx", "yyyy");
^
src/main.cpp:40:30: error: 'mqtt_mqttclientcomponent' was not declared in this scope
mqtt::MQTTClientComponent *mqtt_mqttclientcomponent = App.init_mqtt("192.168.1.222", 1883, "xxx", "yyyy");
^
src/main.cpp:40:61: error: 'class esphomelib::Application' has no member named 'init_mqtt'
mqtt::MQTTClientComponent *mqtt_mqttclientcomponent = App.init_mqtt("192.168.1.222", 1883, "xxx", "yyyy");
^
src/main.cpp:50:94: error: void value not ignored as it ought to be
sensor::MQTTSensorComponent *sensor_mqttsensorcomponent = App.register_sensor(sensor_sensor);
^
src/main.cpp:52:98: error: void value not ignored as it ought to be
sensor::MQTTSensorComponent *sensor_mqttsensorcomponent_2 = App.register_sensor(sensor_sensor_2);
^
src/main.cpp:53:3: error: 'MakeGPIOSwitch' is not a member of 'esphomelib::Application'
Application::MakeGPIOSwitch application_makegpioswitch = App.make_gpio_switch("Watchdog Led", 18);
^
src/main.cpp:54:34: error: 'application_makegpioswitch' was not declared in this scope
switch_::GPIOSwitch *knipper = application_makegpioswitch.switch_;
^
src/main.cpp:55:3: error: 'MQTTSwitchComponent' is not a member of 'esphomelib::switch_'
switch_::MQTTSwitchComponent *switch__mqttswitchcomponent = application_makegpioswitch.mqtt;
^
src/main.cpp:55:33: error: 'switch__mqttswitchcomponent' was not declared in this scope
switch_::MQTTSwitchComponent *switch__mqttswitchcomponent = application_makegpioswitch.mqtt;
^
src/main.cpp:56:3: error: 'MakeGPIOSwitch' is not a member of 'esphomelib::Application'
Application::MakeGPIOSwitch application_makegpioswitch_2 = App.make_gpio_switch("Test Led", 19);
^
src/main.cpp:57:35: error: 'application_makegpioswitch_2' was not declared in this scope
switch_::GPIOSwitch *vermogen = application_makegpioswitch_2.switch_;
^
src/main.cpp:58:3: error: 'MQTTSwitchComponent' is not a member of 'esphomelib::switch_'
switch_::MQTTSwitchComponent *switch__mqttswitchcomponent_2 = application_makegpioswitch_2.mqtt;
^
src/main.cpp:58:33: error: 'switch__mqttswitchcomponent_2' was not declared in this scope
switch_::MQTTSwitchComponent *switch__mqttswitchcomponent_2 = application_makegpioswitch_2.mqtt;
^

OttoWinter, 2019-02-09:
@fransvdscheer Your esphomeyaml and esphomelib versions need to match. If you use esphomelib version dev you must also use esphomeyaml version dev.

fransvdscheer, 2019-02-10:
This is what happens with both on dev.      The generator/ compiler stops with these comments below:

INFO Reading configuration...
INFO Generating C++ source...
ERROR ID sensor_sensor is already registered