esphome/esphome #405: compilation terminated fatal error: Wire.h: No such file or directory

mvturnho, 2019-02-07:
**Operating environment (Hass.io/Docker/pip/etc.):**

Windows 10 commandline and Visual C++
**ESP (ESP32/ESP8266/Board/Sonoff):**

ESP8266 nosemcu
**Affected component:**

https://esphomelib.com/esphomeyaml/custom/custom_component.html

**Description of problem:**
I added the library `I2Cdevlib-esphomelib` to the yaml file to use the MPR121 driver from my custom component.
Then when compiling the yaml file I get the compile errors.

```
**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

#include <Wire.h>
^
compilation terminated.
In file included from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\BMP085.h:36:0,
from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\BMP085.cpp:33:
.piolibdeps\I2Cdevlib-esphomelib_ID3838\src\I2Cdev.h:53:18: fatal error: Wire.h: No such file or directory

```

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: controlpanel
  platform: ESP8266
  board: nodemcuv2
  esphomelib_version:
    branch: dev
  includes:
    - controlpanel/src/my_custom_sensor.h
  libraries:
    - I2Cdevlib-esphomelib

wifi:
  ssid: 'XXXXXXXXXX'
  password: 'XXXXXXXXXXXXXXXXX'
  domain : .visconti.nl
  
mqtt:
  broker: '192.168.3.100'
  port: 1883

# Enable logging
logger:
  level: DEBUG

ota:
  safe_mode: True
  password: VERYSECURE

i2c:
  sda: GPIO12
  scl: GPIO14
  scan: False

custom_component:
    - lambda: |-
        auto my_custom = new MyCustomComponent();
        return {my_custom};
```

Custom component
```cpp
#include "esphomelib/application.h"

using namespace esphomelib;
#include "MPR121.h"

class MyCustomComponent : public Component {
 public:
  void setup() override {
    // This will be called once to set up the component
    // think of it as the setup() call in Arduino

  }
  void loop() override {
    // This will be called very often after setup time.
    // think of it as the loop() call in Arduino

    ESP_LOGD("custom", "Message");
  }
};
```

**Traceback (if applicable):**

```
Scanning dependencies...
Dependency Graph
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|-- <FastLED> 3.2.0
|-- <Hash> 1.0
|-- <I2Cdevlib-esphomelib> 1.1.0
|-- <Wire> 1.0
|-- <esphomelib> 1.11.0-dev #0bea62d
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESPAsyncTCP> 1.2.0
Compiling .pioenvs\controlpanel\src\main.cpp.o
Compiling .pioenvs\controlpanel\lib96b\I2Cdevlib-esphomelib_ID3838\AD7746.cpp.o
Compiling .pioenvs\controlpanel\lib96b\I2Cdevlib-esphomelib_ID3838\ADS1115.cpp.o
Compiling .pioenvs\controlpanel\lib96b\I2Cdevlib-esphomelib_ID3838\BMP085.cpp.o
In file included from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\AD7746.h:38:0,
from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\AD7746.cpp:35:
.piolibdeps\I2Cdevlib-esphomelib_ID3838\src\I2Cdev.h:53:18: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

#include <Wire.h>
^
compilation terminated.
In file included from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\ADS1115.h:40:0,
from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\ADS1115.cpp:36:
.piolibdeps\I2Cdevlib-esphomelib_ID3838\src\I2Cdev.h:53:18: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

#include <Wire.h>
^
compilation terminated.
In file included from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\BMP085.h:36:0,
from .piolibdeps\I2Cdevlib-esphomelib_ID3838\src\BMP085.cpp:33:
.piolibdeps\I2Cdevlib-esphomelib_ID3838\src\I2Cdev.h:53:18: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

#include <Wire.h>
^
compilation terminated.
*** [.pioenvs\controlpanel\lib96b\I2Cdevlib-esphomelib_ID3838\ADS1115.cpp.o] Error 1
*** [.pioenvs\controlpanel\lib96b\I2Cdevlib-esphomelib_ID3838\AD7746.cpp.o] Error 1
*** [.pioenvs\controlpanel\lib96b\I2Cdevlib-esphomelib_ID3838\BMP085.cpp.o] Error 1
 [ERROR] Took 3.95 seconds

```

**Additional information:**

mvturnho, 2019-02-07:
This is solved with <esphomelib> 1.10.1.
After Update the compile did not produce any errors

mvturnho, 2019-02-07:
@OttoWinter could we reopen this issue?

I tried again with another yaml project and with the latest esphomeyaml version 1.11.0-dev and it gives the error:

```
In file included from src\main.cpp:5:0:
src\custom_i2c.h:2:18: fatal error: Wire.h: No such file or directory
```

when using version 1.10.1 it compiles just fine

brandond, 2019-02-07:
Guessing this is due to LDF being disabled - if you don't include an i2c entry in your YAML, it has no idea that your custom component wants to use Wire.h. @ottowinter may have a better solution, but adding Wire to your library list in platformio.ini is probably the easy fix.