esphome/esphome #415: Device Connects With Dynamic IP But Not Static

danieledwardgeorgehitchcock, 2019-02-10:
**Operating environment (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266/Board/Sonoff):**

**Affected component:**

**Description of problem:**
When I compile and upload my YAML to the Sonoff Basic, I cannot get it to work with a static IP. If I comment out the manual_ip section of the configuration, the device connects to my AP with no issues however, if I try and connect with the manual_ip section below, it doesn't connect. I can confirm that my router's DHCP IP range is set from 1-100 so, 101 should be fine for a static IP. I have also done an nslookup and ping on port 101 and can confirm that there are no devices occupying that address.

**Problem-relevant YAML-configuration entries:**
```yaml
# Device Settings
esphomeyaml:
  name: dining_room_airwick_infuser
  platform: ESP8266
  board: esp01_1m

# WiFi Settings
wifi:
  ssid: OBSCURED_FOR_PROTECTION
  password: OBSCURED_FOR_PROTECTION
  manual_ip:
    static_ip: 192.168.1.101
    gateway: 192.168.1.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

# Enable Over The Air Flashing
ota:

# Define I/O
binary_sensor:
  - platform: gpio
    pin:
      number: 0
      mode: INPUT_PULLUP
      inverted: True
    name: Dressing Room AirWick Infuser Button
    on_press:
      - switch.toggle: relay

switch:
  - platform: gpio
    name: Dressing Room AirWick Infuser Relay
    id: relay
    pin:
      number: 12

output:
  - platform: esp8266_pwm
    id: led
    pin:
      number: 13
      inverted: True

light:
  - platform: monochromatic
    name: Dressing Room AirWick Infuser LED
    output: led
```

**Traceback (if applicable):**

brandond, 2019-02-10:
Just out of curiosity, does it work with `fast_connect: True` in your wifi section?

danieledwardgeorgehitchcock, 2019-02-10:
Yes that seems to work - Surely if my SSID is not set to hidden then it should work without that setting?

Could it be due to me having a mesh network, or that my 2.4 and 5ghz occupy the same SSID?

OttoWinter, 2019-02-10:
fast_connect makes esphome use the default wifi connection mechanism (no scanning). Sometimes that works a bit better - with these things though it is often a bit trial and error because the SDK's wifi code is closed source.