esphome/esphome #424: yaml !include whole file

quazzie, 2019-02-11:
Add ability to include a whole file in yaml.
```
test.yaml:
!include other.yaml
```

**Is your feature request related to a problem/use-case? Please describe.**

This would allow templating with substitutions.
base.yaml:
```
esphomeyaml:
  name: $hostname
  platform: ESP8266
  ...
```

node1.yaml:
```
substitutions:
  hostname: test
!include base.yaml
```

**Describe the solution you'd like:**

Top level !include in .yaml files

**Additional context:**

OttoWinter, 2019-02-11:
That's not possible with YAML. Top-level tags are not supported by the language.

You can use this trick:

```yaml
.dummy: !include &dummy base.yaml
<<: *dummy
```

closed 2019-02-11 by OttoWinter

quazzie, 2019-02-11:
Is that the way we are suppose to use it according to https://esphome.io/guides/configuration-types.html?highlight=substitution ?
`With substitutions, you can have a single generic source file for all nodes of one kind and substitute expressions in.`
If that is the case please add it to the documentation :)

quazzie, 2019-02-11:
@OttoWinter but yaml as a language does not support !include either (afaics) so whats different with a toplevel include ?

And yes, your trick works. Now i can have a base.yaml using substitution variables and then define the substitutions in node<1..*>.yaml, super great!

OttoWinter, 2019-02-11:
Yes !include is a YAML extension that HA and ESPHome add. Those are however valid YAML (using tags). A top-level include would not be valid YAML and we would have to change the YAML reader for that.

brandond, 2019-02-16:
FYI, I wasn't even aware that ESPHome supported !include until this issue came by. Definitely could use some documentation. That and substitution would probably make for a good page on meta-programming the YAML templates; I'll put that in my mental backlog.

OttoWinter, 2019-02-16:
@brandond For `!include`, see first entry in FAQ 😉  - though the `<<` could probably be added there too.