esphome/esphome #425: Monochromatic light. PWM. Dimming

erazor666, 2019-02-11:
I am not 100% sure this is am actual issue with esphome, but i'm not sure what is the cause.

I have some pwm driven monochromatic lights and they function like expected mostly. However when dimming them the dim 'curve' is a litte strange. From 0 to about 50-60% intensity they are kind of dull and not lighting up much, and from there and up to 100% it feels like too much is happening. The distrubution of light seems not to be completely linear.

My theories is something esphome related or the mosfet gates are not the optimal ones for this application. I'm using irfz44n mosfets and a wemos d1 esp8266. I dont know if i should expect other results using a esp32?

OttoWinter, 2019-02-11:
ESPHome uses gamma correction by default - the formula to compute the actual output it `x^2.8` (gamma correction factor 2.8). I believe this is what you're seeing; for most lights this gamma correction is good because humans perceive light logarithmically. 

Are you driving the LEDs directly, or is there some controller in between. Maybe also the LEDs in use do something. Anyway, you can disable gamma correction (see docs)

erazor666, 2019-02-11:
They are, driven directly i guess you'd say. The designated pwm pin goes directly to the mosfet. The 'software pwm' issues with esp8266 isn't much related to the problem i'm seeing as far i understand. I have virtually no flickering or flashing problems.

yawor, 2019-02-12:
@erazor666 the irfz44 MOSFET you're using is not meant to be driven with 3.3v. It's gate opens fully at about 4v. Your MOSFET is not fully opening, which means it's resistance doesn't drop to correct value when driven. This means it draws current itself and heats up much much more than it should. It may also affect your lights.
You need to either convert voltage from the esp's pin to be more than 4v or use a different transistor. A good choice is the irlz44, which is almost identical to irfz44, but its minimum gate to source voltage is only 3v (or even lower, I don't remember right now), so it opens fully when driven from 3.3v logic.

OttoWinter, 2019-02-12:
@yawor That is correct too. I do not think that would result in this behavior, but I might be wrong.

@erazor666 Have you tried disabling gamma correction?

erazor666, 2019-02-13:
I will try out both tips as soon as possible, either will be a pretty easy fix since the mosfets aren't soldered to my boards and the code is quick to fix. I will test them individually and report back. 
Either solution sounds likely at this time.

yawor, 2019-02-13:
@erazor666 even if the IRFZ44 is not causing this issue, I'd still replace it with IRLZ44. Depending on your load, it'll run much cooler and with better efficiency. It may affect the LED brightness overall.