esphome/esphome #430: DO NOT POST ISSUES HERE

OttoWinter, 2019-02-13:
For bug reports, please go to https://github.com/esphome/issues

For feature requests, go to https://github.com/esphome/feature-requests

Thank you!

Fusseldieb, 2020-05-28:
Why that? It's kinda unintuitive having two separate empty repos just for that... 

Better make a badge with "Bug" or "Feature Request" and you can then search by badges.

h0jeZvgoxFepBQ2C, 2020-07-22:
Is there any reason why this is like it is? Just asking, no offense :)

OttoWinter, 2020-07-26:
@h0jeZvgoxFepBQ2C It's mostly because of this repo's history (IIRC, it's been some time). First, esphomelib was C++-only project (where users had to type C++ themselves), then esphomeyaml was added as a layer that generates the C++ from YAML (that is this repo today).

Back then the issue was that users would create issues randomly split between in the esphomelib and -yaml repos, which made the bug tracker hard to use. So the bug tracker was moved to a separate bug-tracker only repo.

Then the C++ codebase got merged into this project and everything was renamed `esphome`.  Changing the bug/feature tracker now would be quite hard.

lixianglan1, 2021-09-13:
1

davidglezz, 2023-11-19:
https://docs.github.com/en/issues/tracking-your-work-with-issues/transferring-an-issue-to-another-repository

https://cli.github.com/manual/gh_issue_transfer

I think it would be easy to make a script that transfers all the issues here

RubenKelevra, 2024-09-23:
Sounds like a good idea actually, why don't you do that? :)