esphome/esphome #604: BME280 issue on ESP8266 + ESP32

bladeback, 2019-06-03:
ESPHOME finds it on BME280 (0x76) but cannot communicate with it. How can I fix it?
This happens to me on both the ES8266 and ESP32.
The error message is: Communication with BME280 failed!

closed 2019-06-03 by close-issue-app[bot]

GrowFOUR, 2019-06-06:
Same issue here. Tried multiple sensors

bladeback, 2019-06-06:
> Same issue here. Tried multiple sensors

And what sensors should I include to make it work? I tried HTU21D, BMP280, nothing goes on I2C bus.

GrowFOUR, 2019-06-10:
Just saying I tried 2 different BME280 sensors to see if one was malfunctioning. But I get the same communication error with both. I used the light sensor BH1750 and that worked fine with esphome

bladeback, 2019-06-10:
> Jen jsem řekl, že jsem vyzkoušel 2 různé senzory BME280, abych zjistil, zda nedošlo k poruše. Ale dostanu stejnou komunikační chybu s oběma. Použil jsem světelný senzor BH1750 a ten fungoval dobře s esphome

I agree, I tried 5 different BME280s, no one works, neither at 0x76 nor at 0x77 address. BH1750 works. So there's something wrong with the ESPHome. Unfortunately, I did not find out where to send a question to get things done.