esphome/esphome #3472: esp8285 reboot due to incorrect API or wireless timeout

platima, 2022-05-14:
I have a pile of Espressif MCU's connected to ESPHome within HA. Everything latest.

I've found that the ESP8285 crashes every 15-30 mins, and I narrowed that down to either a Wifi or API timeout. Setting both to 0s works fine.

The YAML I am using now is 
```
esphome:
  name: csh-plgusbpm
  comment: Labelled Plug01

esp8266:
  board: esp8285

logger:

api:
  reboot_timeout: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  reboot_timeout: 0s
```

However it'd be great to actually not have to set these, as they are not actually timing out. My AP does not show it disconnecting until after the reboot, and when I watch the logs everything is working fine until the API says read error on the 15 or 30 min mark, then it reconnects a few seconds later. So it seems like the timeout is not being counted correctly.

nagyrobi, 2022-05-14:
You don't read, don't you?

platima, 2022-05-14:
> You don't read, don't you?

I don't appreciate your passive aggressive nature there. Yes, I do read, and I scoured the internet for two days and tried all sorts of things before putting this issue up - even then, I checked the existing issues first. I am new to esphome, but I am not new to the development process and lodging issues.

Once you're over your faux intelligence bender, could you please actually assist or indicate what I missed?

Thank you.

markus-seidl, 2022-05-15:
It seems there is some aggression ongoing, so I only reference this issue [#1 ](https://github.com/esphome/esphome/issues/430) . 
Poste issues here: https://github.com/esphome/issues

nagyrobi, 2022-05-15:
> could you please actually assist or indicate what I missed?

Hm. Do you see anything special on this link?
https://github.com/esphome/esphome/issues

platima, 2022-05-15:
> It seems there is some aggression ongoing, so I only reference this issue [#1 ](https://github.com/esphome/esphome/issues/430) . Poste issues here: https://github.com/esphome/issues

Thanks - will remove this and move it now. I legit did not see that when I posted last night!

platima, 2022-05-15:
> > could you please actually assist or indicate what I missed?
> 
> Hm. Do you see anything special on this link? https://github.com/esphome/esphome/issues

I do not, I did not last night. It may have not loaded the whole page, or it could have been because I had too many tabs open, but it was an honest mistake and there is no need for the attitude.

closed 2022-05-15 by platima